kizniche/Mycodo #499: Can't save GPIO PWM Output.

Kattihatt, 2018-07-01:
So I have an issue where my water pump is building up too much pressure and leading to some leakage issues. It's controlled by a MOSFET (FQP30N06L) that is connected to pin BCM18, which should be the one to use for hardware PWM, which isn't really necessary in this case since I want to run it with a DC motor and don't need high frequencies. When I try to save 50% duty cycle, it never saves. The regular on/off output works perfectly fine one pin18, but I can't save or use the GPIO PWM option. I've tried "50", "50.0", "50%" and "50.0%. Am I doing something wrong or is there some issue with the software?

Picture of settings 
[image]

Kattihatt, 2018-07-01:
Ah, maybe you set it every time you run it :(

Kattihatt, 2018-07-01:
I want to be able to set an output with a fixed duty cycle at 80% and then activate it in the functions just as a regular output. It seems that I can only regulate PWM but not set a timer or treat the PWM output as a regular one.

kizniche, 2018-07-02:
I wasn't able to reproduce your issue. Do you happen to have anything attached to pin 18 or are initializing it elsewhere in Mycodo or other software on your Pi?
[image]

kizniche, 2018-07-02:
Perhaps there's an issue with your pin 18. Have you tried other pins?

Kattihatt, 2018-07-02:
No, I realised that I can run it as software mode and that's enough for me but I can't save it. I thought that I would be able to create a PWM output with a set duty cycle that I could call in functions, just like a regular output. What I wanted to do was create a PWM output for my peristaltic pump with 80% duty cycle and then call it in a function just as a regular output. But when I choose PWM output, I can just vary the output based on some condition. I want a fixed output, that's what I thouht I was doing but it wasn't :)

kizniche, 2018-07-02:
Which type of function are you desiring to trigger the output, PID controllers or Conditionals? I'm not exactly understanding what you want to trigger the 80% duty cycle.

Kattihatt, 2018-07-02:
I want a conditional to trigger the pump with 80% duty cycle. If the soil moisture goes below 30%, then it should start the pump with 80% duty cycle. So I just need a fixed duty cycle on a PWM output.

kizniche, 2018-07-02:
There is a conditional action called "Output (Duty Cycle)" that will set a duty cycle when the conditional is triggered.

Kattihatt, 2018-07-03:
Yes, but then I can set the duty cycle but not the duration like I can on regular output?

kizniche, 2018-07-03:
You can have the same conditional activate a duration conditional with an offset set, that will both turn the output off and deactivate the duration conditional.

Kattihatt, 2018-07-04:
Are you talking about "deactivate controller" now or something? As far as I can see I can choose between duration and duty cycle but not have both and I just looked and can't figure out how I would get both by adding another action.

Kattihatt, 2018-07-04:
Or do you mean I should put a regular output on for let's say 60 seconds, then put a duty cycle action after it and when it turns the output off after those 60 seconds it's all fine and turns both off?

kizniche, 2018-07-04:
Here's an example:

Conditional A (measurement): If temperature > 30, 1. Set Output G to 80% duty cycle, 2. Activate Conditional B.

Conditional B (timer duration): Every 180 seconds, 60 second offset (this offset is how long the duty cycle will stay on for!), then 1. Turn Output G off, 2. Deactivate Conditional B.

The offset will instruct the Conditional to wait before triggering the actions. Make sure the Conditional duration is longer than the offset.

The effect of this setup will be when the temperature is above 30, Output G will turn on for 60 seconds at a duty cycle of 80%.

Kattihatt, 2018-07-05:
I've been here in the greenhouse for two hours now and there's probably been 50 times more drops from my forehead than there has been from the drippers. If I understood you right, this is what I should do:
[image]
[image]

But nothing happens. No response whatsoever. I'm using the light conditions instead of the soil moisture because I can't read that as often without issues. I'm completely stuck here. I thought I did what you said but apparently I'm doing something completely wrong.

kizniche, 2018-07-05:
Also check the daemon log for errors. If the Test1a conditional isn't even triggering, it could be the list isn't low enough, the measurement is beyond the 1000 second max age, or another issue.

leoneo, 2018-08-03:
Hi,

me to i can not save duty cycle %

kizniche, 2018-08-03:
@leoneo: Are you using an appropriate pin (as outlined in the manual)? Have you attempted to use a different pin?

leoneo, 2018-08-03:
only pin 18 (all like kattihatt) library : hardware

kizniche, 2018-08-03:
Try my recommendation.

leoneo, 2018-08-03:
I have try pin 13 with library any pin and frequency to 40000. It is the same think, impossible to configure duty cycle.

kizniche, 2018-08-03:
Your other comments in other threads suggest you have something wrong with your system. All of my systems can set a PWM duty cycle on pins. I suggest installing Raspbian and Mycodo on a formatted SD card and testing on a Pi with nothing connected to it.

kizniche, 2018-08-03:
I can't do anything if an error can't be provided or I can't replicate the issue.

leoneo, 2018-08-03:
I know but all is a fresh install on a formatted SD with the last raspbian, LCD-show, last mycodo.
All my sensors work.

kizniche, 2018-08-03:
If you have tested without anything connected to the Pi, with multiple different browsers, and the issue persists, you may have a damaged Pi. Do you have a second Pi to test the way I described previously?

closed 2018-10-13 by kizniche