kizniche/Mycodo #501: DHT22 error

pmunz75, 2018-07-10:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.4
Raspberry PI 3B+
16 GB SD Card Class10
System : Linux mycodo 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux
Power 5v 2.0 Amp
Sensor : DHT22 ( 3 meters , DS28 , Original Atlas PH 

I've already tried everything.
other pins, more power, solder tested.
The DHT22 runs fine, but only for a few hours. After that, the Raspberry must be taken off and restarted.

Restart via the console is not possible.
Raspberry must be taken off the mains.
After the restart, everything runs again for a few hours

sorry my bad English !

#### Problem Description
Please list:
DHT 22Humidity sensor always fails after 2-5 hours

 
### Errors from Daemon Log
2018-07-10 17:05:08,952 - mycodo.output - WARNING - Cannot turn on Output with ID 0. It doesn't exist
2018-07-10 17:05:16,201 - mycodo.input_d477bba3 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-07-10 17:05:26,845 - mycodo.inputs.dht22_3 - INFO - Turning off sensor
2018-07-10 17:05:26,851 - mycodo.output - WARNING - Cannot turn off Output with ID 0. It doesn't exist
2018-07-10 17:05:29,854 - mycodo.inputs.dht22_3 - INFO - Turning on sensor
2018-07-10 17:05:29,858 - mycodo.output - WARNING - Cannot turn on Output with ID 0. It doesn't exist
2018-07-10 17:05:47,748 - mycodo.inputs.dht22_3 - INFO - Turning off sensor
2018-07-10 17:05:47,754 - mycodo.output - WARNING - Cannot turn off Output with ID 0. It doesn't exist
2018-07-10 17:05:50,759 - mycodo.inputs.dht22_3 - INFO - Turning on sensor
2018-07-10 17:05:50,763 - mycodo.output - WARNING - Cannot turn on Output with ID 0. It doesn't exist
2018-07-10 17:06:08,662 - mycodo.inputs.dht22_3 - INFO - Turning off sensor
2018-07-10 17:06:08,668 - mycodo.output - WARNING - Cannot turn off Output with ID 0. It doesn't exist
2018-07-10 17:06:11,672 - mycodo.inputs.dht22_3 - INFO - Turning on sensor
2018-07-10 17:06:11,678 - mycodo.output - WARNING - Cannot turn on Output with ID 0. It doesn't exist
2018-07-10 17:06:18,926 - mycodo.input_d477bba3 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

Could you please help me please !

kizniche, 2018-07-10:
Looks like you have an issue with pigpio. Try uninstalling it (after deactivating your sensor) and reinstalling it, from the config menu.

pmunz75, 2018-07-10:
56/5000
how can I uninstall and reinstall it?

Service options:
Configure pigpiod
? ---> Uninstall ?

pmunz75, 2018-07-10:
Deactivating all Sensors or only DHT22

pmunz75, 2018-07-12:
DHT22 not run for me , change to BME280

kizniche, 2018-07-13:
The DHT is the only one you have that uses pigpio, so that's the only one you need to deactivate first.

And yes, that's where to reinstall pigpiod.

closed 2018-10-13 by kizniche