kizniche/Mycodo #502: Output usage not calculating

frodus17, 2018-07-10:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.14
 
#### Problem Description
Not seeing any Output usage kWh calculated. I see on time, but no Power Usage or Cost.
 
### Errors
Does not calculate Power Usage or Cost, no errors.

### Additional Notes
Also, kWh is not power, it's "Energy". Power is kW.

frodus17, 2018-07-10:
Oh, I didn't realize I had to go to outputs, sorry. I thought only Max Amps needed to be configured. I just didn't see it. 

I'll close, it's working now.

Mycodo is working great regulating my kegerator. Have it set for around 50. I need to put the sensor further from the fan, and also have another output to run the fan intermittently to keep airflow.

kizniche, 2018-07-10:
Are you using a PID? If so, what settings are you using for it?

kizniche, 2018-07-10:
Also, Max Amps will merely prevent an output from turning on if the combined power draw (determined from the sum of output amps) is greater than the Max Amps set value. I've never had an over-amp situation but I can see it being an issue if there are a lot of devices.

frodus17, 2018-07-10:
Thanks for changing to "energy. 

Yes I'm using a PID. 

Lower only
Period is 960s
Max age 120s
setpoint 53
band 1
store lower as negative
Kp gain is 50
Ki is 0
Kd is 0
Integrator min is -100
Integrator max is 100
output rise is disabled
output lower is set to my GPIO that controls the SSR
Min On duration (lower) is 180s
Max on duration (lower) is 900s
Min off duration (lower) is 120s
No methods set.

frodus17, 2018-07-13:
OK, so I caused the daemon to crash. I changed the output current to higher than the max amps, and the Daemon won't run, and I also cannot change or save either of them since the Daemon doesn't run.

Tips?

frodus17, 2018-07-13:
Tried to restore a backup. I was on 6.1.4, but it showed 6.1.3.

Tried to do a backup. Which it did, and said it was 6.1.4 in the backups, but at the top left still said 6.1.3. Rebooted and now Mycodo is completely gone.

frodus17, 2018-07-13:
Reinstalling and going to try to start from scratch.

Warm beer everywhere :)

frodus17, 2018-07-13:
For some reason, restoring a backup, stops the Daemom

frodus17, 2018-07-13:
Whenever I do a backup, it says that I'm on software version of that backup, but I'm not. It isn't matching the 6.1.4 that is installed.

frodus17, 2018-07-13:
After restore, rebooted and now get this:
Error 502: Bad Gateway Error

Something bad happened but it's probably not your fault. Try refreshing the page to see if it will load. If that doesn't work, try rebooting your system, then try refreshing it again. If that still doesn't work, submit a new issue on GitHub with any usefull information that can help determine the cause of this issue.

Something is seriously wrong. All i did was change the output current to 8A, and the max amps was 5. Ever since then, nothing is running right, restores don't work, and I've restarted from scratch.

kizniche, 2018-07-13:
Backups are located at /var/Mycodo-backups

You can manually copy one of these backups to /home/pi/mycodo then issue the following commands:

/bin/bash /home/pi/mycodo/scripts/upgrade_commands.sh initialize

/bin/bash /home/pi/mycodo/scripts/upgrade_commands.sh update-permissions

/bin/bash /home/pi/mycodo/scripts/upgrade_post.sh

kizniche, 2018-07-13:
Also, the daemon doesn't need to be running to change settings. If there is an issue with the daemon, it may slow or hinder web interface use, but once you fully kill the daemon, you can edit settings to correct the issue, then start the daemon again.

Could you look in the restore log to see what went wrong during your restore?

/var/log/mycodo/mycodorestore.log

kizniche, 2018-07-13:
Your Mycodo install was most likely not deleted, but moved to the backup location, then an error occurred.

kizniche, 2018-07-13:
It looks like the amp amount isn't the issue with starting the daemon, but rather the duration set between generating output usage reports. It appears it's not set to a time duration, but a null value.

kizniche, 2018-07-13:
...that's at least what I can deduce from the log excerpt you posted. If you have any other errors from the log that hint at another issue, please post it and I can try to figure out a fix.

frodus17, 2018-07-13:
1) I used the Backup/restore section to do a restore through the GUI.
2) When I changed something, it would tell me it was unable to write because the Daemon wasn't running.
3) I can't do anything right now, nothing runs at all.
4) Where is the backup location?

Here is mycodorestore.log:
[2018-07-12 22:13:59] 
[2018-07-12 22:13:59] #### Restore of backup /var/Mycodo-backups/Mycodo-backup-2018-06-15_20-23-57-6.1.2 initiated 2018-07-12_22-13-59 ####
[2018-07-12 22:13:59] #### Stopping Daemon and HTTP server ####
[2018-07-12 22:14:02] 
[2018-07-12 22:14:02] #### Running initialization
[2018-07-12 22:14:02] 
[2018-07-12 22:14:02] #### Creating mycodo user
[2018-07-12 22:14:02] The user `mycodo' is already a member of `adm'.
[2018-07-12 22:14:02] The user `mycodo' is already a member of `dialout'.
[2018-07-12 22:14:02] The user `mycodo' is already a member of `gpio'.
[2018-07-12 22:14:03] The user `mycodo' is already a member of `i2c'.
[2018-07-12 22:14:03] The user `mycodo' is already a member of `video'.
[2018-07-12 22:14:03] 
[2018-07-12 22:14:03] #### Compiling mycodo_wrapper
[2018-07-12 22:14:04] 
[2018-07-12 22:14:04] #### Creating symlinks to Mycodo executables
[2018-07-12 22:14:04] 
[2018-07-12 22:14:04] #### Creating files and directories
[2018-07-12 22:14:04] 
[2018-07-12 22:14:04] #### Setting permissions
[2018-07-12 22:14:07] 
[2018-07-12 22:14:07] #### Setting permissions
[2018-07-12 22:14:07] 
[2018-07-12 22:14:07] Backing up current Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2018-07-12_22-13-59-6.1.4...Done.
[2018-07-12 22:14:07] 
[2018-07-12 22:14:07] Restoring Mycodo from /var/Mycodo-backups/Mycodo-backup-2018-06-15_20-23-57-6.1.2 to /home/pi/Mycodo...Done.
[2018-07-12 22:14:07] Moving env directory...Done.
[2018-07-12 22:16:53] 
[2018-07-12 22:16:53] #### Restore of backup /var/Mycodo-backups/Mycodo-backup-2018-07-12_22-13-59-6.1.4 initiated 2018-07-12_22-16-53 ####
[2018-07-12 22:16:53] #### Stopping Daemon and HTTP server ####
[2018-07-12 22:17:00] 
[2018-07-12 22:17:00] #### Running initialization
[2018-07-12 22:17:01] 
[2018-07-12 22:17:01] #### Creating mycodo user
[2018-07-12 22:17:01] The user `mycodo' is already a member of `adm'.
[2018-07-12 22:17:01] The user `mycodo' is already a member of `dialout'.
[2018-07-12 22:17:01] The user `mycodo' is already a member of `gpio'.
[2018-07-12 22:17:02] The user `mycodo' is already a member of `i2c'.
[2018-07-12 22:17:02] The user `mycodo' is already a member of `video'.
[2018-07-12 22:17:02] 
[2018-07-12 22:17:02] #### Compiling mycodo_wrapper
[2018-07-12 22:17:03] 
[2018-07-12 22:17:03] #### Creating symlinks to Mycodo executables
[2018-07-12 22:17:03] 
[2018-07-12 22:17:03] #### Creating files and directories
[2018-07-12 22:17:03] 
[2018-07-12 22:17:03] #### Setting permissions
[2018-07-12 22:17:05] 
[2018-07-12 22:17:05] #### Setting permissions
[2018-07-12 22:17:06] 
[2018-07-12 22:17:06] Backing up current Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2018-07-12_22-16-53-6.1.2...Done.
[2018-07-12 22:17:06] 
[2018-07-12 22:17:06] Restoring Mycodo from /var/Mycodo-backups/Mycodo-backup-2018-07-12_22-13-59-6.1.4 to /home/pi/Mycodo...Done.
[2018-07-12 22:17:06] Moving env directory...Done.
[2018-07-12 22:17:17] 
[2018-07-12 22:17:17] #### Running initialization
[2018-07-12 22:17:17] 
[2018-07-12 22:17:17] #### Creating mycodo user
[2018-07-12 22:17:17] The user `mycodo' is already a member of `adm'.
[2018-07-12 22:17:17] The user `mycodo' is already a member of `dialout'.
[2018-07-12 22:17:17] The user `mycodo' is already a member of `gpio'.
[2018-07-12 22:17:17] The user `mycodo' is already a member of `i2c'.
[2018-07-12 22:17:18] The user `mycodo' is already a member of `video'.
[2018-07-12 22:17:18] 
[... 1330 lines cut ...]
[2018-07-12 23:09:24]   libpython-all-dev python-all python-all-dev python-keyring
[2018-07-12 23:09:24]   python-keyrings.alt python-secretstorage python-wheel
[2018-07-12 23:09:24] Use 'apt autoremove' to remove them.
[2018-07-12 23:09:24] 0 upgraded, 0 newly installed, 0 to remove and 252 not upgraded.
[2018-07-12 23:09:25] Installing collected packages: testfixtures, pytest, SQLAlchemy, alembic, Flask, Flask-CSV, geocoder, gunicorn, imutils
[2018-07-12 23:09:25]   Found existing installation: testfixtures 6.0.2
[2018-07-12 23:09:26]     Uninstalling testfixtures-6.0.2:
[2018-07-12 23:09:27]       Successfully uninstalled testfixtures-6.0.2
[2018-07-12 23:09:28]   Found existing installation: pytest 3.5.1
[2018-07-12 23:09:30]     Uninstalling pytest-3.5.1:
[2018-07-12 23:09:30]       Successfully uninstalled pytest-3.5.1
[2018-07-12 23:09:30] Searching for pip
[2018-07-12 23:09:30] Best match: pip 10.0.1
[2018-07-12 23:09:30] Processing pip-10.0.1-py2.7.egg
[2018-07-12 23:09:30] pip 10.0.1 is already the active version in easy-install.pth
[2018-07-12 23:09:30] Installing pip script to /usr/local/bin
[2018-07-12 23:09:30] Installing pip2.7 script to /usr/local/bin
[2018-07-12 23:09:30] Installing pip2 script to /usr/local/bin
[2018-07-12 23:09:30] 
[2018-07-12 23:09:30] Using /usr/local/lib/python2.7/dist-packages/pip-10.0.1-py2.7.egg
[2018-07-12 23:09:30] Processing dependencies for pip
[2018-07-12 23:09:30] Finished processing dependencies for pip
[2018-07-12 23:09:31]   Found existing installation: SQLAlchemy 1.2.7
[2018-07-12 23:09:33]     Uninstalling SQLAlchemy-1.2.7:
[2018-07-12 23:09:34]       Successfully uninstalled SQLAlchemy-1.2.7

frodus17, 2018-07-13:
Going to re-install from scratch, because it's having so many issues right now. Going to re-input everything. I didn't have a lot to set up. Will let you know if there are issues.

kizniche, 2018-07-13:
[here](https://github.com/kizniche/Mycodo/issues/502#issuecomment-404768668) is where the backup location is and how to manually restore a backup. You can also reinstall and just move your backed up database (You can install 6.1.3 and use the backed-up 6.1.3 database and then perform an upgrade).

kizniche, 2018-07-13:
> unable to write because the Daemon wasn't running

Do you know the exact error you were receiving? I could trace the issue with that if I had the exact quoted error (or part of it).

frodus17, 2018-07-13:
After I re-start, if I see errors, I'll screen capture it all. Alternatively, I can give you access with VNC.

frodus17, 2018-07-13:
Starting over and haven't seen any issues (except alarm setup and send test email, but testing that now).