kizniche/Mycodo #503: CCS811 Problem

pmunz75, 2018-07-12:
Specific Mycodo Version: 6.1.4
    Raspberry PI 3B+
    16 GB SD Card Class10
    System : Linux mycodo 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux
    Power 5v 2.0 Amp
    Sensor : DHT22 ( 3 meters ) , DS18B20 , Original Atlas PH i2c

I've already tried everything.
solder tested.
The css811 not Run 

sorry my bad English !
Problem Description

Please list:
ccs811 sensor always fails no i2c no data

log

2018-07-12 23:42:08,519 - mycodo.daemon - ERROR - Could not activate Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 480, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 263, in __init__
    self.measure_input = CCS811Sensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/ccs811.py", line 29, in __init__
    self.sensor = Adafruit_CCS811(address=self.i2c_address, busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_CCS811/Adafruit_CCS811.py", line 83, in __init__
    if(self._device.readU8(CCS811_HW_ID) != CCS811_HW_ID_CODE):
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_GPIO/I2C.py", line 148, in readU8
    result = self._bus.read_byte_data(self._address, register) & 0xFF
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_PureIO/smbus.py", line 169, in read_byte_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
OSError: [Errno 121] Remote I/O error

###
Activate Sensor Error ?

Success: Input controller activated in SQL database

Error: Could not activate Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b: [Errno 121] Remote I/O error

Thanks for Help

kizniche, 2018-07-13:
It appears it's unable to communicate with the sensor. Can you verify the sensor works?

Have you successfully taken a measurement with other software or other hardware (arduino)?

Does the device show up in an I2C can with ```i2c-detect```?

pmunz75, 2018-07-13:
ok , i2c found my id 0x5a , and a new error 

Error: Could not activate Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b: Device ID returned is not correct! Please check your wiring.

kizniche, 2018-07-13:
How long is your wire to the sensor? Are you supplying the proper voltage?

kizniche, 2018-07-13:
Try disconnecting everything but the one sensor and use short wires. I suspect your power supply is insufficient to run all your sensors at once.

pmunz75, 2018-07-13:
3 meters , 3 Sensors , i2c 
first = ccs811
second = BH1750
last = BME 280
all on one wire
the ccs811 no function
BH1750 OK
BME280 OK

kizniche, 2018-07-13:
We recently got the CCS811 sensor working in Mycodo (#467).

@wflost, could you confirm the CCS811 still works for you in the latest version (6.1.4)?

kizniche, 2018-07-13:
@pmunz75 Have you been able to acquire a measurement form the sensor with other code or other hardware (arduino)?

wflost, 2018-07-14:
Hi Kyle and @pmunz75 

The SparksFun CCs811 is still working correctly in 6.1.4, just checked. The issues you are encountering @pmunz75 is very similar to the issues I was encountering using a generic CCS811 sensor purchased on Ebay. I have tried two different ones from Ebay the one did not detect at all, no i2c the other was showing similar issues to your issues.

pmunz75, 2018-07-14:
Hi wflost and Kyle,
i use the Original ( Adafruit CCS811 Air Quality Sensor Breakout - VOC and eCO2 )
Link to the shop and Datasheet : https://cdn-shop.adafruit.com/product-files/3566/3566_datasheet.pdf

i2c conectet to 0x5a its OK , issues NO Data , Mycodo Daemon.log

2018-07-13 13:57:52,810 - mycodo.daemon - INFO - 41.72 MB RAM in use
2018-07-13 14:04:42,195 - mycodo.daemon - WARNING - Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b not found
2018-07-13 14:04:58,512 - mycodo.daemon - ERROR - Could not activate Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 480, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 263, in __init__
    self.measure_input = CCS811Sensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/ccs811.py", line 29, in __init__
    self.sensor = Adafruit_CCS811(address=self.i2c_address, busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_CCS811/Adafruit_CCS811.py", line 83, in __init__
    if(self._device.readU8(CCS811_HW_ID) != CCS811_HW_ID_CODE):
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_GPIO/I2C.py", line 148, in readU8
    result = self._bus.read_byte_data(self._address, register) & 0xFF
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_PureIO/smbus.py", line 169, in read_byte_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
OSError: [Errno 121] Remote I/O error
2018-07-13 14:05:01,926 - mycodo.daemon - WARNING - Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b not found
2018-07-13 14:05:10,986 - mycodo.daemon - ERROR - Could not activate Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b: Device ID returned is not correct! Please check your wiring.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 480, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 263, in __init__
    self.measure_input = CCS811Sensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/ccs811.py", line 29, in __init__
    self.sensor = Adafruit_CCS811(address=self.i2c_address, busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_CCS811/Adafruit_CCS811.py", line 84, in __init__
    raise Exception("Device ID returned is not correct! Please check your wiring.")
Exception: Device ID returned is not correct! Please check your wiring.
2018-07-13 16:34:23,530 - mycodo.daemon - WARNING - Input controller with ID 68be24d7-9d3d-4f34-b67b-7ab3d9f56a8b not found

kizniche, 2018-07-14:
Have you attempted to acquire a measurement with other software or hardware (e.g. an arduino)? If not, I would suggest testing that.

Since @wflost's sensor is working, it suggests this is an issue with your hardware (i.e. wiring, sensor, power supply, pi, etc.).

wflost, 2018-08-28:
Hi Dave 

I upgraded to 6.2.2 and everything is still working as before, there is also an issue with Adafruit_TMP it looks like the readme files have been moved. 

This may be an issue for any new installs, mine is still working fine

Thanks
Wayne

[2018-08-23 12:01:22] 
[2018-08-23 12:01:22] #### Checking for updates to optional dependencies
[2018-08-23 12:01:37] 
[2018-08-23 12:01:37] #### Installing/updating MCP342x
[2018-08-23 12:01:38] 
[2018-08-23 12:01:38] #### Installing MCP342x==0.3.3 with pip
[2018-08-23 12:02:08] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2018-08-23 12:02:15] Requirement already up-to-date: MCP342x==0.3.3 in ./env/lib/python3.5/site-packages (0.3.3)
[2018-08-23 12:02:15] Requirement already satisfied, skipping upgrade: smbus-cffi in ./env/lib/python3.5/site-packages (from MCP342x==0.3.3) (0.5.1)
[2018-08-23 12:02:15] Requirement already satisfied, skipping upgrade: cffi>=1.1.0 in ./env/lib/python3.5/site-packages (from smbus-cffi->MCP342x==0.3.3) (1.11.5)
[2018-08-23 12:02:16] Requirement already satisfied, skipping upgrade: pycparser in ./env/lib/python3.5/site-packages (from cffi>=1.1.0->smbus-cffi->MCP342x==0.3.3) (2.18)
[2018-08-23 12:02:48] 
[2018-08-23 12:02:48] #### Installing/updating tsl2561
[2018-08-23 12:02:49] 
[2018-08-23 12:02:49] #### Installing tsl2561 with pip
[2018-08-23 12:03:20] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2018-08-23 12:03:28] Requirement already up-to-date: tsl2561 in ./env/lib/python3.5/site-packages (3.3)
[2018-08-23 12:04:04] 
[2018-08-23 12:04:04] #### No install candidate for RPi.GPIO
[2018-08-23 12:04:04] 
[2018-08-23 12:04:04] #### Installing/updating miflora
[2018-08-23 12:04:05] 
[2018-08-23 12:04:05] #### Installing miflora==0.4 with pip
[2018-08-23 12:04:35] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2018-08-23 12:04:41] Requirement already up-to-date: miflora==0.4 in ./env/lib/python3.5/site-packages (0.4)
[2018-08-23 12:04:41] Requirement already satisfied, skipping upgrade: btlewrap==0.0.2 in ./env/lib/python3.5/site-packages (from miflora==0.4) (0.0.2)
[2018-08-23 12:05:17] 
[2018-08-23 12:05:17] #### Installing/updating bluepy
[2018-08-23 12:05:18] Reading package lists...
[2018-08-23 12:05:21] Building dependency tree...
[2018-08-23 12:05:21] Reading state information...
[2018-08-23 12:05:30] libglib2.0-dev is already the newest version (2.50.3-2).
[2018-08-23 12:05:30] 0 upgraded, 0 newly installed, 0 to remove and 51 not upgraded.
[2018-08-23 12:05:30] 
[2018-08-23 12:05:30] #### Installing bluepy==1.1.4 with pip
[2018-08-23 12:06:05] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2018-08-23 12:06:11] Requirement already up-to-date: bluepy==1.1.4 in ./env/lib/python3.5/site-packages (1.1.4)
[2018-08-23 12:06:46] 
[2018-08-23 12:06:46] #### Installing/updating btlewrap
[2018-08-23 12:06:46] 
[2018-08-23 12:06:46] #### Installing btlewrap==0.0.2 with pip
[2018-08-23 12:07:17] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2018-08-23 12:07:23] Requirement already up-to-date: btlewrap==0.0.2 in ./env/lib/python3.5/site-packages (0.0.2)
[2018-08-23 12:08:01] 
[2018-08-23 12:08:01] #### Installing/updating Adafruit_TMP
[2018-08-23 12:08:01] 
[2018-08-23 12:08:01] #### Installing Adafruit_TMP with pip
[2018-08-23 12:08:32] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2018-08-23 12:08:38] Collecting Adafruit_TMP
[2018-08-23 12:08:39]   Using cached https://files.pythonhosted.org/packages/4d/b1/6921987a5b03b615d8f662ec1bbb5dc15f194d000e3b4a84c0e9dac7a770/Adafruit_TMP-1.6.2.tar.gz
[2018-08-23 12:08:58]     Complete output from command python setup.py egg_info:
[2018-08-23 12:08:58]     Traceback (most recent call last):
[2018-08-23 12:08:58]       File "<string>", line 1, in <module>
[2018-08-23 12:08:58]       File "/tmp/pip-install-mzaqr5rj/Adafruit-TMP/setup.py", line 15, in <module>
[2018-08-23 12:08:58]         with open(path.join(here, 'README.md'), encoding='utf-8') as f:
[2018-08-23 12:08:58]       File "/home/pi/Mycodo/env/lib/python3.5/codecs.py", line 895, in open
[2018-08-23 12:08:58]         file = builtins.open(filename, mode, buffering)
[2018-08-23 12:08:58]     FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-mzaqr5rj/Adafruit-TMP/README.md'
[2018-08-23 12:08:58]     
[2018-08-23 12:08:58]     ----------------------------------------
[2018-08-23 12:09:02] 
[... 214 lines cut ...]
[2018-08-23 12:29:41] #### Updating Mycodo crontab entry
[2018-08-23 12:29:41] Crontab entry does not exist, nothing to do.
[2018-08-23 12:29:41] 
[2018-08-23 12:29:41] #### Updating Mycodo restart monitor crontab entry
[2018-08-23 12:29:42] Removing crontab entry ...
[2018-08-23 12:29:42] Appending to crontab:
[2018-08-23 12:29:42] -----------------------------------------------
[2018-08-23 12:29:42] @reboot /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
[2018-08-23 12:29:42] -----------------------------------------------
[2018-08-23 12:29:42] To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove
[2018-08-23 12:29:42] 
[2018-08-23 12:29:43] 
[2018-08-23 12:29:43] #### Setting permissions
[2018-08-23 12:29:51] 
[2018-08-23 12:29:51] #### Restarting the Mycodo daemon
[2018-08-23 12:30:05] Mycodo previously shut down properly
[2018-08-23 12:31:08] 
[2018-08-23 12:31:08] #### Restarting nginx
[2018-08-23 12:31:18] #### Reloading mycodoflask
[2018-08-23 12:31:20] 
[2018-08-23 12:31:20] #### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
[2018-08-23 12:33:38] #### Successfully connected to http://localhost
[2018-08-23 12:33:38] Done.
[2018-08-23 12:33:38] 
[2018-08-23 12:33:38] Upgrade completed successfully without errors.

Theoi-Meteoroi, 2018-08-28:
Yup - looks like Adafruit is fiddling with their libraries and implementing policy. The TMP library is also deprecated and archived.

I tried installing the newest version manually but it looks like the only changes are in making sure setup.py is broken.  

Going to try to manually install these two sensors using the previous github versions.

Theoi-Meteoroi, 2018-08-28:
Well. Lovely.  If you search for "archived" Adafruit repositories you find that most all the python sensor libs I've been using have been archived, with the last act seeming to break installation.  The setup.py in each was changed to make it broken then set in concrete (archived).  That would seem to be the final finger.

I'll actively avoid Adafruit hardware in the future.  And no, I'm not going to use circuitpython either. Seems google wasn't the only one to forget how not to be evil.

kizniche, 2018-08-28:
That is unfortunate. Luckily I forked all of these repositories a while ago for just this type of occasion. I'll see what I can do to fix this, but if there are any recommendations, let me know. Should I just add the working library fil(s) to Mycodo?

kizniche, 2018-08-28:
I could just specify version numbers of the last working version for pip to sort out, like I've done for other pip packages (https://github.com/kizniche/Mycodo/blob/master/mycodo/scripts/dependencies.sh).

Theoi-Meteoroi, 2018-09-02:
Since I've spent a non-trivial amount with Adafruit, I complained about the final state of the libraries.  Adafruit did respond, took action with the repositories and now says these old libs are fixed:

[https://forums.adafruit.com/viewtopic.php?f=19&t=140256](url)

kizniche, 2018-09-03:
I appreciate you taking the initiative with contacting Adafruit and being thorough with diagnosing the issues with the packages.

closed 2018-10-13 by kizniche