kizniche/Mycodo #504: Daemon stops running

frodus17, 2018-07-16:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.4
 
#### Problem Description
Daemon stops running. It ran for a couple days, then stopped after a restart. Seems to have something to do with usage report. I have my only output set at 8A. Max amps, under configuration, is 15A. I have it run usage once a month, this is all that I've configured differently in the last week.
 
### Errors
This is similar to the issues I had earlier this week. This is a clean install, on a new flash disk. I set everything up again. Restarted and the Daemon shows not running. 

### Steps to Reproduce the issue:
How can this issue be reproduced?
It seems that 6.1.4 is unstable with regards to output usage statistics. Earlier versions had no issues and ran for over a month without a problem. Once I set 0A on my output, and disable running usage reports, I restart the daemon and it seems to run.
 
### Additional Notes
I see this under Daemon logs:
2018-07-15 21:51:58,245 - mycodo.daemon - INFO - All Conditional controllers stopped
2018-07-15 21:51:58,245 - mycodo.daemon - INFO - All LCD controllers stopped
2018-07-15 21:51:58,253 - mycodo.input_df812159 - INFO - Deactivated in 8.6 ms
2018-07-15 21:51:58,263 - mycodo.input_1b9b8ca5 - INFO - Deactivated in 17.9 ms
2018-07-15 21:51:58,274 - mycodo.input_874ae506 - INFO - Deactivated in 28.6 ms
2018-07-15 21:52:05,093 - mycodo.pid_12b78e75 - WARNING - No data returned from influxdb
2018-07-15 21:52:05,431 - mycodo.pid_12b78e75 - INFO - Deactivated in 7186.2 ms
2018-07-15 21:52:05,432 - mycodo.daemon - INFO - All PID controllers stopped
2018-07-15 21:52:05,433 - mycodo.daemon - INFO - All Math controllers stopped
2018-07-15 21:52:05,434 - mycodo.daemon - INFO - All Input controllers stopped
2018-07-15 21:52:05,495 - mycodo.daemon - ERROR - ERROR Starting Mycodo Daemon
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 1080, in start_daemon
    self.mycodo.run()
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 399, in run
    now > self.output_usage_report_next_gen):
TypeError: unorderable types: float() > NoneType()

kizniche, 2018-07-16:
The latest commit should fix this issue. I'll do some more testing later and will try to get a release out later this week.

frodus17, 2018-07-16:
Thanks Kyle. What was it, for my knowledge?

kizniche, 2018-07-16:
It should be. I need to perform some testing with different combinations of settings, but that at least will prevent the error you've experienced. Whether it affects anything else is yet to be seen.

closed 2018-10-13 by kizniche

byhlk, 2023-04-01:
**sudo service influxdb start** 

```
influxdb.service - InfluxDB is an open-source, distributed, time series database
     Loaded: loaded (/lib/systemd/system/influxdb.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: timeout) since Sat 2023-04-01 19:56:10 BST; 54ms ago
       Docs: https://docs.influxdata.com/influxdb/
    Process: 6870 ExecStart=/usr/lib/influxdb/scripts/influxd-systemd-start.sh (code=killed, signal=TERM)
        CPU: 1min 16.434s
```

**sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py**

```
2023-04-01 19:53:49,021 Mycodo daemon v8.15.7 starting
2023-04-01 19:53:49,334 Starting Pyro5 daemon
Object <__main__.PyroServer object at 0xb4612718>:
    uri = PYRO:mycodo.pyro_server@0.0.0.0:9080
Pyro daemon running.
2023-04-01 19:53:57,930 Activated in 4511.5 ms
2023-04-01 19:53:58,432 All activated Conditional controllers started
2023-04-01 19:53:58,435 All activated Trigger controllers started
2023-04-01 19:53:58,445 All activated Input controllers started
2023-04-01 19:53:58,460 All activated PID controllers started
2023-04-01 19:53:58,464 All activated Function controllers started
2023-04-01 19:54:00,078 Activated in 1608.4 ms
2023-04-01 19:54:01,082 Mycodo daemon started in 12.057 seconds
2023-04-01 19:54:03,607 47.42 MB RAM in use
```