kizniche/Mycodo #505: Show me your projects! (for an undergraduate/graduate seminar)

kizniche, 2018-07-16:
Hello all. I'm currently working on a presentation for a seminar for university undergraduate and graduate students, as well as interested faculty, on using feedback loops for biological science (but not limited to that). 

Over the years I've heard of many of you using Mycodo in some great projects. I'd like to ask if anyone is interested in sharing their work, to potentially be included in my presentation, to please [contact me (http://kylegabriel.com/contact)](http://kylegabriel.com/contact). If you willing, please contact me with the above form and I will reply back with an email address (unless you already have it) you can send photos and a brief description. Let me know if you would like to be credited for the photo and how to credit you (e.g. "Photo courtesy of John Doe").

Thanks!

Kyle

frodus17, 2018-07-16:
Sent!

frodus17, 2018-07-30:
Should I have gotten an email?

kizniche, 2018-07-30:
@frodus17, Sorry, I got your email and will be including your project in my presentation. There's a month until the seminar, so I won't send out a copy of the presentation for you to see until it's closer to that time, so I have a chance to finalize the presentation.

Gossen1, 2018-09-11:
Are you still looking for pictures and a little description of projects?

frodus17, 2018-09-12:
Do you have information on the presentation? I'd be interested in following it. Your work is pretty interesting.

kizniche, 2018-09-12:
Just sent you an email with a link to download it.

closed 2018-09-17 by kizniche