kizniche/Mycodo #506: Version 7.0 Changes

kizniche, 2018-07-19:
Changes with version 7.0 that will break current functionality (upgrade will fail, need to recreate database).

1. Full unit conversion system (select what unit to use for each sensor measurement)

kizniche, 2018-07-19:
I'm working on a way to preserve the 6.x database and still accomplish 1. I'll be adding to this list as I think of feature additions and updating my progress with the unit/conversion system.

kizniche, 2018-07-21:
I figured out how to modify database entries to rename to the new naming convention used in the new measurement/unit system. I'm now going through the system making changes and testing stability.

kizniche, 2018-07-21:
Additionally, what I meant to mention in the last comment was that as a result, a database deletion will not be necessary when upgrading to the next release, and it will nto be a major version jump (7.x), but rather only to 6.2,

kizniche, 2018-07-21:
Great progress so far. Here's how the graph I've been testing looks after fixing all the bugs (so far).
[image]

kizniche, 2018-07-22:
Finally ironed out most of the bugs for the dashboard, live, and async graph pages.
[image]

kizniche, 2018-07-22:
With 6.2.0, I'll also introduce the ability to add new measurements, units, and unit conversions. For an example if how this would be used, let's say, as a user, I had weight sensors. Since no weight sensors are currently supported, I would need to create a script to read the weight measurement and configure a Command Input in Mycodo to execute that script and store the measurement. To make a new measurement and unit available to be selected in the command Input, I would then need to add the "weight" measurement, "gram" and "kilogram" units, and the conversions "g_to_kg" = "x/1000", "kg_to_g" = "x*1000". Once this is done, the measurement and units are available via dropdown in the Command Input. This can be done for any new type of measurement and/or unit that's not currently built-in to Mycodo.

kizniche, 2018-07-23:
The new measurement configuration page.

[image]

kizniche, 2018-07-23:
It's nearly all come together.

[image]

kizniche, 2018-07-24:
I think I have the custom measurement system in its final stages. I've ironed out most of the bugs in the system and refined the Measurement Settings page. Now I need to test a bunch of (pre-6.2.0) database configurations through the [database upgrade script](https://github.com/kizniche/Mycodo/blob/648369ea16d00d110c03a95f018383438e4d3bb6/databases/alembic/versions/7dbc5357d3a9_update_conversion_entries.py) and hope everything gets modified correctly. This is the most complex database data upgrade I've performed in Mycodo yet. I hope everyone appreciates not having to delete your database and start from scratch :wink: 

[image]

kizniche, 2018-07-26:
Selecting a measurement/unit is now foolproof because of the new measurement system. two text inputs for measurement and unit are replaced by a dropdown to select the measurement and unit, for Linux Command and ADC Inputs, as well as Math controllers.

[image]

closed 2018-07-30 by kizniche

Theoi-Meteoroi, 2018-07-30:
This is *awesome*

Thank you Kyle for making this change... and also all the database work you have done recently for upgrades.  I've been distracted by other projects lately but these changes are compelling enough to squeeze in some Mycodo time to kick the tires a bit and stand up a controller or two with the new code.

Nice work!

kizniche, 2018-07-30:
Thanks! I'm also experimenting with PID autotuning, so I may be releasing that soon, depending how the testing goes.

commit f5d6069a (2018-07-30): Develop database upgrade script to alleviate having ot delete database (closes #506)