kizniche/Mycodo #507: IP Address on LCD Display ?

pmunz75, 2018-07-28:
Hello, is it possible to display the current IP address on the LCD display?
with Linux Command?
Could you please help me ?
thx Philipp

kizniche, 2018-07-28:
I can't think of a way to do it with the existing features, but that's a good idea. I'll see about adding that this weekend (though the release may take another week). Thanks for the question and good suggestion.

pmunz75, 2018-07-28:
Thank you .!
Perfect would be a startup display, welcome and ip address

we have completely translated mycodo into German.
Do you want to have the translation?

kizniche, 2018-07-28:
That's a good idea. I'll add a second display after the mycodo version. If it can detect an IP address, it will display it after the version. I'll also add an "IP address" option to the line displays as well.

All words/phrases have been translated? wow, that's quite an undertaking! Yes, please send the *.po file and I'll add it to the next release. Thank you!

pmunz75, 2018-07-28:
yes we translated everything, completely
I'll send it to you in the evening.

kizniche, 2018-08-20:
I was just able to test the IP display on my LCD (I have been developing on a system without an LCD). I found that it is working for me, however it is displaying the IP as a byte, and I need to convert it to a string. I will make the fix and it will be available in the next release.

kizniche, 2018-08-22:
The new LCD option to show the IP on the LCD should be working. Feel free to reopen this issue if you're experiencing issues and I'll investigate.

closed 2018-08-22 by kizniche