kizniche/Mycodo #509: problem UART pH data

leoneo, 2018-08-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.4
 
#### Problem Description
Please list: no pH data with atlas pH UART

 2018-08-01 18:07:09,465 - mycodo.daemon - ERROR - Could not activate Input controller with ID a2342661-5493-44c0-ba01-49edb9824690: invalid literal for int() with base 16: 'Tx/Rx'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 480, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 242, in __init__
    self.measure_input = AtlaspHSensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/atlas_ph.py", line 27, in __init__
    self.i2c_address = int(str(input_dev.location), 16)
ValueError: invalid literal for int() with base 16: 'Tx/Rx'

kizniche, 2018-08-01:
That last commit should fix this issue. You can manually edit the file and make those changes, restart the daemon, and see if it works, otherwise I should have the next release out sometime in the next week or so.

leoneo, 2018-08-03:
I have do the change but there is always no pH data. I will check with atlas code.

kizniche, 2018-08-03:
Please copy the error in the daemon log after making the changes I performed in the commit, as that should have fixed the error you originally reported. If there's another error that appears I need to see what it is in order to properly diagnose the new issue and devise a fix.

leoneo, 2018-08-03:
I think that a raspbian upgrade had change the file /boot/cmdline.txt 
I have do the correction in this file and now I have pH data.

Thanks

kizniche, 2018-08-03:
Great

closed 2018-08-03 by kizniche