kizniche/Mycodo #511: enhancement Add Liquid Flow Meter

leoneo, 2018-08-01:
Hi !
I have this Liquid Flow Meter https://www.adafruit.com/product/828 

Can you please add this sensor ? With reset option ?
Thank you so much 
Léo

kizniche, 2018-08-01:
There is already an RPM Input that should be able to work if you connect the device properly. From this you can use an Equation Math controller to convert the RPM to a flow rate. Have you done anything in the way of attempting to connect it to your Pi?

kizniche, 2018-08-01:
The Pi pins require 3.3 volts, so you will have to drop the voltage output from the sensor (5-24VDC in). I would think a voltage divider would suffice for this. Make sure you don't exceed the max current for the Pi's pin.

leoneo, 2018-08-03:
Thanks for your good remark ! I have make a voltage divider with two resistors.

Now i will try with RPM input and equation math

kizniche, 2018-08-20:
Have you had any luck with what you mentioned you were going to attempt?

kizniche, 2018-11-30:
Any update on testing this?

closed 2019-10-07 by kizniche