kizniche/Mycodo #513: Dash page crashes with Tracelog

mrplow, 2018-08-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.2.0
 
#### Problem Description
  - trying to load the dashboard results in an error 500 with a full traceback as below.
 
### Errors
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 489, in page_dashboard
    y_axes=y_axes)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 135, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 117, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["dashboard", _('Dashboard')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 255, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 788, in block "body"
    {% include 'pages/dashboard_options/display_graph.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard_options/display_graph.html", line 246, in top-level template code
    name: '{{each_math.name}} ({{measurement_units[each_measurement]['name']}}
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'Pressure'

kizniche, 2018-08-17:
Have you checked your upgrade log for errors?

mrplow, 2018-08-17:
Good idea 😋
```
[2018-08-17 01:20:05] #### Installing/updating Adafruit_TMP
[2018-08-17 01:20:05]
[2018-08-17 01:20:05] #### Installing Adafruit_TMP with pip
[2018-08-17 01:20:08] Collecting Adafruit_TMP
[2018-08-17 01:20:08]   Downloading https://files.pythonhosted.org/packages/4d/b1/6921987a5b03b615d8f662ec1bbb5dc15f194d000e3b4a84c0e9dac7a770/Adafruit_TMP-1.6.2.tar.gz
[2018-08-17 01:20:09]     Complete output from command python setup.py egg_info:
[2018-08-17 01:20:09]     Traceback (most recent call last):
[2018-08-17 01:20:09]       File "<string>", line 1, in <module>
[2018-08-17 01:20:09]       File "/tmp/pip-install-1p6pj3dv/Adafruit-TMP/setup.py", line 15, in <module>
[2018-08-17 01:20:09]         with open(path.join(here, 'README.md'), encoding='utf-8') as f:
[2018-08-17 01:20:09]       File "/home/pi/Mycodo/env/lib/python3.5/codecs.py", line 895, in open
[2018-08-17 01:20:09]         file = builtins.open(filename, mode, buffering)
[2018-08-17 01:20:09]     FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-1p6pj3dv/Adafruit-TMP/README.md'
[2018-08-17 01:20:09]
[2018-08-17 01:20:09]     ----------------------------------------
[2018-08-17 01:20:10]
[2018-08-17 01:20:10] #### Installing/updating Adafruit_BME280
```

kizniche, 2018-08-17:
Is that all? No errors with the database upgrade? Can you go to the System Information page and check if the database version is colored green or red?

Also, you can restore a previous version if you really need to get your system back operational right away.

kizniche, 2018-08-17:
I'm working on a fix you can try to regain access to the graph page to resave your graph settings (which will fix the graph itself).

kizniche, 2018-08-17:
Okay, I think I found what could be the issue. It doens't require any code fixes. Go to your Math Controllers, disable them all. Then review the settings and make sure the measurement and unit are properly selected and saved. I suspect you have a misconfigured math controller.

mrplow, 2018-08-17:
You nailed it.
It was my kPa math conditional. The Unit Measurement was unset, I guess due to the upgrade.
After disabling, setting the Unit, resaving and finally reactivating the dashboard loads properly.
Thanks!

kizniche, 2018-08-17:
I tried to account for all potential combinations for the database upgrade, but I guess I missed something. Glad it was easily fixable, though.

closed 2018-08-17 by kizniche