kizniche/Mycodo #515: Dashbord does not show

wflost, 2018-08-20:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.2.0
 
#### Problem Description
Please list: I was selecting the dashbord menu item, the live item works correctly. This was all working under 6.1.4,

 
### Errors

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 489, in page_dashboard
    y_axes=y_axes)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 135, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 117, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["dashboard", _('Dashboard')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 255, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 788, in block "body"
    {% include 'pages/dashboard_options/display_graph.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard_options/display_graph.html", line 246, in top-level template code
    name: '{{each_math.name}} ({{measurement_units[each_measurement]['name']}}
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'CO2'

  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Select Dashboard, the web errors report a web timeout
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

This is has only happened since the upgrade, I have restarted the system

kizniche, 2018-08-20:
Can you check that all your math controllers have a measurement unit set?

kizniche, 2018-08-20:
I just released v6.2.1, which adds a new configuration menu that has the option to delete all dashboard elements. I added this because it appears too difficult to diagnose individual issues that may have arisen from the 6.2.0 upgrade, causing a misconfiguration of any of a number of dashboard elements. By deleting all dashboard elements, this will allow you to access the Dashboard page again and rebuild your Dashboard. I'm sorry to have to do this, but I feel it is the best way to approach this without investing a lot of effort to find issues that may be resolved by rebuilding the dashboard. Please let me know if there are any issues with the new feature and if you experience any new issues related to adding new dashboard elements. If there are new issues related to creating new dashboard elements, I will do my best to address them, as I believe addressing these new issues will be more productive than the many potential issues that could have resulted from the 6.2.0 upgrade.

wflost, 2018-08-21:
Hi Kyle,

Thanks it works clearing the Dashboard elements and changing the Math units

Please close the case

closed 2018-08-21 by wflost