kizniche/Mycodo #521: Enhancement - New sensot SI7021 Temp humidoty

wflost, 2018-09-03:
Hi Kyle,

I have ordered a new all in one, it has a ccs811 which is working correctly, I have the si7021 using the sht2x so the calibration may be slightly off the use the same I2C -40

Are you able to add the SI7021 to the sensor list?

Thanks
Wayne

## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

Theoi-Meteoroi, 2018-09-03:
First try with the [pi_si7021](https://github.com/kangasta/pi_si7021) did not go well, I think. Fails on a register write.

pi@Control-1:~/Downloads/pi_si7021-master/pi_si7021 $ python3 si7021.py 
Traceback (most recent call last):
  File "si7021.py", line 82, in <module>
    print("Temperature: " + str(round(RHTEMP.temperature, 2)) + u" \u00B0C")
  File "si7021.py", line 36, in temperature
    self.pi.i2c_write_device(self.i2c, self.__cmds["meas_temp_nohold"])
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 3054, in i2c_write_device
    self.sl, _PI_CMD_I2CWD, handle, 0, len(data), [data]))
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 970, in _u2i
    raise error(error_text(v))
pigpio.error: 'I2C write failed'

Now this is a module that I'm not quite fuzzy about since the CCS811 sensor fails due to:
2018-09-03 16:23:59,657 - mycodo.daemon - ERROR - Could not activate Input controller with ID 6927e0de-1b8b-45a7-a744-4f280cceb196: Device returned an Error! Try removing and reapplying power to the device and running the code again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 481, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 261, in __init__
    self.measure_input = CCS811Sensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/ccs811.py", line 29, in __init__
    self.sensor = Adafruit_CCS811(address=self.i2c_address, busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_CCS811/Adafruit_CCS811.py", line 92, in __init__
    raise Exception("Device returned an Error! Try removing and reapplying power to the device and running the code again.")
Exception: Device returned an Error! Try removing and reapplying power to the device and running the code again.
2018-09-03 16:24:20,584 - mycodo.daemon - WARNING - Input controller with ID 6927e0de-1b8b-45a7-a744-4f280cceb196 not found
2018-09-03 16:24:30,153 - mycodo.daemon - ERROR - Could not activate Input controller with ID 6927e0de-1b8b-45a7-a744-4f280cceb196: Device ID returned is not correct! Please check your wiring.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 481, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 261, in __init__
    self.measure_input = CCS811Sensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/ccs811.py", line 29, in __init__
    self.sensor = Adafruit_CCS811(address=self.i2c_address, busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/Adafruit_CCS811/Adafruit_CCS811.py", line 84, in __init__
    raise Exception("Device ID returned is not correct! Please check your wiring.")
**Exception: Device ID returned is not correct!** Please check your wiring.
2018-09-03 16:24:41,890 - mycodo.daemon - WARNING - Input controller with ID 6927e0de-1b8b-45a7-a744-4f280cceb196 not found

So I know the CCS811 is a little techy and really a SMbus device.   Folks using microcontrollers have to stretch a clock to get the correct behavior. So maybe its my board.
 I'll try again with a separate si7021 breakout, just to be sure it isn't a faulty combo-breakout.

kizniche, 2018-09-04:
Does there happen to be a reset pin that's floating?

Theoi-Meteoroi, 2018-09-04:
Gaaa.  The DFrobot breakout doesn't show up in i2cdetect. So I go looking for a schematic...

[Product Page](https://www.dfrobot.com/product-1181.html)

[TOY0054_Si7021_SCH.PDF]
(https://github.com/kizniche/Mycodo/files/2346731/TOY0054_Si7021_SCH.PDF)

I see. If that board is built to the schematic, Both SDA and SCL are shorted to ground.  Checking with a meter is unrevealing.  I've never used these breakouts and they were very cheap.  Perhaps there is a soldering botch under the package.

Theoi-Meteoroi, 2018-09-05:
I found that the device on the combo board is actually a HDC1080 responding at 0x40. And that device works. But this means I have no functional si7021 sensors. The DFrobot boards I tried don't show up.  The CCS811 on the combo breakout may have downrev app code or something. I haven't been able to find any device history for CCS811 hwID and it seems only one hwID works.   
The other two devices work fine.

I'll order some si7021 (they are cheap) along with some SHT21/AM2320B sensors to have examples. But it will probably be a few weeks to receive the parts and get back on this.

The upshot is these other drivers may work just fine, I need working sensor hardware to run against to test.

kizniche, 2019-10-07:
@wflost I've been going through and closing issues today, and came across this one. I know this is a bit old, but if you have any Python 3 code you can confirm works with the sensor, I'll try to create an Input module for you to test.

wflost, 2019-10-07:
Hi Kyle,

Thank you for following up, I don't have any code, please close the issue

kizniche, 2019-10-07:
Can you test this code?

Save the following to a file, ```~/test_sensor.py```:

```python
from si7021 import Si7021
import time
from smbus2 import SMBus
sensor = Si7021(SMBus(1))
print(sensor.read())
```

Then run the commands:

```~/Mycodo/env/bin/pip install Si7021```

```~/Mycodo/env/bin/python ~/test_sensor.py```

closed 2020-07-23 by kizniche