kizniche/Mycodo #527: Adding notes to Mycodo

Gossen1, 2018-09-11:
Just a thought that keeps popping up. I would love the ability to add notes to Mycodo. For example: "On the 12th of may at 13:00 I added an extra fan to room 1." So I can see the effect more easily in the graph. My 2 pennies. Thanks!

kizniche, 2018-09-11:
This was added in #28, but had to be removed when the whole system was refactored. I'll look into adding this feature again.

frodus17, 2018-09-11:
This is very useful (looked at #28). Right now it's in my Kegerator, but soon I'll be gutting my other temp chamber to run Mycodo. It'll be my Fermentation chamber, so adding notes to it would be super useful.

kizniche, 2018-09-12:
Tags and Notes can now be added, and notes can be searched by tag and contents of the note. Next is to integrate the notes into graphs.

[image]

kizniche, 2018-09-12:
More work on the filtering/sorting of notes:

[image]

I'll tackle making them appear in graphs next. I also still need to implement file attachments, but that should be relatively easy.

kizniche, 2018-09-13:
After much toiling, I finally got note flags to appear on graphs!

[image]

kizniche, 2018-09-13:
Who ever said this was going to be easy? :rofl: 

[image]

kizniche, 2018-09-13:
The notes feature is nearly complete. Here's the current state:

[image]

[image]

kizniche, 2018-09-14:
With that last commit, I think I've accomplished everything on my feature list for a basic note system. I'll review and test all the new features (single-file input module system and the new note system), and try to get a new release out this weekend.

kizniche, 2018-09-14:
[image]

closed 2018-09-17 by kizniche