kizniche/Mycodo #528: Feature

frodus17, 2018-09-11:
I was wondering if there was a way to allow the user, through a browser to save the remote configuration of the MyCodo system? 

That way I can easily back up locally through the browser and save on my windows machine. I've had a couple times where the flash disk I've had either gets corrupt, or something happens to the install, so I have to start over.

kizniche, 2018-09-11:
From the More -> Export Import page, you can save/import a mycodo settings configuration as well as the InfluxDB Database and Metastore (actual measurements). Currently there is no way to export/import a whole-Mycodo backup (from the Config -> Backup Restore page), but I should add that.

frodus17, 2018-09-11:
OK, I can do that for now. I like the one-stop backup idea.

We do something similar with our data logger for Modbus devices. Allows cloning of devices.

kizniche, 2018-09-12:
To accomplish what you're trying to do, you'll need to export the settings database (and influxdb database/metastore if you want to bring over measurements). Then, on the new Pi, download the Mycodo release that corresponds to the exported settings database version, install that release, then import the settings database (also import influxdb database/metastore if desired). Then perform an upgrade.

There are potential issues with merely copying a folder/zip backup to a new system and not initializing it properly (install dependencies, make folders/symlinks, permissions, etc.). The method I describe above is the safe way that I've tested to work.

Perhaps I'm not understanding exactly what you want to do.

frodus17, 2018-09-12:
I think mainly I'm interested in settings, not logged data, so I can start anew if something goes wrong. It recently happened to me where the SD card was corrupted and no longer works. I had to start from scratch, and in a rush.

Just not sure how to install a specific version, if there's something that goes wrong and I need to restore. I only know how to install the current version per the instructions on the main Mycodo page.

kizniche, 2018-09-12:
I'll update the install script so a version number can be specified.

kizniche, 2018-09-12:
To install a specific version, just search through the releases on the main github repo page, download, extract, and run the install/setup.sh script.

kizniche, 2018-09-12:
I'll implement soon the ability to import settings databases that have a version number lower than the installed Mycodo system. Currently only a database with the same version number as the running Mycodo system can be imported. This should be as easy as adding a database upgrade step in the import process.

frodus17, 2018-09-12:
Kyle, that is great! I didn't see the backup page before, I think I was on an older install version (haven't updated in a few weeks or so). I have backed up and it's safe and sound for now.

closed 2019-02-17 by kizniche