kizniche/Mycodo #529: Errors after upgrade from 6.2.x to 6.3.3

frodus17, 2018-09-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.3.3
 
#### Problem Description
Just hit upgrade, then this error on the screen:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 622, in page_dashboard
    y_axes=y_axes)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 135, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 117, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["dashboard", _('Dashboard')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 256, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 828, in block "body"
    {% include 'pages/dashboard_options/display_graph.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard_options/display_graph.html", line 4, in top-level template code
    {% set graph_note_tag_ids = each_graph.note_tag_ids.split(';') %}
jinja2.exceptions.UndefinedError: 'None' has no attribute 'split'

kizniche, 2018-09-17:
Let me push a quick update to fix this. Stay tuned. You should still be able to access the upgrade page.

kizniche, 2018-09-17:
Just made a new release that should fix the issue

frodus17, 2018-09-17:
What is the URL for upgrade? like /upgrades.php?

kizniche, 2018-09-17:
Probably /settings/upgrade but not 100% sure. Can also try /live or /data

frodus17, 2018-09-17:
found it [IP address]/admin/upgrade

updating now.

frodus17, 2018-09-17:
Upgraded and it seems to have gone away. Control loop is running and I can access all pages. Thanks Kyle!

kizniche, 2018-09-17:
Can you describe what you were doing immediately before the error appeared? Add input, save options, etc.?

frodus17, 2018-09-17:
I upgraded to 6.3.4 from 6.3.3 (to fix the first error you just fixed), and was testing to see if any other pages had another 500 error when I found this one.

That's about it. I hadn't changed or adjusted anything. It was 5 minutes after the upgrade. I rebooted as well.

kizniche, 2018-09-17:
Okay. Looking for the issue. It shouldn't be affecting the daemon, but I'll try to find a fix post-haste.

kizniche, 2018-09-17:
Can you list all the inputs you have added?

frodus17, 2018-09-17:
This one doesn't seem to affect the daemon and it's still running just fine, so no quick-fix needed.

Just 1 input, the DS18B20 sensor I've got to measure fridge temp.

kizniche, 2018-09-17:
Released 6.3.5

frodus17, 2018-09-17:
Thanks! that was fast.
How do I update manually? I don't see it available on the upgrades page.

[image]

kizniche, 2018-09-17:
Strange error. Github may have temporarily blocked your IP for too many requests in too short of a period. In any case, the manual upgrade will also try to contact the github server to determine the latest version number to procure the correct archive. You can upgrade to the master branch with:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade-master
```

frodus17, 2018-09-17:
updating now, thanks. I'll save that command. Appreciate it.

Upgraded and that page now working and system is running. Thanks Kyle.

kizniche, 2018-09-17:
Thanks for being the first system in the wild to upgrade and find the issues.

frodus17, 2018-09-17:
Ha, you're welcome. I didn't upgrade cause I needed the features, I pretty much do it to help the project find problems as best I can. Keep it up Kyle! It's an awesome project.