kizniche/Mycodo #531: Adafruit_GPIO Dependency won't install

mrplow, 2018-09-18:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.3.8
 
#### Problem Description
Please list:

  - I noticed a failed upgrade from 6.2.4 to 6.3.3 this afternoon from an upgrade I started then left for work (non-critical environment, just testing). I restored the auto-created backup with `sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh backup-restore /var/Mycodo-backups/Mycodo-backup-2018-07-17-SomeTimeThisMorning`
  - I upgraded from 6.2.4 to 6.3.8, this went well with no errors. Afterwards I noticed my BME280 wasn't working, the daemon logs indicated the Adafruit_BME280 and Adafruit_GPIO python modules were missing.
  - I went to the dependency page and selected Adafruit_BME280 and installed them. Afterwards the Adafruit_GPIO was still missing but Adafruit_BME280 wasn't. I tried installing the Adafruit_GPIO module a couple times, rebooted and tried again but it never successfully installed via the gui.
  - I installed the module manually via ssh with `~/Mycodo/env/bin/pip install Adafruit_GPIO` **This solved my problem.**
 
### Errors
  - Upgrade from 6.2.4 to 6.3.3: `Failed: Error while updating database with alembic.`
  - dependency log, Adafruit_GPIO claims to install but nothing from git is pulled: 
```
[2018-09-17 20:12:29] Dependency installation beginning. Installing: Adafruit_BME280,Adafruit_GPIO

[2018-09-17 20:12:29] 
[2018-09-17 20:12:29] #### Installing/updating git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280 (pip-git)
[2018-09-17 20:12:29] 
[2018-09-17 20:12:29] #### Installing -e git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280 with pip
[2018-09-17 20:12:32] Obtaining adafruit-bme280 from git+git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280
[2018-09-17 20:12:32]   Cloning git://github.com/adafruit/Adafruit_Python_BME280.git to ./env/src/adafruit-bme280
[2018-09-17 20:12:36] Installing collected packages: adafruit-bme280
[2018-09-17 20:12:36]   Running setup.py develop for adafruit-bme280
[2018-09-17 20:12:39] Successfully installed adafruit-bme280

[2018-09-17 20:12:39] Successfully installed Adafruit_BME280

[2018-09-17 20:12:39] Successfully installed Adafruit_GPIO

[2018-09-17 20:12:40] 
[2018-09-17 20:12:40] #### Setting permissions

[2018-09-17 20:12:41] #### All Dependencies have been installed.

```
subsequent runs while trying to install just Adafruit_GPIO via the gui all end up like this
```
[2018-09-17 20:15:58] Dependency installation beginning. Installing: Adafruit_GPIO

[2018-09-17 20:15:58] Successfully installed Adafruit_GPIO

[2018-09-17 20:15:58] 
[2018-09-17 20:15:58] #### Setting permissions

[2018-09-17 20:15:59] #### All Dependencies have been installed.

```

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. I'm assuming manually uninstalling Adafruit_GPIO and then trying to install it via the gui?
 
### Additional Notes
This was solved manually and may be just an issue related to an upgrade to a bad releases that was quickly superseded so may not be a big issue.

kizniche, 2018-09-18:
Yes, I do believe this is an issue with the dependency system installing from pypi with pip. I'll investigate.

kizniche, 2018-09-18:
Fixed the issue and released v6.3.9. Thanks for the bug report.

closed 2018-09-18 by kizniche