kizniche/Mycodo #532: Command line not working

not5, 2018-09-19:
## Mycodo Issue Report
  - Specific Mycodo Version: 6.3.9
 
#### Problem Description
Please list:
  - Attempting to activate conditional controller via command line interface using "mycodo-client --activatecontroller Conditional XXX-XXX-XXX-XXX"
 
### Errors
  - Confirmation of activation received via command line
  - Controller displays inactive in web interface
  - Further upon attempting to activate the controller in the web interface results in:
"Error: Cannot activate Conditional controller with ID XXX-XXX-XXX-XXX: It's already active."
  - Although, the following messages are also reported upon attempting to activate in the web interface:
"Success: Conditional controller activated in SQL database
Success: Activate Conditional"

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Execute command in terminal
 2. Visually see that controller is not activated in web interface and that it is not executing as the conditions are met.
 3. Attempt to perform activation through web interface and see success & error message.
 
### Additional Notes
Same result when trying to deactivate the controller as activate; receive confirmation via command line but intended result is not performed.

kizniche, 2018-09-19:
Thanks for the bug report. I'll investigate. In the meantime, if you need this functionality, attempt to restore a previous version from the Backup Restore page.

kizniche, 2018-09-20:
Are there any errors in the daemon log?

not5, 2018-09-20:
Here's the relevant portion in daemon log:

> 2018-09-20 09:26:49,232 - mycodo.conditional_41544767 - INFO - Conditional controller deactivated in 108.0 ms
> 2018-09-20 09:27:09,851 - mycodo.conditional_41544767 - INFO - Conditional controller activated in 459.8 ms
> 2018-09-20 09:27:16,000 - mycodo.daemon - WARNING - Cannot activate Conditional controller with ID 41544767-1e0d-47b1-bcc4-46acbd683ca2: It's already active.
> 2018-09-20 09:28:28,803 - mycodo.conditional_41544767 - INFO - Conditional controller deactivated in 185.4 ms

kizniche, 2018-09-20:
What are the parameters of that particular conditional and its associated actions?

not5, 2018-09-20:
It's a basic test of the conditional: edge detection tied to a button. Turns on an LED upon detecting rising edge from button press.

kizniche, 2018-09-21:
Are you sure you're on v6.3.9 and not an earlier version? 6.3.9 fixed the known issue with edge detection. The same conditional setup you describe is working fine on one of my test systems running the latest version.

not5, 2018-09-21:
Condition works when active. The issue is that using the command line functions with mycodo-client to activate and deactivate it does not work.
Confirmed that same symptoms occur on other conditionals (like measurement based).

commit 2abf517b (2018-10-08): Fix issue with client activating controllers (fixes #532)
closed 2018-10-08 by kizniche