kizniche/Mycodo #535: Multiple rpi's running Mycodo connected. Just a thought.

Gossen1, 2018-09-27:
Hi Kyle,

First a big thanks and thumbs up. It is such a joy to work with mycodo. I use it to control the climate in 2 reefers to grow oyster mushrooms. My idea came from the following thought. It looks like I will be expanding my farm to 14 containers (yay) and off course I can monitor them with one RPi and some sensors but I wanna make the whole concept more modulair. So I want every container to have its dedicated RPi with mycodo running on it and on the outside of the container a wifi omnidirectional antenna. All this is very doable, but wouldn't it be nice to take all the data from every mycodo installation and merge them into one server/super mycodo. I hope you can follow my train of thoughts.

Like the title says: Just a thought. 

Maybe you have some ideas, pointers. Just felt like sharing. 

Btw The notes update is so helpful. Keeps things very clear.

leoneo, 2018-09-27:
Hi, 

Me too, I need this option please.

Thank you

kizniche, 2018-10-03:
This was first discussed in #45, and there's currently a rudimentary Remote Admin section in the Config menu.

kizniche, 2018-10-03:
As far as a full-feature remote administrayion system, that's fairly far off. There was enough progress made the last time I worked on this that the next time should be able to demo a lot of cool new multi-system features.

closed 2018-10-03 by kizniche