kizniche/Mycodo #536: Config > Raspberry Pi - Error 500

mschatte, 2018-10-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.3.9
 
#### Problem Description
Trying to get my DHT22 setup on a fresh install of Mycodo but I found that it's saying pigpiod isn't running. Attempted to go to Config > Raspberry Pi (as suggested) to reinstall but get error 500. 

### Errors
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_settings.py", line 320, in settings_pi
    pi_settings = get_raspi_config_settings()
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_settings.py", line 390, in get_raspi_config_settings
    settings['hostname'] = hostname_out.decode("utf-8")
AttributeError: 'str' object has no attribute 'decode'

### Steps to Reproduce the issue:
How can this issue be reproduced?
1. Fresh Raspbian install
2. Fresh install of Mycodo
3. Hooked up DHT22 sensor to GPIO 4 and configured as input.
4. Could never get the sensor to read data in Live. 
5. Found this in the Daemon log. 
2018-09-30 23:03:10,774 - mycodo.inputs.dht22_1 - ERROR - Could not connect to pigpiod. Ensure it is running and try again.
6. I then found on here that if I go to Config > Raspberry Pi I can reinstall pigpiod. However, I get an error 500 as soon as I click that. Full error is above

stevenvdr, 2018-10-02:
The `.decode("utf-8")` part of that line is unnecessary, I think. If I remove it, the page works properly. Though I'm not sure why it was there in the first place. Maybe it has something to do with locales that are not installed?

mschatte, 2018-10-04:
I checked the locale and it has en_US.UTF-8 enabled. I also tried to regenerate the locale stuff as explained here. https://www.jaredwolff.com/blog/raspberry-pi-setting-your-locale/

I ended up commenting out this line like you said and it started working fine. No more Error 500. The settings page is missing Hostname but oh well! Thanks!!!

`settings['hostname'] = hostname_out.decode("utf-8")`

kizniche, 2018-10-05:
I'll fix this in the next release. I'll be away all weekend, but if I get back soon enough I'll release late Sunday.

mschatte, 2018-10-05:
No worries! I've got a workaround for now. I'm still blown away by how much work has been put into this software. Thank you!

commit bfa6cddd (2018-10-08): Fix 500 Error on Raspberry Pi Config page (fixes #536)
closed 2018-10-08 by kizniche