kizniche/Mycodo #537: Crashes on linux command execution

SAM26K, 2018-10-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.39
 
#### Problem Description
Please list:
  - Using waveshare ADS1256 8 ch A/D converter to monitor analog soil moisture sensors.
     Since this pcb is not on the sensor list, I'm using some python code (called by some Linux bash code) to return the selected channels voltage.  This is the bash code: 
    sudo /etc/getch0.sh
    Above command calls the following command:
    sudo python ~/Desktop/ads1256code1/py-ads1256/read_volts_ch0.py

   read_volts_ch0.py contains the following python code:
   import ads1256       # import this lib                             

gain = 1			 # ADC's Gain parameter
sps = 25			 # ADC's SPS parameter

# Create the first list. It will receive ADC's absolute values
AllChannelValuesVolts = [0,0,0,0,0,0,0,0]       

# Create the second list. It will received absolute values converted to Volts
AllChannelValues = [0,0,0,0,0,0,0,0]		

# Initialize the ADC using the parameters
ads1256.start(str(gain),str(sps))  
#Fill the first list with all the ADC's absolute channel values
AllChannelValues = ads1256.read_all_channels()        
 for i in range(0, 8):
	# Fill the second list  with the voltage values
	AllChannelValuesVolts[i] = (((AllChannelValues[i] * 100) /167.0)/int(gain))/1000000.0   
#for i in range(0, 8):     
#    # Print all the absolute values
#    print AllChannelValues[i]              
# Print a new line
#print ("\n");							   
#for i in range(0, 8):     
    # Print all the Volts values converted from the absolute values
#    print AllChannelValuesVolts[i]
 print AllChannelValuesVolts[0]
# Stop the use of the ADC
ads1256.stop() 	

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

SAM26K, 2018-10-01:
Errors out when selecting "Live". Error log:
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 994, in page_live
    use_unit=use_unit)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 135, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 117, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 3, in top-level template code
    {% set help_page = ["live-measurements", _('Live Measurements')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 256, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 129, in block "body"
    <b><span id="{{each_input.unique_id}}-{{each_measure}}-value">0.0</span> {% if use_unit[each_input.unique_id][each_measure] and use_unit[each_input.unique_id][each_measure] in dict_units %}{{dict_units[use_unit[each_input.unique_id][each_measure]]['unit']}}{% endif %}</b> ({{dict_measurements[each_measure]['name']}}) | <span id="{{each_input.unique_id}}-{{each_measure}}-time">&lt;{{_('Please wait')}}&gt;</span><br>
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute ''

SAM26K, 2018-10-01:
Code is modified version of py-ads1256/read_volts_examply.py available on github.  When executed from a terminal on the pi itself (pi3B) it works fine and returns the voltage for channel 0 only one time. If I use the "Linux command" input function, it generates above error when "Live" tab is clicked.

SAM26K, 2018-10-01:
Sorry didn't mean to close issue.

SAM26K, 2018-10-01:
Code borrowed from here:
https://github.com/fabiovix/py-ads1256

SAM26K, 2018-10-01:
Using this RPi shield PCB:
https://www.amazon.com/Raspberry-Expansion-High-Precision-Functions-Interface/dp/B071JGZM4B/ref=pd_lpo_sbs_147_img_1?_encoding=UTF8&psc=1&refRID=86DDDF05ZWS6N44E88JJ

SAM26K, 2018-10-01:
Sorry, this is the first time Im posting on github.  Here; is second attempt to   post the modified python module  code to get the voltage from one channel:

import ads1256       # import this lib                             

gain = 1			 # ADC's Gain parameter
sps = 25			 # ADC's SPS parameter

# Create the first list. It will receive ADC's absolute values
AllChannelValuesVolts = [0,0,0,0,0,0,0,0]       

# Create the second list. It will received absolute values converted to Volts
AllChannelValues = [0,0,0,0,0,0,0,0]		

# Initialize the ADC using the parameters
ads1256.start(str(gain),str(sps))  

# Fill the first list with all the ADC's absolute channel values
AllChannelValues = ads1256.read_all_channels()        
                
for i in range(0, 8):
	# Fill the second list  with the voltage values
	AllChannelValuesVolts[i] = (((AllChannelValues[i] * 100) /167.0)/int(gain))/1000000.0   

#for i in range(0, 8):     
#    # Print all the absolute values
#    print AllChannelValues[i]              

# Print a new line
#print ("\n");							   

#for i in range(0, 8):     
    # Print all the Volts values converted from the absolute values
#    print AllChannelValuesVolts[i]
         
print AllChannelValuesVolts[0]

SAM26K, 2018-10-01:
That didn't work any better than the first  attempt,  Don't understand the weird formatting.
the bold print is not what I pasted so ignore those bits.

kizniche, 2018-10-03:
Thanks. I'll look into the issues and integrating the ADC

kizniche, 2018-10-08:
I fixed this issue with the Linux Command input, however, after you've updated to the latest release (not yet released), you will have to deactivate your controller, save the settings again, then reactivate it for it to be fixed.

kizniche, 2018-10-08:
I'll also work on adding this ADC to the builtin inputs.

SAM26K, 2018-10-10:
> I just added the ADC module. You will be able to test it once I make the
> 6.4.0 release, which could be any day now (even today, I'm not sure how
> much progress I'll make today toward a release, but I've had a pretty
> successful bug hunt so far).

Thank you very much for all your hard work.  Is there a way to force the
upgrade? If so, how?

kizniche, 2018-10-10:
You can upgrade to the master branch with the instructions at https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch

When the new release comes out, you will be able to upgrade from this version in the normal way via the upgrade page.

SAM26K, 2018-10-11:
Hi Kylem
I just tried to do a master upgrade and it failed.  Tried a second time
with a complete reinstall and has the same issue.  Gives the following
feedback:

[2018-10-11 06:22:56] #### Continuing Upgrade: Stage 2 of 2 ####
[2018-10-11 06:22:56]
[2018-10-11 06:22:56] Moving old Mycodo from /home/pi/Mycodo to
/var/Mycodo-backups/Mycodo-backup-2018-10-11_06-21-31-6.3.9...Done.
[2018-10-11 06:22:56] Moving new Mycodo from /tmp/Mycodo-6.3.9 to
/home/pi/Mycodo...Done.
[2018-10-11 06:23:26]
[2018-10-11 06:23:26] #### Upgrading Mycodo database with alembic (if needed)
[2018-10-11 06:23:33] Failed: Error while updating database with alembic.
[2018-10-11 06:23:33]
[2018-10-11 06:23:33]
[2018-10-11 06:23:33] There was an error during the upgrade.
[2018-10-11 06:23:33] Initial steps to try to fix:
[2018-10-11 06:23:33] 1. Reboot
[2018-10-11 06:23:33] 2. If that doesn't fix the issue, run the
following command:
[2018-10-11 06:23:33] sudo /bin/bash
~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade
[2018-10-11 06:23:33] 3. If that command returns that you are running
the latest version, run the following command:
[2018-10-11 06:23:33] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh

kizniche, 2018-10-11:
What was the error encountered during the fresh install. Your log excerpt is only from the upgrade.

kizniche, 2018-10-11:
So your fresh install works (fresh installs install from the latest release, not the master branch). Have you attempted an upgrade to the master branch after the fresh install?

SAM26K, 2018-10-11:
Just tried a third time. Complete reinstall of mycoda works ok but force
upgrade to master branch causes errors.  If I click on dashboard tab it
generates a 500 error after upgrading,  The upgrade worked once yesterday
or the day before but couldn't get the waveshare a/d working.  I noticed
you had done some more updates to the master so tried to upgrade again and
now have these errors.

kizniche, 2018-10-11:
Can you paste the contents of the upgrade log?

kizniche, 2018-10-11:
Sorry, but you'll have to log on to github.com to attach files. You can't do it by replying to the email notification. Thanks.

SAM26K, 2018-10-11:
Oh ok.  Here it is:
[2018-09-30 23:25:38]
[2018-09-30 23:25:38]
[2018-09-30 23:25:38] Installed version: 6.2.2
[2018-09-30 23:25:38] Latest version: 6.3.9
[2018-09-30 23:25:38]
[2018-09-30 23:25:38] #### Upgrade to v6.3.9 initiated 2018-09-30_23-25-33
####
[2018-09-30 23:25:38]
[2018-09-30 23:25:38] #### Beginning Upgrade: Stage 1 of 2 ####
[2018-09-30 23:25:42] Stopping the Mycodo daemon...Done.
[2018-09-30 23:26:39] Downloading latest Mycodo version to
/home/pi/mycodo-6.3.9.tar.gz...Done.
[2018-09-30 23:26:39] Creating /tmp/Mycodo-6.3.9...Done.
[2018-09-30 23:26:40] Extracting /home/pi/mycodo-6.3.9.tar.gz to
/tmp/Mycodo-6.3.9...Done.
[2018-09-30 23:26:40] Removing /home/pi/mycodo-6.3.9.tar.gz...Done.
[2018-09-30 23:26:40] Copying /tmp/Mycodo-6.3.9/.upgrade status file to
/tmp/Mycodo-6.3.9...Done.
[2018-09-30 23:26:40] Moving env directory...Done.
[2018-09-30 23:26:40] Copying databases from /home/pi/Mycodo/databases/ to
/tmp/Mycodo-6.3.9/databases...Done.
[2018-09-30 23:26:40] Copying flask_secret_key from
/home/pi/Mycodo/databases/ to /tmp/Mycodo-6.3.9/databases...Done.
[2018-09-30 23:26:40] Copying statistics ID...Done.
[2018-09-30 23:26:40] Copying SSL certificates...Done.
[2018-09-30 23:26:40] #### Stage 1 of 2 Complete ####
[2018-09-30 23:26:40]
[2018-09-30 23:26:40] #### Continuing Upgrade: Stage 2 of 2 ####
[2018-09-30 23:26:40]
[2018-09-30 23:26:40] Moving old Mycodo from /home/pi/Mycodo to
/var/Mycodo-backups/Mycodo-backup-2018-09-30_23-25-33-6.2.2...Done.
[2018-09-30 23:26:40] Moving new Mycodo from /tmp/Mycodo-6.3.9 to
/home/pi/Mycodo...Done.
[2018-09-30 23:27:10]
[2018-09-30 23:27:10] #### Upgrading Mycodo database with alembic (if
needed)
[2018-09-30 23:27:18]
[2018-09-30 23:27:18] Running post-upgrade script...
[2018-09-30 23:27:18]
[2018-09-30 23:27:18] #### Running initialization
[2018-09-30 23:27:18]
[2018-09-30 23:27:18] #### Creating mycodo user
[2018-09-30 23:27:19] The user `mycodo' is already a member of `adm'.
[2018-09-30 23:27:19] The user `mycodo' is already a member of `dialout'.
[2018-09-30 23:27:19] The user `mycodo' is already a member of `gpio'.
[2018-09-30 23:27:19] The user `mycodo' is already a member of `i2c'.
[2018-09-30 23:27:19] The user `mycodo' is already a member of `video'.
[2018-09-30 23:27:19]
[2018-09-30 23:27:19] #### Compiling mycodo_wrapper
[2018-09-30 23:27:21]
[2018-09-30 23:27:21] #### Creating symlinks to Mycodo executables
[2018-09-30 23:27:21]
[2018-09-30 23:27:21] #### Creating files and directories
[2018-09-30 23:27:21]
[2018-09-30 23:27:21] #### Setting permissions
[2018-09-30 23:27:22]
[2018-09-30 23:27:22] #### Removing statistics file
[2018-09-30 23:27:22]
[2018-09-30 23:27:22] #### Checking if swap size is 100 MB and needs to be
changed to 512 MB
[2018-09-30 23:27:22] #### Swap not currently set to 100 MB. Not changing.
[2018-09-30 23:27:22]
[2018-09-30 23:27:22] #### Checking python 3 virtualenv
[2018-09-30 23:27:22] #### Virtualenv already exists, skipping creation
[2018-09-30 23:27:22]
[2018-09-30 23:27:22]
[2018-09-30 23:27:22] #### Updating apt repositories
[2018-09-30 23:27:23] Hit:1 http://archive.raspberrypi.org/debian stretch
InRelease
[... 1106 lines cut ...]
/home/pi/Mycodo/install/requirements.txt (line 31)) (2.19.1)
[2018-10-10 14:02:53] Requirement already up-to-date: rpyc==3.4.4 in
./env/lib/python3.5/site-packages (from -r
/home/pi/Mycodo/install/requirements.txt (line 32)) (3.4.4)
[2018-10-10 14:02:54] Requirement already up-to-date: SQLAlchemy==1.2.7 in
./env/lib/python3.5/site-packages (from -r
/home/pi/Mycodo/install/requirements.txt (line 33)) (1.2.7)
[2018-10-10 14:02:54] Requirement already up-to-date: WTForms==2.1 in
./env/lib/python3.5/site-packages (from -r
/home/pi/Mycodo/install/requirements.txt (line 34)) (2.1)
[2018-10-10 14:02:55] Requirement already up-to-date: picamera==1.13 in
/usr/lib/python3/dist-packages (from -r
/home/pi/Mycodo/install/requirements.txt (line 37)) (1.13)
[2018-10-10 14:02:55] Requirement already up-to-date: RPi.GPIO in
/usr/lib/python3/dist-packages (from -r
/home/pi/Mycodo/install/requirements.txt (line 38)) (0.6.3)
[2018-10-10 14:02:55] Requirement already satisfied, skipping upgrade:
pbr>=0.11 in ./env/lib/python3.5/site-packages (from mock==2.0.0->-r
/home/pi/Mycodo/install/requirements.txt (line 6)) (4.3.0)
[2018-10-10 14:02:55] Requirement already satisfied, skipping upgrade:
six>=1.9 in /usr/lib/python3/dist-packages (from mock==2.0.0->-r
/home/pi/Mycodo/install/requirements.txt (line 6)) (1.10.0)
[2018-10-10 14:02:55] Requirement already satisfied, skipping upgrade:
pathlib2>=2.2.0; python_version < "3.6" in
./env/lib/python

kizniche, 2018-10-11:
From your log, above, you were performing an upgrade from 6.2.2 to 6.3.9, not upgrading to the master branch from a fresh install of the latest release (v6.3.9).

```
[2018-09-30 23:25:38] Installed version: 6.2.2 
[2018-09-30 23:25:38] Latest version: 6.3.9
```

kizniche, 2018-10-11:
When upgrading to the master branch, the first line of the upgrade log (excerpt) will be:

```
Upgrade script executed with the 'force-upgrade-master' argument. Upgrading from github repo master.
```

kizniche, 2018-10-11:
You happened to have caught me right as I am making the 6.4.0 release. It's now released, so you can either upgrade normally or perform a fresh install to get it.

kizniche, 2018-10-11:
v6.4.1 released that fixes the database upgrade issue

kizniche, 2018-10-11:
You can probably restore your previous install that worked and go through the normal upgrade and it will work (and preserve your previous settings so you don't have to start anew).

SAM26K, 2018-10-11:
Hi Kyle, That worked. Clean install but still have a problem with
ADS1256. Device not found error after clicking 'activate' on the add data
page.
Here is the last few lines of the log file.  You did write the code for the
waveshare  shield PCB and not just the chip right?

2018-10-11 10:01:59,664 - mycodo.daemon - INFO - Mycodo daemon v6.4.1
starting
2018-10-11 10:02:01,309 - mycodo.daemon - INFO - Anonymous statistics
enabled
2018-10-11 10:02:01,374 - mycodo.daemon - INFO - Starting rpyc server
2018-10-11 10:02:10,489 - mycodo.output - INFO - Output controller
activated in 252.6 ms
2018-10-11 10:02:10,991 - mycodo.daemon - INFO - All activated Conditional
controllers started
2018-10-11 10:02:10,993 - mycodo.daemon - INFO - All activated Input
controllers started
2018-10-11 10:02:10,993 - mycodo.daemon - INFO - All activated Math
controllers started
2018-10-11 10:02:10,995 - mycodo.daemon - INFO - All activated PID
controllers started
2018-10-11 10:02:10,996 - mycodo.daemon - INFO - All activated LCD
controllers started
2018-10-11 10:02:11,497 - mycodo.daemon - INFO - Mycodo daemon v6.4.1
started in 3.471 seconds
2018-10-11 10:02:11,508 - mycodo.daemon - INFO - 38.06 MB RAM in use
2018-10-11 10:05:51,592 - mycodo.daemon - ERROR - Could not activate Input
controller with ID 0afab39b-a337-4780-b5ea-ec45edb3d9df:
/home/pi/Mycodo/env/src/ads1256/ads1256.cpython-35m-arm-linux-gnueabihf.so:
undefined symbol: Py_InitModule3
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 507, in
controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 182, in __init__
    self.adc = input_loaded.ADCModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/ads1256_input.py", line 74, in
__init__
    import ads1256
ImportError: /home/pi/Mycodo/env/src/ads1256/
ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol: Py_InitModule3

https://www.waveshare.com/product/mini-pc/raspberry-pi/hats/high-precision-ad-da-board.htm

Thanks again.

   - Error: CError: Could not activate Input controller with ID
   0afab39b-a337-4780-b5eError: Could not activate Input controller with ID
   0afab39b-a337-4780-b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3 a-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3 ould not activate Input controller with ID
   0afab39b-a337-4780-Error: Could not activate Input controller with ID
   0afab39b-a337-4780-b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3 b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3

   -
      - Error: Could not activate Input controller with ID
      0afab39b-a337-4780-b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
      ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
      Py_InitModule3
   Error: Could not activate Input controller with ID
   0afab39b-a337-4780-b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3

   - Error: Could not activate Input controller with ID
   0afab39b-a337-4780-b5ea-ec45edb3d9df:
   - /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3

kizniche, 2018-10-11:
When you attempted to add the Input, did it prompt you to install bcm2835 and ads1256? Could you paste the dependency log of it installing these libraries?

kizniche, 2018-10-11:
It appears to be an issue with the driver you initially linked. It is developed for python 2.7, however Mycodo runs on python 3.5. Exploring options to fix this.

SAM26K, 2018-10-11:
it did prompt to install ads1256. I think bcm2835 is already installed.
Here's the dependency log:

[2018-10-10 14:37:16] Dependency installation beginning. Installing: ads1256

[2018-10-10 14:37:17]
[2018-10-10 14:37:17] #### Installing/updating git://
github.com/fabiovix/py-ads1256.git#egg=ads1256 (pip-git)
[2018-10-10 14:37:23] Looking in indexes: https://pypi.org/simple,
https://www.piwheels.org/simple
[2018-10-10 14:37:23] Obtaining ads1256 from git+git://
github.com/fabiovix/py-ads1256.git#egg=ads1256
[2018-10-10 14:37:23]   Cloning git://github.com/fabiovix/py-ads1256.git to
./env/src/ads1256
[2018-10-10 14:37:28] Installing collected packages: unknown
[2018-10-10 14:37:28]   Found existing installation: UNKNOWN 0.0.0
[2018-10-10 14:37:28]     Not uninstalling unknown at
/usr/local/lib/python3.5/dist-packages, outside environment
/home/pi/Mycodo/env
[2018-10-10 14:37:28]     Can't uninstall 'UNKNOWN'. No files were found to
uninstall.
[2018-10-10 14:37:28]   Running setup.py develop for unknown
[2018-10-10 14:37:38] Successfully installed unknown

[2018-10-10 14:37:38] Successfully installed ads1256

[2018-10-10 14:37:39]
[2018-10-10 14:37:39] #### Setting permissions

[2018-10-10 14:37:42] #### All Dependencies have been installed.

[2018-10-11 09:49:46] Dependency installation beginning. Installing: ads1256

[2018-10-11 09:49:46]
[2018-10-11 09:49:46] #### Installing/updating git://
github.com/fabiovix/py-ads1256.git#egg=ads1256 (pip-git)
[2018-10-11 09:49:54] Looking in indexes: https://pypi.org/simple,
https://www.piwheels.org/simple
[2018-10-11 09:49:54] Obtaining ads1256 from git+git://
github.com/fabiovix/py-ads1256.git#egg=ads1256
[2018-10-11 09:49:54]   Cloning git://github.com/fabiovix/py-ads1256.git to
./env/src/ads1256
[2018-10-11 09:50:00] Installing collected packages: unknown
[2018-10-11 09:50:00]   Found existing installation: UNKNOWN 0.0.0
[2018-10-11 09:50:00]     Not uninstalling unknown at
/usr/local/lib/python3.5/dist-packages, outside environment
/home/pi/Mycodo/env
[2018-10-11 09:50:00]     Can't uninstall 'UNKNOWN'. No files were found to
uninstall.
[2018-10-11 09:50:00]   Running setup.py develop for unknown
[2018-10-11 09:50:08] Successfully installed unknown

[2018-10-11 09:50:09] Successfully installed ads1256

[2018-10-11 09:50:10]
[2018-10-11 09:50:10] #### Setting permissions

[2018-10-11 09:50:13] #### All Dependencies have been installed.

kizniche, 2018-10-11:
It appears the ADS1256 library didn't compile properly, according to your log. It looks very odd. I'm still exploring whether I can port the library to Python 3 or find another suitable library.

SAM26K, 2018-10-11:
Thanks very much for all your help.

kizniche, 2018-10-11:
Can you test the library at https://github.com/ul-gh/PiPyADC ?

Follow these instructions to test:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh install-wiringpi
cd ~
git clone github.com/ul-gh/PiPyADC
cd PiPyADC
```

Then try:

```
~/Mycodo/env/bin/python ./example.py
```

and

```
~/Mycodo/env/bin/python ./example_2.py
```

kizniche, 2018-10-11:
If that doesn't work in python3, there's a python3 library at https://github.com/jaxbulsara/pyadda we can try next.

kizniche, 2018-10-11:
Actually, Could you also test the pyadda library as well. Execute the following commands:

```
~/Mycodo/env/bin/pip install RPi.GPIO
cd ~
git clone github.com/jaxbulsara/pyadda
cd pyadda
~/Mycodo/env/bin/python setup.py install
```

then try:

```
~/Mycodo/env/bin/python channel_read_example.py
```

and

```
sudo ~/Mycodo/env/bin/python channel_read_example.py
```

kizniche, 2018-10-11:
Please paste the entire output from the last commands that attempt to acquire measurements, as well as any output from previous commands that show any errors. Thanks.

SAM26K, 2018-10-11:
Not sure what you mean it never measures because it errors when I try to
activate it.  Get this message:
Error: Could not activate Input controller with ID
0afab39b-a337-4780-b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol: Py_InitModule3

   - Error: Could not activate Input controller with ID
   0afab39b-a337-4780-b5ea-ec45edb3d9df: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3

kizniche, 2018-10-11:
I just provided two step-by-step instructions to test two different libraries. I'm not sure what your last comment is in response to, but neither of the two instructions I mentioned requires interactive with the Mycodo interface. The instructions are for the command line.

SAM26K, 2018-10-12:
OK I didn't see that email before.  Will try now.
Thanks.

SAM26K, 2018-10-12:
Here's the result for above email:
pi@raspberrypi3_2:~ $ sudo /bin/bash
~/Mycodo/mycodo/scripts/upgrade_commands.sh install-wiringpi

#### Installing wiringpi
Cloning into '/home/pi/Mycodo/install/wiringPi'...
remote: Counting objects: 1177, done.
remote: Compressing objects: 100% (980/980), done.
remote: Total 1177 (delta 822), reused 212 (delta 142)
Receiving objects: 100% (1177/1177), 369.48 KiB | 153.00 KiB/s, done.
Resolving deltas: 100% (822/822), done.
wiringPi Build script
=====================

WiringPi Library
[UnInstall]
[Compile] wiringPi.c
[Compile] wiringShift.c
[Compile] wiringSerial.c
[Compile] piHiPri.c
[Compile] piThread.c
[Compile] wiringPiSPI.c
[Compile] wiringPiI2C.c
[Compile] softPwm.c
[Compile] softTone.c
[Compile] mcp23016.c
[Compile] mcp23008.c
[Compile] mcp23017.c
[Compile] mcp23s08.c
[Compile] mcp23s17.c
wiringPi.c:1309:21: warning: ‘digitalWrite8Dummy’ defined but not used
[-Wunused-function]
 static         void digitalWrite8Dummy       (UNU struct
wiringPiNodeStruct *node, UNU int pin, UNU int value) { return ; }
                     ^~~~~~~~~~~~~~~~~~
wiringPi.c:1308:21: warning: ‘digitalRead8Dummy’ defined but not used
[-Wunused-function]
 static unsigned int digitalRead8Dummy        (UNU struct
wiringPiNodeStruct *node, UNU int UNU pin)            { return 0 ; }
                     ^~~~~~~~~~~~~~~~~
[Compile] sr595.c
[Compile] pcf8574.c
[Compile] pcf8591.c
[Compile] mcp3002.c
[Compile] mcp3004.c
[Compile] mcp4802.c
[Compile] mcp3422.c
[Compile] max31855.c
[Compile] max5322.c
[Compile] ads1115.c
[Compile] bmp180.c
[Compile] sn3218.c
[Compile] htu21d.c
[Compile] ds18b20.c
[Compile] rht03.c
[Compile] drcSerial.c
[Compile] drcNet.c
[Compile] pseudoPins.c
[Compile] wpiExtensions.c
[Link (Dynamic)]
[Install Headers]
[Install Dynamic Lib]

WiringPi Devices Library
[UnInstall]
[Compile] ds1302.c
[Compile] maxdetect.c
[Compile] piNes.c
[Compile] gertboard.c
[Compile] piFace.c
[Compile] lcd128x64.c
[Compile] lcd.c
[Compile] scrollPhat.c
[Compile] piGlow.c
[Link (Dynamic)]
[Install Headers]
[Install Dynamic Lib]

GPIO Utility
[Compile] gpio.c
[Compile] readall.c
[Link]
[Install]

All Done.

NOTE: To compile programs with wiringPi, you need to add:
    -lwiringPi
  to your compile line(s) To use the Gertboard, MaxDetect, etc.
  code (the devLib), you need to also add:
    -lwiringPiDev
  to your compile line(s).

pi@raspberrypi3_2:~ $ cd ~
pi@raspberrypi3_2:~ $ git clone github.com/ul-gh/PiPyADC
fatal: repository 'github.com/ul-gh/PiPyADC' does not exist
pi@raspberrypi3_2:~ $
pi@raspberrypi3_2:~ $ git clone https://github.com/ul-gh/PiPyADC
fatal: destination path 'PiPyADC' already exists and is not an empty
directory.
pi@raspberrypi3_2:~ $ dir
ADS1256_py1      Documents    Mycodo Public     Templates
bcm2835-1.55      Downloads    oldconffiles  py-ads1256     Videos
bcm2835-1.55.tar.gz  helloworld.c  Pictures python_games
Desktop      Music    PiPyADC rpi_qsmhbzlyzi.sh
pi@raspberrypi3_2:~ $ cd PiPyADC
bash: cd: PiPyADC: Not a directory
pi@raspberrypi3_2:~ $ sudo rm -r PiPyADC
pi@raspberrypi3_2:~ $ git clone https://github.com/ul-gh/PiPyADC
Cloning into 'PiPyADC'...
remote: Enumerating objects: 15, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 80 (delta 4), reused 9 (delta 2), pack-reused 65
Unpacking objects: 100% (80/80), done.
pi@raspberrypi3_2:~ $ cd PiPyADC
pi@raspberrypi3_2:~/PiPyADC $ ~/Mycodo/env/bin/python ./example.py
Traceback (most recent call last):
  File "./example.py", line 13, in <module>
    from pipyadc import ADS1256
  File "/home/pi/PiPyADC/pipyadc.py", line 21, in <module>
    import wiringpi as wp
ImportError: No module named 'wiringpi'
pi@raspberrypi3_2:~/PiPyADC $
pi@raspberrypi3_2:~/PiPyADC $
pi@raspberrypi3_2:~/PiPyADC $ ~/Mycodo/env/bin/python ./example_2.py
  File "./example_2.py", line 102
    ads2.drate = DRATE_100
    ^
IndentationError: unexpected indent
pi@raspberrypi3_2:~/PiPyADC $

SAM26K, 2018-10-12:
Here is result of pyadda steps. The last command line,

sudo ~/Mycodo/env/bin/python channel_read_example.py,

was working displaying the analog voltages in a fast loop.  I had to hit
ctrl c to copy the terminal text and that's when some errors or warnings
appeared.

pi@raspberrypi3_2:~ $ git clone https://github.com/jaxbulsara/pyadda
Cloning into 'pyadda'...
remote: Enumerating objects: 38, done.
remote: Total 38 (delta 0), reused 0 (delta 0), pack-reused 38
Unpacking objects: 100% (38/38), done.
pi@raspberrypi3_2:~ $ cd pyadda
pi@raspberrypi3_2:~/pyadda $ ~/Mycodo/env/bin/python setup.py install
running install
running build
running build_ext
building 'pyadda' extension
creating build
creating build/temp.linux-armv7l-3.5
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall
-Wstrict-prototypes -g
-fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=.
-fstack-protector-strong -Wformat -Werror=format-security -Wdate-time
-D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m
-I/home/pi/Mycodo/env/include/python3.5m -c wrapper.c -o
build/temp.linux-armv7l-3.5/wrapper.o
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall
-Wstrict-prototypes -g
-fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=.
-fstack-protector-strong -Wformat -Werror=format-security -Wdate-time
-D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m
-I/home/pi/Mycodo/env/include/python3.5m -c ADDAlibrary.c -o
build/temp.linux-armv7l-3.5/ADDAlibrary.o
creating build/lib.linux-armv7l-3.5
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions
-Wl,-z,relro -Wl,-z,relro -g
-fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=.
-fstack-protector-strong -Wformat -Werror=format-security -Wdate-time
-D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/wrapper.o
build/temp.linux-armv7l-3.5/ADDAlibrary.o -lbcm2835 -o
build/lib.linux-armv7l-3.5/pyadda.cpython-35m-arm-linux-gnueabihf.so
running install_lib
copying build/lib.linux-armv7l-3.5/pyadda.cpython-35m-arm-linux-gnueabihf.so
-> /home/pi/Mycodo/env/lib/python3.5/site-packages
running install_egg_info
Writing
/home/pi/Mycodo/env/lib/python3.5/site-packages/UNKNOWN-0.0.0.egg-info
pi@raspberrypi3_2:~/pyadda $

pi@raspberrypi3_2:~/pyadda $ ~/Mycodo/env/bin/python channel_read_example.py
Segmentation fault
pi@raspberrypi3_2:~/pyadda $

sudo ~/Mycodo/env/bin/python channel_read_example.py

ID=
Ok, ASD1256 Chip ID = 0x3
Channel 0 - 1667.686mV
Channel 1 - 1702.297mV
^CTraceback (most recent call last):
Channel 2 - 1593.012mV
  File "channel_read_example.py", line 44, in <module>
Channel 3 - 1577.654mV
Channel 4 - 1649.254mV
    pass7 - 1151.542mV
KeyboardInterrupt
Fatal Python error: could not acquire lock for <_io.BufferedWriter
name='<stdout>'> at interpreter shutdown, possibly due to daemon threads

Thread 0x759c3470 (most recent call first):
  File "channel_read_example.py", line 30 in interruptInterpreter

Current thread 0x76fbb010 (most recent call first):
Aborted

SAM26K, 2018-10-12:
After doing all of the above, activation of an ADS1256 device yields same
error as before.

   - × Error: Could not activate Input controller with ID
   d4436c80-a68f-48de-8a64-e5d2b7cb8f1f: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3 Error: Could not activate Input controller with ID
   d4436c80-a68f-48de-8a64-e5d2b7cb8f1f: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3Error: Could not activate Input controller with ID
   d4436c80-a68f-48de-8a64-e5d2b7cb8f1f: /home/pi/Mycodo/env/src/ads1256/
   ads1256.cpython-35m-arm-linux-gnueabihf.so: undefined symbol:
   Py_InitModule3

SAM26K, 2018-10-12:
This is the mycodo rev info (frorm clicking the rev at top left):
Mycodo Version: 6.4.1
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516]
Database Version: da10608dbc1c
Daemon Status: Running
Daemon Process ID: 1185
Daemon RAM Usage: 45.812 MB
Daemon Virtualenv: Yes
Frontend RAM Usage: 50.396 MB
Frontend Virtualenv: Yes

SAM26K, 2018-10-12:
> You're not going to see the Mycodo module work until we figure out what
> code works and I update the module.
> Good. The pyadda module works.
> I'd like to ask if you could retry the PiPyADC module...
> ~/Mycodo/env/bin/pip install wiringpi

First feed back :
pi@raspberrypi3_2:~ $ git clone https://github.com/ul-gh/PiPyADC
Cloning into 'PiPyADC'...
remote: Enumerating objects: 15, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 80 (delta 4), reused 9 (delta 2), pack-reused 65
Unpacking objects: 100% (80/80), done.
pi@raspberrypi3_2:~ $ cd PiPyADC
pi@raspberrypi3_2:~/PiPyADC $ ~/Mycodo/env/bin/python ./example.py

PiPyADC: Example file for class ADS1256 in module pipyadc:

ADS1256 cycling through eight input channels.

Hardware: Waveshare ADS1256 board interfaced to the Raspberry Pi 3

Ulrich Lukas 2017-03-10

Press CTRL-C to exit.
Segmentation fault
pi@raspberrypi3_2:~/PiPyADC $ sudo ~/Mycodo/env/bin/python ./example.py

PiPyADC: Example file for class ADS1256 in module pipyadc:

ADS1256 cycling through eight input channels.

Hardware: Waveshare ADS1256 board interfaced to the Raspberry Pi 3

Ulrich Lukas 2017-03-10

Press CTRL-C to exit.
Segmentation fault

kizniche, 2018-10-12:
Okay. It seems of the 3 libraries, only one works with python3. I'll integrate that one into Mycodo and let you know when you can upgrade to the master branch and test.

SAM26K, 2018-10-12:
Excellent let me know when you want more testing.

SAM26K, 2018-10-12:
> Quick question: when you tested the pyadda module, did you have a channel
> connected to a known voltage to test if the returned voltage was accurate?
> If not, could you connect a source (like a AA battery) and rerun the test
> script?

I do have moisture sensors connected on six of the channels that looked ok
maybe sample rate too high but reasonable numbers.  The     RPi is already
embedded into an active garden and not accessible for physically
experimenting with.

SAM26K, 2018-10-12:
Nearing the end of the grow cycle though and would like to help you develop
calibration,

\\\\\\

kizniche, 2018-10-12:
I just pushed a fix for the ADS1256 module. Follow these steps to properly test if it works:

1. Deactivate and delete any ADS1256 inputs currently on your Data page.
2. Perform an upgrade to the master branch.
3. Add a ADS1256 Input on the Data page, configure, and activate.

SAM26K, 2018-10-13:
Hi Kyle,  Still not finding the device.  Here is the feedback and upgrade
log:
Activate:
Success: Input controller activated in SQL database
Error: Could not activate Input controller with ID
7c56228d-cdb5-464d-bc12-58b0bb01de3d: No module named 'adc_consts'

deactivate:
Success: Input controller deactivated in SQL database
Error: Input controller with ID 7c56228d-cdb5-464d-bc12-58b0bb01de3d not
found

Dependency Log:

30
Last 30 lines of /var/log/mycodo/mycododependency.log:
[2018-10-10 14:37:38] Successfully installed ads1256

[2018-10-10 14:37:39]
[2018-10-10 14:37:39] #### Setting permissions

[2018-10-10 14:37:42] #### All Dependencies have been installed.

[2018-10-11 09:49:46] Dependency installation beginning. Installing: ads1256

[2018-10-11 09:49:46]
[2018-10-11 09:49:46] #### Installing/updating git://
github.com/fabiovix/py-ads1256.git#egg=ads1256 (pip-git)
[2018-10-11 09:49:54] Looking in indexes: https://pypi.org/simple,
https://www.piwheels.org/simple
[2018-10-11 09:49:54] Obtaining ads1256 from git+git://
github.com/fabiovix/py-ads1256.git#egg=ads1256
[2018-10-11 09:49:54]   Cloning git://github.com/fabiovix/py-ads1256.git to
./env/src/ads1256
[2018-10-11 09:50:00] Installing collected packages: unknown
[2018-10-11 09:50:00]   Found existing installation: UNKNOWN 0.0.0
[2018-10-11 09:50:00]     Not uninstalling unknown at
/usr/local/lib/python3.5/dist-packages, outside environment
/home/pi/Mycodo/env
[2018-10-11 09:50:00]     Can't uninstall 'UNKNOWN'. No files were found to
uninstall.
[2018-10-11 09:50:00]   Running setup.py develop for unknown
[2018-10-11 09:50:08] Successfully installed unknown

[2018-10-11 09:50:09] Successfully installed ads1256

[2018-10-11 09:50:10]
[2018-10-11 09:50:10] #### Setting permissions

[2018-10-11 09:50:13] #### All Dependencies have been installed.

Upgrade Log:

[2018-10-13 00:46:01] Upgrade script executed with the
'force-upgrade-master' argument. Upgrading from github repo master.
[2018-10-13 00:46:01]
[2018-10-13 00:46:01] #### Upgrade to v6.4.1 initiated 2018-10-13_00-45-56
####
[2018-10-13 00:46:01]
[2018-10-13 00:46:01] #### Beginning Upgrade: Stage 1 of 2 ####
[2018-10-13 00:46:04] Stopping the Mycodo daemon...Done.
[2018-10-13 00:46:58] Downloading latest Mycodo version to
/home/pi/Mycodo-master.tar.gz...Done.
[2018-10-13 00:46:58] Creating /tmp/Mycodo-6.4.1...Done.
[2018-10-13 00:46:59] Extracting /home/pi/Mycodo-master.tar.gz to
/tmp/Mycodo-6.4.1...Done.
[2018-10-13 00:46:59] Removing /home/pi/Mycodo-master.tar.gz...Done.
[2018-10-13 00:46:59] Copying /tmp/Mycodo-6.4.1/.upgrade status file to
/tmp/Mycodo-6.4.1...Done.
[2018-10-13 00:46:59] Moving env directory...Done.
[2018-10-13 00:46:59] Copying databases from /home/pi/Mycodo/databases/ to
/tmp/Mycodo-6.4.1/databases...Done.
[2018-10-13 00:46:59] Copying flask_secret_key from
/home/pi/Mycodo/databases/ to /tmp/Mycodo-6.4.1/databases...Done.
[2018-10-13 00:46:59] Copying statistics ID...Done.
[2018-10-13 00:46:59] Copying SSL certificates...Done.
[2018-10-13 00:46:59] Copying mycodo/inputs/custom_inputs...Done.
[2018-10-13 00:46:59] #### Stage 1 of 2 Complete ####
[2018-10-13 00:46:59]
[2018-10-13 00:46:59] #### Continuing Upgrade: Stage 2 of 2 ####
[2018-10-13 00:46:59]
[2018-10-13 00:46:59] Moving old Mycodo from /home/pi/Mycodo to
/var/Mycodo-backups/Mycodo-backup-2018-10-13_00-45-56-6.4.1...Done.
[2018-10-13 00:46:59] Moving new Mycodo from /tmp/Mycodo-6.4.1 to
/home/pi/Mycodo...Done.
[2018-10-13 00:47:29]
[2018-10-13 00:47:29] #### Upgrading Mycodo database with alembic (if
needed)
[2018-10-13 00:47:33]
[2018-10-13 00:47:33] Running post-upgrade script...
[2018-10-13 00:47:33]
[2018-10-13 00:47:33] #### Running initialization
[2018-10-13 00:47:34]
[2018-10-13 00:47:34] #### Creating mycodo user
[2018-10-13 00:47:34] The user `mycodo' is already a member of `adm'.
[2018-10-13 00:47:34] The user `mycodo' is already a member of `dialout'.
[2018-10-13 00:47:34] The user `mycodo' is already a member of `gpio'.
[2018-10-13 00:47:34] The user `mycodo' is already a member of `i2c'.
[2018-10-13 00:47:35] The user `mycodo' is already a member of `video'.
[2018-10-13 00:47:35]
[2018-10-13 00:47:35] #### Compiling mycodo_wrapper
[2018-10-13 00:47:36]
[2018-10-13 00:47:36] #### Creating symlinks to Mycodo executables
[2018-10-13 00:47:36]
[2018-10-13 00:47:36] #### Creating files and directories
[2018-10-13 00:47:37]
[2018-10-13 00:47:37] #### Setting permissions
[2018-10-13 00:47:38]
[2018-10-13 00:47:38] #### Removing statistics file
[2018-10-13 00:47:38]
[2018-10-13 00:47:38] #### Checking if swap size is 100 MB and needs to be
changed to 512 MB
[2018-10-13 00:47:38] #### Swap not currently set to 100 MB. Not changing.
[2018-10-13 00:47:38]
[2018-10-13 00:47:38] #### Checking python 3 virtualenv
[2018-10-13 00:47:38] #### Virtualenv already exists, skipping creation
[2018-10-13 00:47:38]
[2018-10-13 00:47:38]
[2018-10-13 00:47:38] #### Updating apt repositories
[2018-10-13 00:47:39] Hit:1 http://archive.raspberrypi.org/debian stretch
InRelease
[2018-10-13 00:47:39] Get:2 http://raspbian.raspberrypi.org/raspbian
stretch InRelease [15.0 kB]
[2018-10-13 00:47:41] Get:3 http://raspbian.raspberrypi.org/raspbian
stretch/main armhf Packages [11.7 MB]
[2018-10-13 00:49:02] Fetched 11.7 MB in 1min 23s (140 kB/s)
[2018-10-13 00:49:10] Reading package lists...
[2018-10-13 00:49:11]

SAM26K, 2018-10-13:
Here is current  version info:
Mycodo Version: 6.4.1
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516]
Database Version: da10608dbc1c
Daemon Status: Running
Daemon Process ID: 1212
Daemon RAM Usage: 46.236 MB
Daemon Virtualenv: Yes
Frontend RAM Usage: 53.104 MB
Frontend Virtualenv: Yes
uptime

 01:33:28 up 23 min,  1 user,  load average: 0.49, 0.89, 0.79

Kernel Information: uname -a

Linux raspberrypi3_2 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST
2018 armv7l GNU/Linux

SAM26K, 2018-10-13:
I think part of the upgrade log was missing in the previous email.  Here's
the other half:
[2018-10-13 00:47:38] #### Checking python 3 virtualenv
[2018-10-13 00:47:38] #### Virtualenv already exists, skipping creation
[2018-10-13 00:47:38]
[2018-10-13 00:47:38]
[2018-10-13 00:47:38] #### Updating apt repositories
[2018-10-13 00:47:39] Hit:1 http://archive.raspberrypi.org/debian stretch
InRelease
[2018-10-13 00:47:39] Get:2 http://raspbian.raspberrypi.org/raspbian
stretch InRelease [15.0 kB]
[2018-10-13 00:47:41] Get:3 http://raspbian.raspberrypi.org/raspbian
stretch/main armhf Packages [11.7 MB]
[2018-10-13 00:49:02] Fetched 11.7 MB in 1min 23s (140 kB/s)
[2018-10-13 00:49:10] Reading package lists...
[2018-10-13 00:49:11]
[2018-10-13 00:49:11] #### Installing prerequisite apt packages and update
pip
[2018-10-13 00:49:11] Hit:1 http://archive.raspberrypi.org/debian stretch
InRelease
[2018-10-13 00:49:11] Hit:2 http://raspbian.raspberrypi.org/raspbian
stretch InRelease
[2018-10-13 00:49:17] Reading package lists...
[2018-10-13 00:49:18] Reading package lists...
[2018-10-13 00:49:20] Building dependency tree...
[2018-10-13 00:49:20] Reading state information...
[2018-10-13 00:49:22] Package 'apache2' is not installed, so not removed
[2018-10-13 00:49:22] The following packages were automatically installed
and are no longer required:
[2018-10-13 00:49:22]   libpython-all-dev lxkeymap python-all
python-all-dev python-cairo
[2018-10-13 00:49:22]   python-gobject python-gobject-2 python-gtk2
python-keyring
[2018-10-13 00:49:22]   python-keyrings.alt python-secretstorage
python-wheel python-xklavier
[2018-10-13 00:49:22] Use 'apt autoremove' to remove them.
[2018-10-13 00:49:22] 0 upgraded, 0 newly installed, 0 to remove and 20 not
upgraded.
[2018-10-13 00:49:22] Reading package lists...
[2018-10-13 00:49:23] Building dependency tree...
[2018-10-13 00:49:23] Reading state information...
[2018-10-13 00:49:24] fswebcam is already the newest version (20140113-1).
[2018-10-13 00:49:24] gawk is already the newest version (1:4.1.4+dfsg-1).
[2018-10-13 00:49:24] gcc is already the newest version (4:6.3.0-4).
[2018-10-13 00:49:24] git is already the newest version (1:2.11.0-3+deb9u4).
[2018-10-13 00:49:24] libffi-dev is already the newest version (3.2.1-6).
[2018-10-13 00:49:24] libi2c-dev is already the newest version (3.1.2-3).
[2018-10-13 00:49:24] logrotate is already the newest version (3.11.0-0.1).
[2018-10-13 00:49:24] moreutils is already the newest version (0.60-1).
[2018-10-13 00:49:24] nginx is already the newest version (1.10.3-1+deb9u1).
[2018-10-13 00:49:24] python-setuptools is already the newest version
(33.1.1-1).
[2018-10-13 00:49:24] python3 is already the newest version (3.5.3-1).
[2018-10-13 00:49:24] python3-dev is already the newest version (3.5.3-1).
[2018-10-13 00:49:24] python3-smbus is already the newest version (3.1.2-3).
[2018-10-13 00:49:24] sqlite3 is already the newest version
(3.16.2-5+deb9u1).
[2018-10-13 00:49:24] wget is already the newest version (1.18-5+deb9u2).
[2018-10-13 00:49:24] The following packages were automatically installed
and are no longer required:
[2018-10-13 00:49:24]   libpython-all-dev lxkeymap python-all
python-all-dev python-cairo
[2018-10-13 00:49:24]   python-gobject python-gobject-2 python-gtk2
python-keyring
[2018-10-13 00:49:24]   python-keyrings.alt python-secretstorage
python-wheel python-xklavier
[2018-10-13 00:49:24] Use 'apt autoremove' to remove them.
[2018-10-13 00:49:24] 0 upgraded, 0 newly installed, 0 to remove and 20 not
upgraded.
[2018-10-13 00:49:30] Searching for pip
[... 286 lines cut ...]
[2018-10-13 00:51:07] #### Updating Mycodo restart monitor crontab entry
[2018-10-13 00:51:07] Removing crontab entry ...
[2018-10-13 00:51:07] Appending to crontab:
[2018-10-13 00:51:07] -----------------------------------------------
[2018-10-13 00:51:07] @reboot /var/mycodo-root/env/bin/python
/var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
[2018-10-13 00:51:07] -----------------------------------------------
[2018-10-13 00:51:07] To remove previously added crontab entry, run:
/home/pi/Mycodo/install/crontab.sh restart_daemon --remove
[2018-10-13 00:51:07]
[2018-10-13 00:51:08]
[2018-10-13 00:51:08] #### Setting permissions
[2018-10-13 00:51:09]
[2018-10-13 00:51:09] #### Restarting the Mycodo daemon
[2018-10-13 00:51:12] Mycodo previously shut down properly
[2018-10-13 00:51:18]
[2018-10-13 00:51:18] #### Restarting nginx
[2018-10-13 00:51:23] #### Reloading mycodoflask
[2018-10-13 00:51:24]
[2018-10-13 00:51:24] #### Connecting to http://localhost (creates Mycodo
database if it doesn't exist)
[2018-10-13 00:51:31] #### Successfully connected to http://localhost
[2018-10-13 00:51:31] Done.
[2018-10-13 00:51:31]
[2018-10-13 00:51:31] Upgrade completed successfully without errors.

kizniche, 2018-10-13:
Pasting long logs in this thread makes it difficult for reading. So, if you have a long long you want to share, please log into github.com with a browser and attach the file. Do not try to attach it in an email or paste it in an email. I've gone through and deleted the long log portions here so it's easier for me to navigate this thread.

> Still not finding the device. Here is the feedback and upgrade log: Activate: Success: Input controller activated in SQL database Error: Could not activate Input controller with ID 7c56228d-cdb5-464d-bc12-58b0bb01de3d: No module named 'adc_consts'

Did you follow all 3 instructions I gave that would have you deactivate/delete the current input device and add it again? This is required to install the new dependency. The ```No module named 'adc_consts'``` error indicates you didn't perform steps 1 and 2.

SAM26K, 2018-10-13:
Sorry about that Kyle, Would you mind repeating step 1 and 2 instructions?

SAM26K, 2018-10-14:
> I was asking if you've deleted all current ADS1256 Inputs you may have on
> the Data page. After that, when you re-added one, there may be a prompt to
> install a dependency. If you have already performed this, before the error
> you pasted (#537 (comment)
> <https://github.com/kizniche/Mycodo/issues/537#issuecomment-429522575>),
> then I need to investigate something else.

I did that.  ie deleted existing data inputs before upgrading to the
master.    It still can't find the device.

kizniche, 2018-10-14:
```adc_consts``` is not the device, they are a set of variables, and is a part of the pyadda python library. Looking why you're unable to access them.

kizniche, 2018-10-14:
I'm unable to replicate your issue. It's working find from my end (except the actual measurement because I don't have the device).

I just released v6.4.2. Could you delete your entire existing install (```mv ~/Mycodo ~/Mycodo-backup```) and install from the beginning?

kizniche, 2018-10-14:
These would be the proper steps to installing new:

    1. sudo service mycdoo stop
    2. sudo service mycodoflask stop
    3. mv ~/Mycodo ~/Mycodo-backup

Then install according the instructions in the README

SAM26K, 2018-10-14:
6.4.2 has the same problem. Input adds ok but activate generates same error
as before,  no prompts to add depency,
×

   - × Error: Could not activate Input controller with ID
   6c39deac-c01c-407e-961a-5ea6689015f4: No module named 'adc_consts'Error:
   Could not activate Input controller with ID
   6c39deac-c01c-407e-961a-5ea6689015f4: No module named 'adc_consts'

×

   - Error: Could not activate Input controller with ID
   6c39deac-c01c-407e-961a-5ea6689015f4: No module named 'adc_consts'

kizniche, 2018-10-14:
If you're not prompted to install the dependency, that means you have not deleted your current install and installed fresh.

SAM26K, 2018-10-14:
I thought renaming mycodo did that.  What else should I do to make sure its
uninstalled?

kizniche, 2018-10-14:
See https://github.com/kizniche/Mycodo/issues/537#issuecomment-429586522

[92 more replies]

commit 8f9cbea8 (2018-10-08): Fix issue with Linux Command Input (fixes #537)
closed 2018-12-09 by kizniche