kizniche/Mycodo #538: Ver 6.3.3 AM2302 sensor

James11325, 2018-10-03:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
Three months ago I am operating 5 mushroom growing houses without any problem. Each room has 5 ASAIR 2302 sensors. Everything worked fine on a Raspberry PI3 B +
Two days ago I switched to version 6.3.3 and the sensors no longer work.
What should I do?
Thank you
How can this issue be reproduced?
 1. Just connect a sensor. 
 2. Add the only Input option you have is AOSONG DHT22
 3. All is OK but no data is displayed.
 Thanks
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-10-03:
Are there any errors in the daemon log?

James11325, 2018-10-03:
Kyle
Thanks for your response.  Right now I resolve the problem starting demon
with sudo pigpiod. I don´t know if after every reboot I need to activate
daemon again.
Part of the problem was that I have a Cozir CO2 sensor working at the same
time and if I activate first the daemon I can´t read any data in CO2 sensor.
Then I first define de CO2 sensor, activate the daemon and at last define
the DHT22 sensor.

Thanks again and your system is very, very good.

kizniche, 2018-10-08:
I'm closing this issue so I can track that I fixed it. However, once I make a release and you can test the fix, please report whether it worked or not. If it did not work, please reopen this issue and let me know any details about any errors you received when following my instructions, above. Thanks.

closed 2018-10-08 by kizniche

Theoi-Meteoroi, 2018-10-08:
I found this reference that includes this sensor compared with the others available in Mycodo.   Really interesting stuff:  [Hygrometers compared](http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html)

kizniche, 2018-10-08:
Wow, that's some great data! I'll check it out later. ATM, I'm on a roll smashing these bugs. I'd like to make a bit more progress before I call it a night.