kizniche/Mycodo #539: Gateway time-out when activating GPIO/DHT22 input

stevenvdr, 2018-10-03:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:
I am trying to connect my DHT22 sensor to Mycodo, working on a clean install (both Pi and Mycodo). Sensor data line is on GPIO4. 

Using command line tool `Adafruit_Python_DHT` or the backend Mycodo Input driver, I am able to read from the sensor, so I know for sure the sensor is working and properly connected.

When I try to activate the sensor, it seems to work at first but whenever I go to another page or refresh the page I get a 504 error, caused by the gunicorn process that is taking up 100% of CPU, leaving no room to handle threads. The same happens if I try to access the last value of `GET /last/gpio_state/9dccaf09-8118-4a69-8d34-b30ce499c2e7/16`, which should return the last db entry of the given sensor. Instead it hangs on `mycodo_flask/routes_general.py` line 272 

```raw_data = dbcon.query(query_str).raw```

(if I return a 200 response before this line, everything keeps working, but of course I don't get any sensor data)

I was thinking of trying out the query (`SELECT LAST(value) FROM gpio_state WHERE device_id='9dccaf09-8118-4a69-8d34-b30ce499c2e7' AND time > now() - 16s`), but I can't figure out how to go to the influxdb to run the query manually.

[image]
[image]

Any ideas on what could be going wrong?

kizniche, 2018-10-03:
Are there any errors in the daemon log?

stevenvdr, 2018-10-07:
No errors, and nothing that I wouldn't expect... (Time difference is because I just plugged in the Pi - mycodo daemon started before the time was updated to the correct time)
```
2018-10-06 19:52:09,906 - mycodo.daemon - INFO - Mycodo daemon v6.3.9 starting
2018-10-06 19:52:10,412 - mycodo.daemon - INFO - Anonymous statistics enabled
2018-10-06 19:52:10,438 - mycodo.daemon - INFO - Starting rpyc server
2018-10-06 19:52:10,771 - mycodo.output - INFO - Output controller activated in 108.7 ms
2018-10-06 19:52:11,273 - mycodo.daemon - INFO - All activated Conditional controllers started
2018-10-06 19:52:11,274 - mycodo.daemon - INFO - All activated Input controllers started
2018-10-06 19:52:11,274 - mycodo.daemon - INFO - All activated Math controllers started
2018-10-06 19:52:11,275 - mycodo.daemon - INFO - All activated PID controllers started
2018-10-06 19:52:11,275 - mycodo.daemon - INFO - All activated LCD controllers started
2018-10-06 19:52:11,776 - mycodo.daemon - INFO - Mycodo daemon v6.3.9 started in 1.868 seconds
2018-10-06 19:52:11,781 - mycodo.daemon - INFO - 38.17 MB RAM in use
2018-10-07 19:12:24,689 - mycodo.input_9dccaf09 - INFO - Activated in 424.4 ms
```
Is there a way to inspect the data in the db?

Just tried to do `GET /last/gpio_state/9dccaf09-8118-4a69-8d34-b30ce499c2e7/16`, with the data input disable in Mycodo. The result is the same: gateway timeout + gunicorn process taking 100% CPU and never ending...

Theoi-Meteoroi, 2018-10-07:
Just an observation - It isn't unexpected to have a positive time correction while running in most programs and OS.  What is unusual is are large negative corrections (while running) and I have seen that create problems for databases and other time-series oriented programs.  It is rare that the programmer codes for running suddenly in the past. Clocks matter in computing.

Can you reproduce this without the large negative time offset?  Is NTP enabled?

stevenvdr, 2018-10-07:
There are no negative corrections, only a positive correction to account for the 24 hours that the Pi was not getting any power. Behavior is exactly the same when no time correction happens. Is there supposed to be some logging info each time the daemon saves a datapoint?

As for NTP:
```      Local time: Sun 2018-10-07 21:47:50 CEST
  Universal time: Sun 2018-10-07 19:47:50 UTC
        RTC time: n/a
       Time zone: Europe/Brussels (CEST, +0200)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
```

What's the config of the database that should have been set up? Because I cannot get to open the connection to the database...

In config.py: 
```
# Influx sensor/device measurement database
INFLUXDB_HOST = 'localhost'
INFLUXDB_PORT = 8086
INFLUXDB_USER = 'mycodo'
INFLUXDB_PASSWORD = 'some_random_key'
INFLUXDB_DATABASE = 'mycodo_db'
```

However 
```
pi@raspberrypi:~/Mycodo/mycodo $ sudo -u mycodo influx -database 'mycodo_db' -host 'localhost' -port '8086'
Failed to connect to http://localhost:8086: Get http://localhost:8086/ping: dial tcp [::1]:8086: getsockopt: connection refused
Please check your connection settings and ensure 'influxd' is running.
```
With: 
```
pi@raspberrypi:~ $ systemctl status influxd
● influxdb.service - InfluxDB is an open-source, distributed, time series database
   Loaded: loaded (/lib/systemd/system/influxdb.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-10-06 19:52:00 CEST; 1 day 2h ago
     Docs: https://docs.influxdata.com/influxdb/
 Main PID: 339 (influxd)
   CGroup: /system.slice/influxdb.service
           └─339 /usr/bin/influxd -config /etc/influxdb/influxdb.conf

Oct 06 19:52:00 raspberrypi systemd[1]: Started InfluxDB is an open-source, distributed, time series database.
Oct 06 19:52:03 raspberrypi influxd[339]: ts=2018-10-06T17:52:03.026966Z lvl=info msg="InfluxDB starting" log_id=0AzcyPpW000 version=1.6.0 branch=1.6 
Oct 06 19:52:03 raspberrypi influxd[339]: ts=2018-10-06T17:52:03.027131Z lvl=info msg="Go runtime" log_id=0AzcyPpW000 version=go1.9.2 maxprocs=4
Oct 06 19:52:03 raspberrypi influxd[339]: ts=2018-10-06T17:52:03.167125Z lvl=info msg="Using data dir" log_id=0AzcyPpW000 service=store path=/var/lib/
Oct 06 19:52:03 raspberrypi influxd[339]: ts=2018-10-06T17:52:03.167374Z lvl=info msg="Open store (start)" log_id=0AzcyPpW000 service=store trace_id=0
```

Theoi-Meteoroi, 2018-10-08:
Sorry @stevenvdr  I don't know how I missed the date-stamp that shows it was not negative. My bad :).   About the only thing I have seen do this is when I had a I2C bus problem with the data line pulled low - that makes the _**i2cdetect**_ program spin hard and slows the webpages almost to a halt when you try to select the status page.    Are you using the i2c bus?

kizniche, 2018-10-08:
@stevenvdr, can you test querying with ```curl``` (from [here](https://docs.influxdata.com/influxdb/v1.6/guides/querying_data/), exmaple:

```bash
curl -G 'http://localhost:8086/query?pretty=true' --data-urlencode "db=mycodo_db" --data-urlencode "q=SELECT LAST(value) FROM gpio_state WHERE device_id=\"9dccaf09-8118-4a69-8d34-b30ce499c2e7\" AND time > now() - 120m"
```

kizniche, 2018-10-09:
Also, after these lines:

https://github.com/kizniche/Mycodo/blob/bfa6cddd637e064de0425f3b6aa9974c68363fd6/mycodo/mycodo_flask/routes_general.py#L258-L260

Try adding the following line and see if this helps with debugging by preventing the process from hanging:

```python
current_app.config['INFLUXDB_TIMEOUT'] = 5
```

stevenvdr, 2018-10-10:
I reinstalled influxdb (with fresh config file), which did the trick for running the influx db. (Unfortunately I didn't save the old config file :( )

After that I ran the `setup.sh` again to create the correct influx user and tables. 
```
#### Ensuring compatible version of influxdb is installed ####
#### Incorrect InfluxDB version (v1.0.2+dfsg1-1) installed. Installing v1.6.0-1...
(Reading database ... 125106 files and directories currently installed.)
Preparing to unpack influxdb_1.6.0_armhf.deb ...
Unpacking influxdb (1.6.0-1) over (1.0.2+dfsg1-1) ...
dpkg: error processing archive influxdb_1.6.0_armhf.deb (--install):
 trying to overwrite '/usr/bin/influx', which is also in package influxdb-client 1.0.2+dfsg1-1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Processing triggers for man-db (2.7.6.1-2) ...
Errors were encountered while processing:
 influxdb_1.6.0_armhf.deb

#### Creating InfluxDB database and user
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 30 seconds then trying again...
#### Attempting to connect...
#### Influxdb database and user successfully created
```

Still not getting any datapoints, but I'm guessing that's just because there is probably due to some local change I made. Will try to do a clean install.

----
Post I was writing before influx started working...
----

Influx db is not reachable from curl either...
```
pi@raspberrypi:~ $ curl -G 'http://localhost:8086/query?pretty=true' --data-urlencode "db=mycodo_db" --data-urlencode "q=SELECT LAST(value) FROM gpio_state WHERE device_id=\"9dccaf09-8118-4a69-8d34-b30ce499c2e7\" AND time > now() - 120m"curl: (7) Failed to connect to localhost port 8086: Connection refused
```

Timeout doesn't seem to help, it still hangs. Maybe the timeout only works for the query itself and not for setting up the connection to the db?

I've tried getting some sign of life from the influxdb, but without success:
- I've uncommented the following in `/etc/influxdb/influxdb.conf`
```[http]
  # Determines whether HTTP endpoint is enabled.
  enabled = true
  # The bind address used by the HTTP service.
  bind-address = ":8086"
  # Determines whether user authentication is enabled over HTTP/HTTPS.
  auth-enabled = false
```
- Ran netstat, which shows that influxdb seems to be running on 8088 instead of 8086
```pi@raspberrypi:/etc/nginx $ sudo netstat -lptun
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      723/nginx: master p 
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      583/sshd            
tcp        0      0 127.0.0.1:8088          0.0.0.0:*               LISTEN      1591/influxd        
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      723/nginx: master p 
tcp        0      0 0.0.0.0:18813           0.0.0.0:*               LISTEN      799/python          
tcp6       0      0 :::22                   :::*                    LISTEN      583/sshd            
tcp6       0      0 :::8888                 :::*                    LISTEN      444/pigpiod         
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           351/avahi-daemon: r 
udp        0      0 0.0.0.0:51143           0.0.0.0:*                           351/avahi-daemon: r 
udp        0      0 0.0.0.0:68              0.0.0.0:*                           356/dhcpcd          
udp6       0      0 :::5353                 :::*                                351/avahi-daemon: r 
udp6       0      0 :::57185                :::*                                351/avahi-daemon: r 
udp6       0      0 :::546                  :::*                                356/dhcpcd```
- Curl'ing on `localhost:8088` is more hopefull:
```pi@raspberrypi:/etc/nginx $ curl -G 'http://localhost:8088/query?pretty=true' --data-urlencode "db=mycodo_db" --data-urlencode "q=SELECT LAST(value) FROM gpio_state WHERE device_id=\"9dccaf09-8118-4a69-8d34-b30ce499c2e7\" AND time > now() - 120m"
curl: (56) Recv failure: Connection reset by peer```
with syslog 
```Oct 10 21:13:43 raspberrypi influxd[1591]: [tcp] 2018/10/10 21:13:43 tcp.Mux: handler not registered: 71. Connection from 127.0.0.1:46848 closed
Oct 10 21:13:55 raspberrypi influxd[1591]: [tcp] 2018/10/10 21:13:55 tcp.Mux: handler not registered: 71. Connection from 127.0.0.1:53744 closed```
- Setting the influxdb port to 8088 in `config.py`, doens't seem to change anything.
- Also in the `setup.log` the influxdb was not running (or not able to connect): 
```
#### Ensuring compatible version of influxdb is installed ####
Correct version of InfluxDB currently installed

#### Creating InfluxDB database and user
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 30 seconds then trying again...
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 30 seconds then trying again...
#### Attempting to connect...
#### Could not connect to Influxdb. Waiting 30 seconds then trying again...
```

closed 2018-10-10 by stevenvdr