kizniche/Mycodo #544: Quick bug in update_command.sh

not5, 2018-10-10:
Looks like you've hard coded the location for python3.5 as /usr/bin/python3.5. You should probably use a variable here and look up the system location for python3.5's bin.

kizniche, 2018-10-10:
At least with the environment I'm developing for, the binary location should be the same across all systems. Though, I agree there is a benefit to your recommendation, that being the eventual ability to operate on systems besides Raspberry Pis running Raspbian Stretch (such as banana Pi, beagle board, etc).

closed 2018-10-10 by kizniche

not5, 2018-10-11:
Running Jessie on KivyPie.
I manually changed the location in the script for my own use, but less savy users may not have figured it out right away.

commit 8c5f6b5a (2018-10-10): Update upgrade_commands.sh (closes #544)