kizniche/Mycodo #547: Identifying Input devices across reboots

kizniche, 2018-10-10:
This issue was briefly discusses with @Theoi-Meteoroi. Essentially, USB devices are assigned as /dev/ttyUSB*, where * is a an integer assigned by the kernel upon detection. If there are several USB devices connected, the linux device for the physical device may change across reboots. For instance, after plugging in two USB devices, the linux devices may be as follows:

Device 1: /dev/ttyUSB1
Device 2: /dev/ttyUSB2

However, after a reboot, the devices may now be:

Device 1: /dev/ttyUSB2
Device 2: /dev/ttyUSB1

This will obviously cause an issue if the location is set based on the linux device. What is needed is either a way to preserve the linux device designation across reboots or a way to probe the deivces for some sort of identifier that can be used to save the location in the Mycodo settings.

Theoi-Meteoroi, 2018-10-10:
I covered this in a very narrow way in my Hackaday project for the ZH03B.  I use udev to create a persistent device name ( '/dev/dust-sensor')  that will be linked to the /dev/ttyUSBn that is chosen at device arrival in the kernel.   The only requirement is some attribute returned from the USB device that is unique.   The common circumstance is that none of the attributes are unique and you get stuck with just VID and PID, which is ok as long as you don't have any other adapters that report the same VID and PID.   If you have multiple adapters with the same VID and PID, then hopefully they have some unique attribute.   This command will walk the attributes. Run on each USB device and then compare differences to possibly find some attribute to use.

<prompt> udevadm info --name=/dev/ttyUSB0 --attribute-walk  

I ended up using the serial number on the ZH03B to program the USB adapter serial field. This way guarantees unique serial numbers rather than me trying to remember what was the last serial number I used to increment by 1.

Here is the text from the Hackaday log:

When you plug a USB device in it can be enumerated to different device names by the operating system.  To fix this problem for this sensor on linux, I changed attributes that make the connection unique.

First - find the VID and PID for the USB device:

pi@GUIdev:~ $ lsusb
Bus 001 Device 008: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

In this case the Vendor ID is 10c4   The Product ID is ea60

Since I changed the serial number field - this will be unique.

pi@GUIdev:~ $ udevadm info --name=/dev/ttyUSB0 --attribute-walk  | grep serial
    SUBSYSTEMS=="usb-serial"
    *_ATTRS{serial}=="ZH03B180904"_*
    ATTRS{serial}=="3f980000.usb"

Now I have an attribute to tell udev what to do.  I create a file in /etc/udev/rules.d with a name like "99-dustsensor.rules".  In that file I tell udev what device name to create when it sees this device plugged in:

SUBSYSTEM=="tty", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", ATTRS{serial}=="ZH03B180904" SYMLINK+="dust-sensor"

To test the new rule:

pi@GUIdev:/dev $  sudo udevadm trigger

pi@GUIdev:/dev $ ls -al dust-sensor

lrwxrwxrwx 1 root root 7 Oct  6 21:04 dust-sensor -> ttyUSB0

Now, every time I plug in my dust sensor, it shows up at

/dev/dust-sensor

Theoi-Meteoroi, 2018-10-10:
The tool I used for the CP2102 is [Simplicity Studio](https://www.silabs.com/products/development-tools/software/simplicity-studio)

The other vendor you commonly run into is FTDI.  Their tools seem somewhat less easy to use, hard to find and the parts are relatively expensive. But they are huge in the market.   I'm looking into their 4-port [device](https://www.ftdichip.com/Products/ICs/FT4232H.htm) (have some on order) to create a 'sensor pod' that would connect over a single USB port and provide a mix of serial and I2C ports.   I'll write up the process when I have some experience.  Their customization tool is [FTprog](https://www.ftdichip.com/Support/Utilities.htm#FT_PROG)

closed 2018-11-26 by kizniche