kizniche/Mycodo #549: Issue May Only Affect Me...

not5, 2018-10-16:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.4
 
#### Problem Description
Please list:
mycodo.service & mycodoflask.service do not start successfully because smbus is not present on my distribution.
I have had no success in finding the source to build smbus for my system, but I have remedied this by installing smbus2, which I've heard is the exact same (or improved?) over smbus and is more widely available. I imported smbus2 as smbus in the two locations it is used (controller_lcd.py & chirp.py) with no issue and success.

Maybe it necessary that I create my own branch going forward?

kizniche, 2018-10-16:
I think you may have had an issue installing python3-smbus during the Mycodo install. You can try manually with ```sudo apt-get install python3-smbus```

kizniche, 2018-10-16:
Oh, that's right... you aren't using vanilla raspbian

kizniche, 2018-10-16:
You'll have to provide more details of your system if you're unable to use the apt version of python3 smbus

not5, 2018-10-16:
Yup- no package exists for python3-smbus for my distro. As I said, I might just need to create a branch to solve this as I'm sure most are just running vanilla raspbian. As I mentioned, smbus2 works just fine instead and it's readily available.

kizniche, 2018-10-16:
I'm willing to port a few things to allow cross-compatibility. There are just more places than the 2 files you mentioned that require smbus. I'm sure a few Inputs also require smbus, which may be more difficult to remedy.

kizniche, 2018-10-16:
Having a second look, smbus2 should work fine for all changes.

closed 2018-10-17 by kizniche