kizniche/Mycodo #550: Major changes for v7.0

kizniche, 2018-10-16:
Changes that will potentially cause issues with current systems:

  - [ ] Analog-to-digital controllers: Allow multiple channel reads from single Input controller

kizniche, 2018-10-22:
I've fully-implemented the refactoring of the conditional system and ADCs.

[image]

kizniche, 2018-10-22:
All ADC measurements have the same timestamp and align on graphs as a result:

[image]

kizniche, 2018-10-22:
This ADC refactoring should also substantially increase the possible sample rate from the controller.

kizniche, 2018-10-22:
Added the amg8833 as an input!

Here's some of the output:

[image]

[image]

[image]

kizniche, 2018-10-23:
If you only want to monitor certain pixels (channels), you can only enable those and attach them to a Function. It's a pretty cool sensor for measuring/monitoring several temperatures in an area. I'm looking forward to creating some sort of video/still image rendering/playback system.

kizniche, 2018-10-23:
I wonder if I could generate an image on the fly and place it in the graph tooltip.

[image]

kizniche, 2018-10-23:
I can now generate images from the influxdb data. I've increased the scale a bit. It really does look like from my left side, leaning in, with my hand on my chin.

[image]

kizniche, 2018-10-23:
I figured how to render the image server-side and serve it to the HighStock graph tooltip. I can also have it open a new window with the image when a point is clicked on the graph. I think I'll incorporate the latter, as rendering each image server-side takes a long time (1.5 sec per image) if you across across several data points it can take a significant amount of time to catch up. In the video, below, you can see the first few frames take some time to load, but then once cached, load immediately later in the video when I move the mouse to the left side of the graph.

Youtube video:

<a href="http://www.youtube.com/watch?feature=player_embedded&v=ewULkI3WmlU" target="_blank"><img src="http://img.youtube.com/vi/ewULkI3WmlU/0.jpg" 
alt="Mycodo thermal image graph preview" width="240" height="180" border="10" /></a>

kizniche, 2018-10-24:
I developed a nice alternative to the request-heavy tooltip method, above. It only makes a request when you click a data point.

Video:

<a href="http://www.youtube.com/watch?feature=player_embedded&v=g5yR87AJ7ak" target="_blank"><img src="http://img.youtube.com/vi/g5yR87AJ7ak/0.jpg" 
alt="Mycodo thermal image graph preview" width="240" height="180" border="10" /></a>

Theoi-Meteoroi, 2018-10-24:
Really good work Kyle! I'll grab a zip of this branch and give it a go.  The AMG8833 is a pretty amazing device for the cost.  The sensing range is 0-80 degrees C.  There is another model that will range -20-100 degrees C.

eap904, 2018-10-28:
any ETA on when this will be complete? Wanting to update lots of old installations from V5, but not if I'm going to need to do it again right away. Would a beta of 7 be up-gradable to the stable version once released?

Thanks Kyle, great work as always.

kizniche, 2018-10-28:
I'm not sure. I'm currently working on improving the multi-channel support (right now, in fact), and after that I'll need to run a lot of tests before release. At the moment, the 7.0 branch is stable and can be installed, but with the changes I have yet to push to github would necessitate a database deletion in order to upgrade. I'd recommend waiting a week or so until I get the release ready.

kizniche, 2018-10-28:
I'm now completely refactoring the Input measurement system so each measurement from an input has it's own SQL table, which will make organizing options for each measurement less of a headache, and enable the use of multi-channel inputs, such as ADCs, thermal imaging devices, and any input that may measure more than one of the same measurement (multiple voltages, temperatures, etc.). Initiating this refactoring has broken almost every part of Mycodo, though I'm confident this system will be vastly superior than the last. This may potentially extend the time til completion, though I'm not sure by how long.

eap904, 2018-10-29:
Thank you for the update. Great work as always

kizniche, 2018-10-30:
I've succesfully refactored a set of test Input modules to use the new measurement/unit framework. I've successfully started these three Inputs and have been acquiring measurements and storing them, and I've fixed the Live page and graphs on the Dashboard page (many more frontend/backend to go, but it's a good start). This new system makes it much easier to work with the data, because the measurement and unit it was measured as is stored with the measurement in the influx database. I've also made database schema changes that makes it easily known what channel and unit is used for every measurement (each measurement of an input has its own SQL table entry), which reduces code complexity and expands the possibility of more complex manipulation of measurement data. This next commit should be a hefty one.

kizniche, 2018-10-30:
If anyone wants to run the latest in the [major_release_v7.0](https://github.com/kizniche/Mycodo/tree/major_release_v7.0) branch, make sure you get the commit https://github.com/kizniche/Mycodo/commit/1b311fe305ec5dd200d98ef9a86f1c0ff68c2e89 instead of the latest commit, if you want it to work.

magichammer, 2018-11-12:
Good morning Kyle, I would have a proposal to make a new implementation for Mycodo, I asked you before about the connection of the Arduino, I looked for an interesting solution to solve the problem.
I found Nanpy, https://github.com/nanpy/nanpy-firmware, I think it is advantageous to use arduino as a slave data acquisition platform of raspberry pi, many analog and digital pins, usb connection, low cost, many addons.
as I understand, since I am not a programmer and I have a lot of difficulties for this type of activity.
try to give an eye to nanpy, since it makes transparent arduino to python, with a few additional parameters.
thank you very much for the commitment and the excellent Mycodo.
kind regards
Magic

kizniche, 2018-11-24:
Just a brief update. It took a lot more effort to accomplish the goals listed the original post, but I now have most parts of the system working again. We're close to release, but I have a few more parts I need to test. To conclude this post, I'll leave a screenshot of a graph of the past week's testing data which would bring most browsers to their knees:

[image]

mrplow, 2018-11-28:
@kizniche do you want new issues created by any of us who are crazy enough to be running the 7.0 branch?

I'm running f34616ab6eccd70e25677381371f01890aec477f
Creating a Daily (Time-based) method results in this:
```
Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_method.py", line 243, in method_builder
    MethodData.unique_id == display_order[-1]).first()
IndexError: list index out of range
```

kizniche, 2018-11-30:
@mrplow Thanks, it's now fixed in 7.0

closed 2018-12-09 by kizniche