kizniche/Mycodo #554: Implement single-file controller modules

kizniche, 2018-10-20:
Similar to #525, but for controllers. Allow user-created controller type modules to be imported 

Built-in controllers include (Conditional)[https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_conditional.py], (Input)[https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_input.py], (LCD)[https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_lcd.py], (Math)[https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_math.py], [Output)[https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_output.py], (PID)[https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_pid.py], and [Trigger](https://github.com/kizniche/Mycodo/blob/eae5e92861ab3f912d4fcd5c36634d1f19ab3b15/mycodo/controller_trigger.py).

kizniche, 2018-10-20:
I'm currently only working toward this, but without intention of including it in the v7.0 release, since implementing this won't break anyone's systems (maybe?). So, I'm not going to let the release of v7.0 hing on whether I get this implemented.

magichammer, 2018-11-11:
Good morning Kyle, I would have a proposal to make a new implementation for Mycodo, I asked you before about the connection of the Arduino, I looked for an interesting solution to solve the problem.
I found Nanpy, https://github.com/nanpy/nanpy-firmware, I think it is advantageous to use arduino as a slave data acquisition platform of raspberry pi, many analog and digital pins, usb connection, low cost, many addons.
as I understand, since I am not a programmer and I have a lot of difficulties for this type of activity.
try to give an eye to nanpy, since it makes transparent arduino to python, with a few additional parameters.
thank you very much for the commitment and the excellent Mycodo.
kind regards
  Magic