kizniche/Mycodo #556: Timelapse No Longer Working

berserktron3k, 2018-10-23:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.4
 
#### Problem Description
Please list: After the last few upgrades, Timelapse seems to have stopped working. Using either a PiCamera or WebCam, timelapse no longer works. You can set it up, however it won't take the pictures. Taking stills still seems to work, as well as stream (for the PiCamera)

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
 Can't see anything in the logs off hand.

### Steps to Reproduce the issue:
1. Set up a new timelapse
2. Watch it not take any pictures... ;)
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-10-23:
I just tested and couldn't reproduce your issue. What is indicating to you that it isn't working?

kizniche, 2018-11-30:
Are you still experiencing this issue? Do you have any extra details? Could you supply me with SSH access (user pi) and Mycodo access (admin user) to test?

berserktron3k, 2018-11-30:
Hey Kyle. Apologies. Yes, I'm still having the issue, but I haven't been
able to spend time on it lately. This weekend, I'll work on getting you SSH
access to it.

Brady Hummel

berserktron3k, 2018-12-10:
Looks like it was a database issue.

Once I upgraded to 7.0.1 it seems to be working again.

Brady Hummel

closed 2018-12-10 by kizniche