kizniche/Mycodo #558: Command   output controls still don"t    work.

SAM26K, 2018-10-27:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-10-27:
Fixed in 7.0

closed 2018-10-27 by kizniche

SAM26K, 2018-10-27:
Cool is there a way to patch files early pre-release of 7.0?

kizniche, 2018-10-27:
You can manually make these edits: https://github.com/kizniche/Mycodo/commit/29c048cb3492a0a6a07e6ecd113f849231fc0f62

SAM26K, 2018-10-27:
> You can manually make these edits: 29c048c

Hi Kyle,
Was comparing the code in your link above and the existing output.html file
on the RPi looks the same.
Is it possible that when I did the re-install it had the changes already?
If so then it didn't fix the problem of "command control does not
configure."  If not then I'm not seeing any changes.
What changes exactly did you make to output.html?
Thanks again.

SAM26K, 2018-10-27:
Update: I have 2 RPi3B's. The command (on/off) works on one but not the
other so something is messed up with one of the RPi's.
Any ideas of what could be interfering with it?  I hate to do another
"start from scratch" routine.

SAM26K, 2018-10-28:
Sorry but I'm not that familiar with Github. It looks like you replaced
lines 134-142 with a blank line (output.html)?

SAM26K, 2018-10-28:
I tried the edit above and got this error after a  system restart and click
on setup > output:Error 500: Internal Server Error

Something bad happened but it's *probably* not your fault. Letting the
developers know about these issues is crucial to supporting Mycodo. Please
submit a new issue on GitHub
<https://github.com/kizniche/Mycodo/issues/new> with
the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py",
line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py",
line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line
1372, in page_output
    user=user)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py",
line 134, in render_template
    return _render(ctx.app.jinja_env.get_or_select_template(template_name_or_list),
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py",
line 869, in get_or_select_template
    return self.get_template(template_name_or_list, parent, globals)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py",
line 830, in get_template
    return self._load_template(name, self.make_globals(globals))
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py",
line 804, in _load_template
    template = self.loader.load(self, name, globals)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/loaders.py",
line 125, in load
    code = environment.compile(source, name, filename)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py",
line 591, in compile
    self.handle_exception(exc_info, source_hint=source_hint)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py",
line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py",
line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/output.html",
line 139, in template
    {% endif %}
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py",
line 497, in _parse
    return Parser(self, source, name, encode_filename(filename)).parse()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 901, in parse
    result = nodes.Template(self.subparse(), lineno=1)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 883, in subparse
    rv = self.parse_statement()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 130, in parse_statement
    return getattr(self, 'parse_' + self.stream.current.value)()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 268, in parse_block
    node.body = self.parse_statements(('name:endblock',), drop_needle=True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 165, in parse_statements
    result = self.subparse(end_tokens)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 883, in subparse
    rv = self.parse_statement()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 144, in parse_statement
    self.fail_unknown_tag(token.value, token.lineno)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 97, in fail_unknown_tag
    return self._fail_ut_eof(name, self._end_token_stack, lineno)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 90, in _fail_ut_eof
    self.fail(' '.join(message), lineno)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/parser.py",
line 59, in fail
    raise exc(msg, lineno, self.name, self.filename)
jinja2.exceptions.TemplateSyntaxError: Encountered unknown tag
'endif'. Jinja was looking for the following tags: 'endblock'. The
innermost block that needs to be closed is 'block'.

SAM26K, 2018-10-28:
If I restore the original output,html file it doesn't have the  500 error
but command control still doesn't work.

SAM26K, 2018-10-29:
I figured out how to read the commit page and made the changes.  No more
errors but still doesn't execute the ssh commands.

kizniche, 2018-10-29:
Keep in mind the commands are executed as the user 'mycodo', not user 'pi'

SAM26K, 2018-10-29:
I just found that out.  What is the password for user mycodo? I think I
need to create a key so mycodo doesn't need a password.

kizniche, 2018-10-29:
There is no password set.

SAM26K, 2018-10-29:
If I use putty to logon as user mycodo it asks for a password and simply
hitting return doesn't work.  I tried the different passwords I use and
none work.  Should I delete mycodo user and recreate it?

kizniche, 2018-10-29:
That will probably destroy Mycodo. Just change the password.

SAM26K, 2018-10-30:
Well I got ssh without the password from  'mycodo' user on the moisture
sensor pi working.
I can log into the master pi as user mycodo and then send ssh commands to
the relay pi and switch the relays.  Those same commands in a
output>command control still don't switch the relays on the remote pi.
I notice the mycodo user is not a regular user.  I had to create a
/home/mycodo folder that is standard with the default pi user.
Not sure what to try next I should be up to date with all your fixes for
these issues so far.
Is there any way to change the user that mycodo uses to 'pi'?
Thanks.
ps these are the commands again:
ssh pi@192.168.1.51 mycodo-client --outputon
74958db7-1e20-4229-8164-bb085e5c7e33
ssh pi@192.168.1.51 mycodo-client --outputoff
74958db7-1e20-4229-8164-bb085e5c7e33

kizniche, 2018-10-30:
You can modify the code any way you choose, if you want it to use the user Pi, however you won't be able to use the upgrade system anymore.

SAM26K, 2018-10-30:
It's definitely a problem with the ssh access.  If I create a command
control on the local machine that has the relay board, the commands work.
eg:
 mycodo-client --outputon  74958db7-1e20-4229-8164-bb085e5c7e33
 mycodo-client --outputoff  74958db7-1e20-4229-8164-bb085e5c7e33
There must be a difference with logging on with putty vs what is seen from
the mycodo program.

kizniche, 2018-10-30:
You'll need to debug. First, verify commands are reaching the remote system. Then, send command output to a log to determine any errors.

SAM26K, 2018-10-30:
I tried collecting feedback from the command but it doesn't write to the
error.log file. I set error.log permissions to 777:
ssh pi@192.168.1.51 mycodo-client --outputon
74958db7-1e20-4229-8164-bb085e5c7e33 >> /var/log/mycodo/error.log
Any ideas?

kizniche, 2018-10-30:
My suggestion was to first verify the commands are reaching the remote host. You can't possibly have output if the commands are not reaching it. If you don't debug in a logical progression your results may not be interpretable.

SAM26K, 2018-10-30:
Do you use a particular development environment to write your code other
than the terminal CLI?

SAM26K, 2018-10-30:
Changed the button commands like this:
ssh mycodo@192.168.1.51 mycodo-client --outputon
74958db7-1e20-4229-8164-bb085e5c7e33 >> /var/log/mycodo/error.log 2>&1

generated these errors to the error.log file:
These are the errors I'm getting on the master pi after pressing the
command on button once:
Permission denied, please try again.

Permission denied, please try again.

Permission denied (publickey,password).

SAM26K, 2018-10-30:
Finally found a solution with sshpass. Passes the password with the ssh
command.  Not very secure but it works.
https://linux.die.net/man/1/sshpass.
Now the command lines look like this:
sshpass -p <password> ssh mycodo@192.168.1.51 mycodo-client --outputoff
74958db7-1e20-4229-8164-bb085e5c7e33

kizniche, 2018-10-31:
There has to be a more secure solution. Though at least you got it working. That's the first step.

SAM26K, 2018-10-31:
There must be a difference between logging on as user mycodo and
mycodo itself sending commands.  The two seem to look different to the
remote pi.

kizniche, 2018-10-31:
Of course. Mycodo executes commands with subprocess.Popen