kizniche/Mycodo #559: Request: Sensiron SHT31 Smart Gadget

eap904, 2018-11-03:
There are now a slew of 'wireless sensor tags' available which are extremely useful to plop around the grow room and find out temp/RH trends in different areas. 

The key is that they are wireless, with long battery life which makes using them much more likely. 

I stumbled upon the Sensiron SHT31 Smart Gadget... they have an example APP for iOS / Android that lets you see the collected data, and connects via BLE. At $28 shipped from Digi-Key. It's reasonable to deploy many of them. 

https://www.sensirion.com/en/environmental-sensors/humidity-sensors/development-kit/

Sensiron provides firmware for the gadget, but I have no idea about using this to connect it to the rPi, or then using a bluetooth device in Mycodo. 

https://github.com/Sensirion/SmartGadget-Firmware

Another one I have been using is SensorPush (http://a.co/d/10wGxID) which uses the SHT20. But it's more expensive at $50. I contacted the company and they may open source the device soon, but its currently proprietary.

And these I have not tried, although they seem to be open with their code etc. 
https://store.wirelesstag.net/collections/all

All I can do right now is request that someone work on integrating specifically the SHT31 device, but would be interested in learning how to make that contribution myself, if anyone has basic info on where to start.. 

Thanks

kizniche, 2018-11-03:
It seems there is a python library they provide: https://github.com/Sensirion/libsensors-python

eap904, 2018-11-03:
So, it seems the literature is there for accessing the SHT31 sensor itself, but how would an attached bluetooth device be accessed by that library? When I click 'add bluetooth device' I can see it appear.. but I'm presuming I need to somehow install their firmware for the pi

kizniche, 2018-11-03:
I would perhaps start with pybluez or bluepy for bluetooth device access from the RPi.

kizniche, 2018-12-06:
I just purchased one of these from Digikey. I'll let you know when I receive it and get a chance to test with my Pi.

kizniche, 2018-12-06:
[Sensirion/SmartGadget-Firmware/issues/4](https://github.com/Sensirion/SmartGadget-Firmware/issues/4) has some good information about connecting and measuring, so I'm including it for later reference.

eap904, 2018-12-07:
I'm sure you'll figure it out pretty quick. I don't really know the proper method of reading from bluetooth LE devices. There's trusting a device, pairing a device, and connecting to it...

The sensorpush tag I got works automatically with my phone, where once it's in range it just connects and fills in missing data automatically. The sensiron app is manual connection to sync the data, but I'm pretty sure it's just for demonstration. For the gadgets to have a long battery life, not sure the ideal method of connecting, rate of data collection etc, but it seems like once it's working, it would be pretty easy to make it a linux command input. 

For the Pi, it would be able to connect to the device 24/7, but suppose I lose power, etc, not sure through the UUID's etc how to access data logged to the device. 

Anyway, thanks as always. I have very little time to tinker with it

eap904, 2018-12-07:
I should also say, as it turns out the sensor tag's built in housing is a very good thing to have in the grow room... maybe someone has made like a 3d printed basic case for the smart gadget, but it certainly needs something to protect it.

kizniche, 2018-12-07:
What is the "sensorpush tag" you are referring to?

kizniche, 2018-12-09:
The commit I just pushed is for the SHT31 Smart Gadget. I was able to get an Input module working using bluepy. This will be in the 7.0.1 release I plan to create tomorrow.

kizniche, 2018-12-09:
I see some sporadic errors that I can probably iron out (disconnects, reconnects, etc.), so I'll let it run through the night and see what happens.

eap904, 2018-12-09:
Here's a link to the sensorpush sensor on Amazon: https://www.amazon.com/SensorPush-Wireless-Thermometer-Hygrometer-Android/dp/B01AEQ9X9I

It uses the sht21 chip. 

Can't wait to try it out!

kizniche, 2018-12-11:
I just stumbled upon a [full python implementation](https://github.com/Mailblocker/SmartGadgetReader) for reading and writing to the SHT31 Smart Gadget!

I'm scrapping my work so far because this does it all! The best part, which I've been struggling with, is its ability to download the data stored on the device. This will allow me to add options such as:

Download Stored Data (Boolean): Whether to download just the latest measurement (unchecked) or all stored data and the latest measurement (checked).
Set Logging Interval (Integer, seconds): How often the device will measure and store data to its internal memory.

Then I can add a check to make sure the download interval isn't too long based on the Logging Interval, due to the limited memory size on the device.

kizniche, 2018-12-11:
I think the SHT31 Smart Gadget module is finished. I added the options mentioned above and it appears to be working as expected. I have my Logging Interval set to 5 minutes and my Period to download the data set to 1 hour. I'll see how it performs overnight.

kizniche, 2018-12-15:
I fixed a critical bug that shifted all downloaded date back one interval. Since that, data downloading has been working flawlessly. At the time of data download there will also be battery percentage checked and logged as well. It seems this module has a very stable operation now.

Theoi-Meteoroi, 2018-12-15:
This is a very good addition to the Mycodo feature set.   Between AMS semiconductor and Sensiron, there are some really fine sensor technologies coming out of Europe that are vast improvements on what had been available at similar or even higher component costs.   This one is particularly good since it does not rely on the Pi power supply or any wired connections.  As much as I like building my own hardware - this device is cheaper to buy than building something like it.

eap904, 2018-12-16:
Is there a way to make it download only new data instead of all data? 

Curious for implementing something like this with other wired sensors, if your new input system gives the option of having sort of a backup sensor to control outputs, so if for instance the wireless sensor has an interruption or battery dies etc, it will then check the backup sensor or other logic before just breaking whatever control system is relying on it.

kizniche, 2018-12-16:
> this device is cheaper to buy than building something like it.

I really like how it's designed. It's been nice to test conditions in different areas so easily. I can see a lot of uses with determining optimal positions for my wired sensors.

> Is there a way to make it download only new data instead of all data?

Yes, this is how I designed the Mycodo module. If the start timestamp is set, only data after that will be downloaded. The module operated as follows:

  - Set start timestamp if there has been a previous download of data since the module has started, otherwise set start timestamp to 0
  - Sync current timestamp
  - Download data
  - Store current timestamp for next start timestamp

> option of having sort of a backup sensor

Nothing like this currently exists. However, you can have a conditional set up to notify you if measurements stop being acquired by a sensor. Adding the ability to set a backup sensor sounds interesting. It could be a type of math controller. For instance, return sensor 1 measurement if it exists in the past x seconds, else return sensor 2 measurement if it exists in the past x seconds.

eap904, 2018-12-21:
awesome! is this in the current version, 7.01?

kizniche, 2018-12-21:
No, this will be in 7.0.2

closed 2018-12-22 by kizniche