kizniche/Mycodo #560: Error when activating TSL2561

drgrumpy, 2018-11-05:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.5
Clean install
 
#### Problem Description
When activating a TSL2561 sensor  the following error is received:

- Error: Could not activate Input controller with ID f026114d-5447-4bbb-9734-9327ed55184c: No module named 'Adafruit_GPIO'

From the logs:
mycodo.daemon - WARNING - Input controller with ID f026114d-5447-4bbb-9734-9327ed55184c not found
2018-11-05 15:43:44,571 - mycodo.daemon - ERROR - Could not activate Input controller with ID f026114d-5447-4bbb-9734-9327ed55184c: No module named 'Adafruit_GPIO'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 506, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 204, in __init__
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/drgrumpy/Mycodo/mycodo/inputs/tsl2561.py", line 35, in __init__
    from tsl2561 import TSL2561
  File "/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/__init__.py", line 1, in <module>
    from .tsl2561 import *
  File "/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/tsl2561.py", line 5, in <module>
    from Adafruit_GPIO import I2C
ImportError: No module named 'Adafruit_GPIO'

It looks like Adafruit_GPIO is a dependency, but not listed as such....
And it seems cannot install without also installing a device that is dependent on it.

kizniche, 2018-11-05:
Thanks for catching this. I'll add it as a dependency for the 7.0 release. In the meantime, you should be able to add another Input that has it as a dependency to get it installed.

drgrumpy, 2018-11-05:
So I installed Adafruit_GPIO by installing MAX31855
deleted completely and then re-added, activated the tsl2561, but now there is 'No data'

[Was all working fine with 5.7.2 - due to sd card corruption after non use for a couple of months, I have upgraded with clean install of 6.4.5, but much has changed so not finding so easy to troubleshoot]

kizniche, 2018-11-05:
Are you able to acquire measurements with other software? Any errors in the daemon log?

drgrumpy, 2018-11-05:
I can get a sensible reading with python script from here:
<https://github.com/ControlEverythingCommunity/TSL2561>

No errors in the daemon log...

drgrumpy, 2018-11-06:
Restarted the daemon in debug mode and no useful clues, deactivated and reactivated it appears to be activated without error:
2018-11-06 11:03:03,895 - mycodo.input_cf9640bf - INFO - Activated in 2736.6 ms

but we then get (as I am displaying the data on the lcd):
2018-11-06 11:05:18,443 - mycodo.influxdb - DEBUG - No measurement available in the past 360 seconds.
2018-11-06 11:05:18,452 - mycodo.lcd_70aa9de2 - DEBUG - No data returned from influxdb

so apparently no data is being saved...
Reminder was working fine for me in version 5.7.2 (are there any other users of curent version with tsl2561 ?)

kizniche, 2018-11-06:
I believe I have one, but testing will have to wait until I return from a conference.

drgrumpy, 2018-11-06:
Was trying to look at figuring out myself.....
but much has changed and the sensor module seems more complex and more dependent on external files, I am curious what does all the
```
convert_units(
                    'light', 'lux', self.convert_to_unit,
self.tsl.lux())
```
do ? Why is it needed ? Is this the issue?

kizniche, 2018-11-07:
I just tested my TSL2561 with the latest 7.0 branch and it's working. I'm not sure what's wrong with your install or the stable version, but I'm putting effort into getting 7.0 released.

drgrumpy, 2018-11-08:
I modified tsl2561.py as follows to try and track down the issue:
```
        try:
            self._lux = self.get_measurement()
            self.logger.error("Measurement obtained")
            if self._lux is not None:
                self.logger.error("Lux value = {}".format(self._lux)) 
                return  # success - no errors
```
and get expected output....
2018-11-08 14:46:21,139 - mycodo.tsl2561_cf9640bf - ERROR - Measurement obtained
2018-11-08 14:46:21,147 - mycodo.tsl2561_cf9640bf - ERROR - Lux value = 2939

drgrumpy, 2018-11-08:
Also if I add:
```
   def get_measurement(self):
...
        try:
            lux = convert_units(
                'light', 'lux', self.convert_to_unit,
                self.tsl.lux())
                self.logger.error("lux value = {}".format(lux))
            return lux
```
I also get expect value, so it looks like the tsl2561.py module is reading ok, but perhaps the data is not being picked by the main module ?

kizniche, 2018-11-08:
Can you verify if the issue is maybe only related to the Live page? Can you check PIDs, Math, Graphs, Gauges, etc. and see what exactly cannot retrieve the measurements fro the database?

drgrumpy, 2018-11-08:
Also in the debug log:
2018-11-08 15:25:19,383 - mycodo.influxdb - DEBUG - No measurement available in the past 360 seconds.
2018-11-08 15:25:19,389 - mycodo.lcd_70aa9de2 - DEBUG - No data returned from influxdb
(as value is supposed to display on lcd)

drgrumpy, 2018-11-08:
I should say that value from other sensors - ds18b20s and and a temp and rh sensor data from command file are all working fine as expected...

drgrumpy, 2018-11-08:
I also deactivated, deleted, re-added, reactivated....
no effect....

kizniche, 2018-11-08:
Change this line:

https://github.com/kizniche/Mycodo/blob/e7699f0cf408d77851c5d679f0a842a27337d44e/mycodo/inputs/tsl2561.py#L60

to this:

```python
        return dict(light=float('{0:.2f}'.format(self._lux))) 
```

Save. Restart the daemon, and it should fix your issues.

drgrumpy, 2018-11-08:
Yay!  It works. Kyle you are a star.... and two issues fixed in one go! 

I guess this was one of the changes that were needed as a result of enabling the customisation of measurements ... too many options leads to complications that become difficult to maintain?

kizniche, 2018-11-08:
I'm not exactly sure when or how this issue arose, but the unreleased version 7.0 has undergone significant changes to input modules to enable a more robust measurement/unit system. The overall size of input modules has been reduced, but I wouldn't say the complexity has. Users that make their own input modules will have an easier time, hopefully, but I still have some ways to go before 7.0 is ready for release. Since I changed a core component of Mycodo, many other parts of the system must change as well, and I'll have to perform rigorous testing to be confident things will work when it's released. However, things are looking good. I have inputs working and dashboard elements retrieving data. Now it's about testing every other part of the system and fixing bugs as I find them.