kizniche/Mycodo #561: BH1750 Sensor issue - no readings

krillik, 2018-11-07:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.5(issue happening on 6.4.4 as well)
 
### Problem Description
I have added a BH1750 light sensor in the Data page under setup but when looking at live page or trying to get sensor data it comes back with 0 lux and no data for the last 16 seconds. I am able to get the light readings from a separate python file so I know the sensor is in fact working. 

A little about my setup:

I have a TCA9548A multiplexer connected per the instructions listed in your manual. I then have one BH1750 on smbus 5 and one on smbus 7. I can access both of these sensors with an independent python file unrelated to mycodo. 
 
### Errors
No errors are shown on the front end when activating, deactivating, or adding a BH1750. Unknown if errors are shown in logs as I am not sure where this would be. 

### Steps to Reproduce the issue:
How can this issue be reproduced?
Add the sensor and activate it then look at the live data page to see no data coming in from the sensor.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue? 

[image]

[image]

kizniche, 2018-11-07:
> I am able to get the light readings from a separate python file

What file are you referring to?

kizniche, 2018-11-07:
Are there any errors in the daemon log?

krillik, 2018-11-07:
I just used this to verify my sensor was working correctly. 
https://bitbucket.org/MattHawkinsUK/rpispy-misc/raw/master/python/bh1750.py 

### /var/log/mycodo/mycodo.log

2018-11-07 14:51:43,151 - mycodo.daemon - INFO - Mycodo daemon v6.4.5 starting
2018-11-07 14:51:43,216 - mycodo.daemon - DEBUG - Refreshing camera settings
2018-11-07 14:51:43,419 - mycodo.daemon - DEBUG - Refreshing misc settings
2018-11-07 14:51:43,452 - mycodo.daemon - INFO - Anonymous statistics enabled
2018-11-07 14:51:43,472 - mycodo.daemon - INFO - Starting rpyc server
2018-11-07 14:51:43,653 - mycodo.daemon - DEBUG - Starting Output controller
2018-11-07 14:51:43,685 - mycodo.output - DEBUG - Initializing Outputs
2018-11-07 14:51:43,735 - mycodo.output - DEBUG - Outputs Initialized
2018-11-07 14:51:43,736 - mycodo.output - INFO - Output controller activated in 81.9 ms
2018-11-07 14:51:44,238 - mycodo.daemon - DEBUG - Starting all activated Conditional controllers
2018-11-07 14:51:44,249 - mycodo.daemon - INFO - All activated Conditional controllers started
2018-11-07 14:51:44,250 - mycodo.daemon - DEBUG - Starting all activated Input controllers
2018-11-07 14:51:44,573 - mycodo.input_f30574d0 - INFO - Activated in 194.6 ms
2018-11-07 14:51:44,574 - mycodo.daemon - INFO - All activated Input controllers started
2018-11-07 14:51:44,575 - mycodo.daemon - DEBUG - Starting all activated Math controllers
2018-11-07 14:51:44,575 - mycodo.daemon - INFO - All activated Math controllers started
2018-11-07 14:51:44,576 - mycodo.daemon - DEBUG - Starting all activated PID controllers
2018-11-07 14:51:44,576 - mycodo.daemon - INFO - All activated PID controllers started
2018-11-07 14:51:44,576 - mycodo.daemon - DEBUG - Starting all activated LCD controllers
2018-11-07 14:51:44,577 - mycodo.daemon - INFO - All activated LCD controllers started
2018-11-07 14:51:45,078 - mycodo.daemon - INFO - Mycodo daemon started in 1.926 seconds
2018-11-07 14:51:45,093 - mycodo.daemon - INFO - 37.69 MB RAM in use
2018-11-07 14:52:15,274 - mycodo.input_f55c8332 - INFO - Activated in 236.0 ms
2018-11-07 14:53:45,196 - mycodo.stats - DEBUG - Sent anonymous usage statistics

that is the output of the log after a restart in debug mode and when I activated the sensor. I had previously deactivated it prior to starting in debug mode to see if any exceptions were thrown.

krillik, 2018-11-07:
> Run this script and see if you can obtain a measurement:
> 
> https://gist.github.com/oskar456/95c66d564c58361ecf9f

I was able to run this script no probem. 

pi@Psych:~ $ python test.py
Sensitivity: 69
Low Res  Light Level : 0.00 lx
HighRes  Light Level : 3.33 lx
HighRes2 Light Level : 2.92 lx
--------
Sensitivity: 79
Low Res  Light Level : 0.00 lx
HighRes  Light Level : 2.91 lx
HighRes2 Light Level : 2.91 lx
--------
Sensitivity: 89
Low Res  Light Level : 0.00 lx
HighRes  Light Level : 3.23 lx
HighRes2 Light Level : 2.91 lx

and so on...

krillik, 2018-11-07:
> Run this script and see if you can obtain a measurement:
> 
> https://gist.github.com/oskar456/95c66d564c58361ecf9f

I was able to get this to run without any issues

kizniche, 2018-11-07:
Try running it in a python 3 environment:

```~/Mycodo/env/bin/python /path/to/script.py```

kizniche, 2018-11-07:
If that works, then try changing smbus to smbus2 and smbus.SMBus() to SMBus(), from the following ines:

```import smbus```

to

```from smbus2 import SMBus```

and

```bus = smbus.SMBus(1)```

to

```bus = SMBus(1)```

krillik, 2018-11-07:
> Try running it in a python 3 environment:
> 
> `~/Mycodo/env/bin/python /path/to/script.py`

When trying to run the script you linked this way, I was unable to get it to load. 

$ ~/Mycodo/env/bin/python test.py
  File "test.py", line 125
    print "Sensitivity: {:d}".format(sensor.mtreg)
                            ^
SyntaxError: invalid syntax

kizniche, 2018-11-07:
Python 3 requires print statements to be encased in parentheses, so change:

```print "blah {}".format(variable)```

to

```print("blah {}".format(variable))

kizniche, 2018-11-07:
Here is the modified code for python 3 compatibility:

```python
    while True:
        print("Sensitivity: {:d}".format(sensor.mtreg))
        for measurefunc, name in [(sensor.measure_low_res, "Low Res "),
                                  (sensor.measure_high_res, "HighRes "),
                                  (sensor.measure_high_res2, "HighRes2")]:
            print("{} Light Level : {:3.2f} lx".format(name, measurefunc()))
        print("--------")
        sensor.set_sensitivity((sensor.mtreg + 10) % 255)
        time.sleep(1)
```

krillik, 2018-11-07:
Ok after changing the code to be compliant, I was able to run the script and get output. I then changed what you asked: 

>  smbus to smbus2 and smbus.SMBus() to SMBus(), from the following lines:

The script runs with the changes to smbus and gives me the lux output.

kizniche, 2018-11-08:
I think I found the issue, from a similar issue in #560 

Change this line:

https://github.com/kizniche/Mycodo/blob/e7699f0cf408d77851c5d679f0a842a27337d44e/mycodo/inputs/bh1750.py#L94

to this:

```python
        return dict(light=float('{0:.2f}'.format(self._lux)))
```

Save, restart the daemon, and your issue should be resolved.

krillik, 2018-11-12:
I have changed line 94 to:

> `return dict(light=float('{0:.2f}'.format(self._lux)))`

but it did not change anything. I am still not able to get light readings.

kizniche, 2018-11-12:
Then I'm not sure what the issue could be. Are you able to provide me with SSH access (user pi) and a Mycodo login (admin user)? If so, I can diagnose the issue from your Pi. You can contact me privately at http://kylegabriel.com/contact

closed 2018-11-16 by krillik