kizniche/Mycodo #562: Addition: please add support for atlas-scientifc peristaltic pumps EZO-PMP

eyespoint, 2018-11-08:
## Mycodo Issue Report:
  - Specific Mycodo Version:

 6.4.5

#### Problem Description
Please list

- no support for using I2C EZO-PMP from atlas-scientific
  
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-11-08:
Can you provide any details about this pump?

Documentation, python libraries, how it operates, how it can be best integrated into Mycodo, etc.?

kizniche, 2018-11-17:
Since I've been putting all my effort into the 7.0 release, I haven't been pushing fixes for the current stable version. However, I have already diagnosed the Atlas pH Input issue and the issue has already been resolved in 7.0.

To fix the pH Input, change the following line:

https://github.com/kizniche/Mycodo/blob/e7699f0cf408d77851c5d679f0a842a27337d44e/mycodo/inputs/atlas_ph.py#L70

to this:

```python
        return dict(ion_concentration=float(self._ph))
```

Save, then restart the backend from the Config menu.

As for the EC Input, I haven't received any issue reports and have yet to diagnose that issue, though it doesn't seem like it has the same issue as the pH Input.

If you can provide me SSH access (user pi) and a Mycodo UI admin user, I could perform some tests to diagnose and hopefully fix your EC issue. You can contact me privately at http://kylegabriel.com/contact

kizniche, 2018-11-17:
Also, that Atlas I2C code is already implemented in Mycodo to communicate with Atlas devices, so integrating the pump should be relatively easy. I would just have to figure out how it would be best integrated. When I asked for your opinion of this, I was more asking how you would be using the pump as it relates to Mycodo (i.e. where you would like to see it in the UI, how it could be configured, where it would get its signal from (10 ml), etc.). I am leaning toward it being another type of output, which could be controlled by PIDs or other Functions.

eyespoint, 2018-11-22:
Thanks,
I have sent you the login details. After more reading about the EC EZO circuit it was necessary to specify the probe type (K 1.0 in my case) and the EC readings started to show up immediately in mycodo. The only thing that I can't get working is the EZO-PMP from Atlas. I am available for testing when you have the chance. 
Regarding how the pump will be used how you described it is exactly the use case scenario. It should be in the outputs where it can be controlled by the functions. I am using the pumps to set a certain Ph and EC values by either adding more water up to a certain level to dilute or add chemicals to adjust Ph/EC.
The signal on how much to dispense will come from a calculation after I calibration of the tank. More or less I know how much ml/L needs to be added to get the right dose. What I would like to try is to get this to be done automatically via PIDs. It can be done in iterative steps as well. Smallest unit  dispensed by the pump is 0.5ml. So I could start adding  0,5ml by 0.5ml and measure in between the additions until desired values are reached. Hope this helps you. Please let me know how I can help with getting the pump working. I am thinking to go route of a script that gets executed as well. What do you think would be the best approach?

kizniche, 2018-11-22:
I was going to diagnose the EC issue, but since that's resolved, I don't have a need to remotely connect. I'll see about adding the pump as an output in the 7.0 branch and then return to see if you can test it.

kizniche, 2018-11-22:
The pump should now be working in the 7.0 branch, if you want to install and test it.

eyespoint, 2018-11-22:
I will install today and will let you know how it went

kizniche, 2018-11-23:
This is how I would install the 7.0 branch if you already have a previous version running:

```sudo service mycodo stop```
```sudo service mycodoflask stop```
```sudo mv ~/Mycodo ~/Mycodo-backup```
```cd```
```git clone github.com/kizniche/Mycodo```
```cd ~/Mycodo```
```git checkout major_release_v7.0```
```sudo /bin/bash ~/Mycodo/install/setup.sh```

eyespoint, 2018-11-23:
Thanks a lot for the info. That worked well. Did the install, now it shows
as ver. 7.0 and i can see the ezo-pmp in outputs and set it up. However it
does not work properly yet. I do not have option to activate the output and
I'm not sure about 10ml/min setting. The pump has several modes of
operation it can be set to, like constant dispensing mode. I believe that
is what you were refering to with the 10ml/min setting. In my case I do not
want it to constantly dispense but conditionaly on ph and ec values. Also I
will be using 3 pumps via I2C to dispense different liquids. The button
where I can put the ml value and dispense  the volume on the bottom of
settings does not work properly. Pump starts and stops immediately and
disregards the value. Also, field for value should accept the - value
(example -6) that would run the pump in opposite direction. At the moment
if i enter negative value the system throws an error. Pump can also be
calibrated etc...
Remote login is still valid if you need to dial in. Let me know what I can
try to troubleshoot further.

kizniche, 2018-11-23:
The current code will calculate how long it needs to run to dispense a particular volume of fluid based on the set flow rate (ml/min). You can review the calculation in the following lines of the output controller:

https://github.com/kizniche/Mycodo/blob/ea64043559e97109b41923e9aab5a8db199b7cf9/mycodo/controller_output.py#L197-L220

The command uses the Dose over time ("D,[ml],[min]", p. 57 of datasheet), however since I don't have the pump, I cannot test whether a decimal can be used instead of a integer (for dispensing for less than 1 minute). This is where I need you to attempt to dispense an amount (expand the output with the plus icon, then enter an amount in the "Amount to Dispense" box, then select "Pump Volume". Now, go to the log page (Config -> Mycodo Logs) and select Daemon, then copy any errors you see.

kizniche, 2018-11-23:
If the Dose over time cannot accept a decimal input, then I will have to modify the code to only use Volume dispensing ("D,[ml]", p. 56 of the datasheet).

kizniche, 2018-11-23:
At the moment, I cannot remotely log in, but I can review the errors you find and try to fix the code. I may be able to remote connect later today or tomorrow.

kizniche, 2018-11-28:
I tried to connect to your system a few days ago but it said it was offline. I planned to implement the rate-set and rate-unset methods for dispensing, though I need to get at least one method working before I attempt to add another.

kizniche, 2018-11-28:
One issue I'll have when connecting remotely is I won't know if the pump is actually dispensing unless you set up a camera and have two vessels I can move fluid back and forth (preferably colored liquid).

eyespoint, 2018-11-30:
> I just updated the EZO-PMP code to be more sane and to include debugging
> lines that will show up in the Daemon long whenever the pump is supposed to
> be actuated so we can see the exact commands being given to it. If you are
> updating your code with cd ~/Mycodo && git fetch && git pull you will
> most likely need to delete your database because there have been schema
> changes that will cause errors. You can fix this by recreating your
> database with sudo rm -rf ~/Mycodo/databases/mycodo.db && sudo servie
> mycodoflask restart then load the web page to create a new admin user.
> After doing that (must refresh the web page to generate the new database),
> then restart the daemon with sudo service mycodo restart and you can then
> add the pump as an output and test.

Thanks Kyle,

I have tried to do as you instructed but I got a torrent of error along the
way. I will try to be as detailed as possible to help you troubershoot with
me. Here we go.
I get an error below when I run the first command to fetch and pull

error: cannot open .git/FETCH_HEAD: Permission denied

To resolve it i went chmod 777 on all of Mycodo folder recursively

Tried again and got an error below on the pull

Please commit your changes or stash them before you merge

This was resolved by command

git reset -- hard

Fetch and pull got the new files after this without any errors.
After that I ran the second command to delete and recreate the db. I got
the following msg

Warning: mycodoflask.service changed on disk Run 'systemctl daemon- reload'
to reload units

I run the 'systemctl daemon- reload'  but a raspbian gui comes prompting
for pi user password. I punched it in but the command timed out.

This was resolved by reboot of raspberry pi. After restart i ran the
command again and it was a success.  I logged in and created new user and
logged in to web gui.

I ran the third command to restart mycodo service and was greeted by below
error

Job for mycodo.service failed because the control process exited with error
code .
See "systemctl status mycodo.service" and "journalctl _xe" for details

When I run the  "systemctl status mycodo.service"   I get the output below

pi@Aquarium1:~ $ systemctl status mycodo.service
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor
prese
   Active: failed (Result: exit-code) since Fri 2018-11-30 15:24:05 MST;
4min 4s
  Process: 3012 ExecStart=/var/mycodo-root/env/bin/python
/var/mycodo-root/mycod

Nov 30 15:24:05 Aquarium1 python[3012]:   File
"/var/mycodo-root/mycodo/mycodo_d
Nov 30 15:24:05 Aquarium1 python[3012]:     from
mycodo.controller_conditional i
Nov 30 15:24:05 Aquarium1 python[3012]:   File
"/var/mycodo-root/mycodo/controll
Nov 30 15:24:05 Aquarium1 python[3012]:     exec each_module
Nov 30 15:24:05 Aquarium1 python[3012]:                    ^
Nov 30 15:24:05 Aquarium1 python[3012]: SyntaxError: Missing parentheses in
call
Nov 30 15:24:05 Aquarium1 systemd[1]: mycodo.service: Control process
exited, co
Nov 30 15:24:05 Aquarium1 systemd[1]: Failed to start Mycodo server.
Nov 30 15:24:05 Aquarium1 systemd[1]: mycodo.service: Unit entered failed
state.
Nov 30 15:24:05 Aquarium1 systemd[1]: mycodo.service: Failed with result
'exit-c
lines 1-15/15 (END)...skipping...
● mycodo.service - Mycodo server
[... 126 lines cut ...]

×

   - Error: Add Output: Could not connect to Daemon: Connection refused. Is
   the daemon running?

From what I see it all points to the file below  file missing some
parentheses in call to exec
I'm not python expert so not sure how to troubleshoot from here. Please
help!

File "/var/mycodo-root/mycodo/controller_conditional.py", line 280
Nov 30 15:41:34 Aquarium1 python[1748]:     exec each_module
Nov 30 15:41:34 Aquarium1 python[1748]:                    ^
Nov 30 15:41:34 Aquarium1 python[1748]: SyntaxError: Missing parentheses in
call to 'exec

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
11/30/18,
5:50:52 PM

eyespoint, 2018-11-30:
> Try these commands:
>
> sudo service mycodo stop
> sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
> sudo service mycodo start

I have tried and the result is the same error when I try to start the
service.
Should I just try to install fresh copy and what would be the best way to
go about it?

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
11/30/18,
6:33:02 PM

kizniche, 2018-11-30:
Try this:
```sudo /bin/bash ~/Mycodo/install/setup.sh```

eyespoint, 2018-12-01:
still the same. No change. I have even deleted the Mycodo folder and run
the install from scratch with

sudo service mycodo stop
sudo service mycodoflask stop
sudo mv ~/Mycodo ~/Mycodo-backup
cd
git clone https://www.github.com/kizniche/Mycodo
cd ~/Mycodo
git checkout major_release_v7.0
sudo /bin/bash ~/Mycodo/install/setup.sh

after a new install I still have the same issue.
Any ides what else can I try to make it work?

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
11/30/18,
7:39:17 PM

eyespoint, 2018-12-01:
Daemon refused to start with the same error no matter what. So I googled around and found this article

https://stackoverflow.com/questions/25445439/what-does-syntaxerror-missing-parentheses-in-call-to-print-mean-in-python

it shows that parentheses were needed in python 3.0 versus 2.7. I edited the file  controller_conditional.py, line 280  from
exec each_module 
to
exec (each_module) 

that made daemon start again and now I can activate inputs and get reading on Ph and EC. 
 I still get an error when I try to add EZO-PMP in web gui
[image](https://user-images.githubusercontent.com/44875884/49322858-9639f600-f4e1-11e8-9ee5-3282896518f3.png)

I tried to add other type of output as well with same result.
This is from daemon log.

2018-11-30 18:52:03,615 - mycodo.daemon - INFO - Mycodo daemon v7.0.0 starting
2018-11-30 18:52:03,968 - mycodo.daemon - INFO - Anonymous statistics enabled
2018-11-30 18:52:03,992 - mycodo.daemon - INFO - Starting rpyc server
2018-11-30 18:52:04,329 - mycodo.output - INFO - Output controller activated in 100.3 ms
2018-11-30 18:52:04,830 - mycodo.daemon - INFO - All activated Conditional controllers started
2018-11-30 18:52:04,832 - mycodo.daemon - INFO - All activated Trigger controllers started
2018-11-30 18:52:05,305 - mycodo.input_fbb67378 - INFO - Activated in 363.5 ms
2018-11-30 18:52:05,591 - mycodo.input_dc4cff8f - INFO - Activated in 203.5 ms
2018-11-30 18:52:05,592 - mycodo.daemon - INFO - All activated Input controllers started
2018-11-30 18:52:05,593 - mycodo.daemon - INFO - All activated Math controllers started
2018-11-30 18:52:05,593 - mycodo.daemon - INFO - All activated PID controllers started
2018-11-30 18:52:05,593 - mycodo.daemon - INFO - All activated LCD controllers started
2018-11-30 18:52:06,094 - mycodo.daemon - INFO - Mycodo daemon started in 2.478 seconds
2018-11-30 18:52:06,097 - mycodo.daemon - INFO - 39.45 MB RAM in use
2018-11-30 18:52:18,033 - mycodo.output - ERROR - 1
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_output.py", line 826, in add_mod_output
    self.output_type[output_id] = output.output_type
AttributeError: 'Query' object has no attribute 'output_type'
2018-11-30 18:52:27,304 - mycodo.output - ERROR - 1
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_output.py", line 826, in add_mod_output
    self.output_type[output_id] = output.output_type
AttributeError: 'Query' object has no attribute 'output_type'
2018-11-30 18:52:34,679 - mycodo.output - ERROR - 1
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_output.py", line 826, in add_mod_output
    self.output_type[output_id] = output.output_type
AttributeError: 'Query' object has no attribute 'output_type'

Any steps you can recommend to try?

eyespoint, 2018-12-01:
I have troubleshooted your remote access issue. You should be able to log in remotely now.
Le me know if you have any problems connecting.

kizniche, 2018-12-01:
Have you tried rebooting after the initializations/installs? It's always good to just reboot to see if a fresh boot brings services up properly.

kizniche, 2018-12-01:
Try also disabling the input, change the data pin to another BCM pin, change the setting on the input in the web UI, save, then reactivate the input.

kizniche, 2018-12-01:
I will see about connecting my DHT22 later today to see if I have an issue acquiring measurements.

eyespoint, 2018-12-01:
Hi Kyle,

I have tried many things including reboots. This is what it had boiled down
to.

https://github.com/kizniche/Mycodo/issues/562#issuecomment-443389606

Please let me know if you have any suggestions on how to continue. daemon
was not starting due to python version parenthesis missing issue. On the
link above you can se onn github I left a comment for the issue.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
12/01/18,
1:38:56 PM

eyespoint, 2018-12-01:
> Try also disabling the input, change the data pin to another BCM pin, change the setting on the input in the web UI, save, then reactivate the input.

One question,  how can I change I2C to another PIN? I use atlas circuits +T3 on I2C. I was not aware I can change I2C pins on Pi to other. Please advise.

kizniche, 2018-12-01:
It appears I've bene replying to the wrong issue. My mistake.

> it shows that parentheses were needed in python 3.0 versus 2.7. I edited the file controller_conditional.py

You are correct, I did not test this when I implemented it. I'll change it.

Sorry for the confusion.

kizniche, 2018-12-01:
> AttributeError: 'Query' object has no attribute 'output_type'

I believe the database schema has been updated and you're using an old database. Please follow these instructions to build a new database and test:

```
sudo service mycodo stop
sudo rm -rf ~/Mycodo/databases/mycodo.db
sudo service mycodoflask restart
```

Reload the webpage and you should see the "create Admin User" page.

```sudo service mycodo start```

Create an admin user, add the pump as an output, and try to test.

eyespoint, 2018-12-01:
Ok, I will try that now and let you know.
Feel free to join me remotely, should be ok.

kizniche, 2018-12-01:
Actually, I'm now not quite sure why you're getting the error that output_type isn't in the schema. In any case, I've progressed a bit more with the VNC to connect to your system, but I'm now at a new login screen. I've emailed you the details.

kizniche, 2018-12-06:
Have you had a chance to test the new code after the fix I pushed 4 days ago?

alonrab, 2019-01-27:
I currently don't have this piece yet -was running the add on to see how it work - will follow up if  I end up getting it. these are a bit expensive/limited. Working on setting up other pump set up and will share what I find out

kizniche, 2019-01-27:
Okay, thanks.

eyespoint, 2019-09-10:
Hi Kyle, I'm back at my project and have set up the EZO-PMP from Atlas again. I believe the issue is in the code referenced below, and how it calculates the command. I am available for testing and would like to sort this out. Let me know when you have the time to test. In short this is the issue I was able to pinpoint.

https://github.com/kizniche/Mycodo/blob/master/mycodo/controllers/controller_output.py

In the lines of code below you are calculating the rate of flow to be dispensed. What is desired, is the ability to calibrate and give desired quantity only without time limit, but as fast as possible to dispense the desired volume. Basically, Mycodo should execute D,10 and nothing else. The way it is set up if I enter 10ml to dispense in output Daemon logs the following command. So it multiplies it by trigger 10 and it comes to D,10.00,100 and dispenses 10ml over 100min. Even when trigger field is adjusted to 1 it will generate D,10.00, 10 *10ml over 10min) This is too slow when you need to inject a certain volume and do it as fast as possible.

2019-09-09 17:07:44,640 - ERROR - mycodo.controllers.controller_output - EZO-PMP command: **D,10.00,100.00**

Code snipet that generates this is below:

# Atlas EZP-PMP
        if self.output_type[output_id] == 'atlas_ezo_pmp':
            **volume_ml = duration**
            if state == 'on' and volume_ml > 0:
                # Calculate command, given flow rate
                **minutes_to_run = self.output_flow_rate[output_id] * volume_ml**

                **write_cmd = 'D,{ml:.2f},{min:.2f}'.format(
                        ml=volume_ml, min=minutes_to_run)
                self.logger.error("EZO-PMP command: {}".format(write_cmd))**

                self.atlas_command[output_id].write(write_cmd)

                write_db = threading.Thread(
                    target=write_influxdb_value,
                    args=(self.output_unique_id[output_id],
                          'ml',
                          volume_ml,),
                    kwargs={'measure': 'volume',
                            'channel': 0})
                write_db.start()
            elif state == 'off' or volume_ml == 0:
                write_cmd = 'X'
                self.logger.error("EZO-PMP command: {}".format(write_cmd))
                self.atlas_command[output_id].write(write_cmd)
            else:
                self.logger.error(
                    "Invalid parameters: ID: {id}, "
                    "State: {state}, "
                    "volume: {vol}".format(
                        id=output_id, state=state, vol=volume_ml))
[image]

I hope I have managed to explain the desired behavior vs the way it is working now by extending the time. Please let me know when you have time so we can troubleshoot this. I'm on the latest version below. 

Mycodo Version: 7.7.1
Python Version: 3.7.3 (default, Apr 3 2019, 05:39:12) [GCC 8.2.0]
Database Version: 65271370a3a9
Daemon Status: Running
Daemon Process ID: 988
Daemon RAM Usage: 48.444 MB
Daemon Virtualenv: Yes
Frontend RAM Usage: 52.54 MB
Frontend Virtualenv: Yes

kizniche, 2019-09-15:
I refactored the Output measurement system to operate as the Input/Math controllers do, with the ability to store multiple measurements per device (until now only one measurement per Output was feasible). Now I am able to add the amount of time that the pump runs for (in minutes) as a stored measurement. This way you can know how long the pump ran for the volume of fluid dispensed. This may also help with Conditional programming if you needed to incorporate the flow rate into your conditional statement.

[image]

This code hasn't been released yet, but can be obtained by [upgrading to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch). Keep in mind that this was a pretty big change and I may still be making fixes to issues I find, until it's stable enough for release. What that means is if you upgrade to master, it may mess up your install from being able to happily upgrade to the next official release, and you may have to delete your database to generate a new one that has the correct schema. If you do decide to try the pre-release, after the upgrade to master, you will have to delete any current EZO Pump Outputs and add them again for this new measurement to appear.

kizniche, 2019-09-15:
More fixes being committed. Now Output measurements can be used in measurement widgets.

[image]

kizniche, 2019-09-15:
I think I finally have all parts of the Dashboard working with the new Output measurement system.

[image]

kizniche, 2019-09-15:
If you want to see the commands being sent to the pump, you'll also need to check the Enable Daemon Debug Logging option in the General Settings on the Configuration page.

eyespoint, 2019-09-15:
Hi Kyle,

I have tested the update before last, and this is what I have found. The math now works as you stated. The max flow rate of the pump is 105ml/min according to documentation, and the math from photo below generates the below in the Daemon log

[image]

2019-09-13 19:37:25,130 - ERROR - mycodo.controllers.controller_output - EZO-PMP command: D,17.50,0.17

What I have tried, as you suggested, is use the pumps max flow rate of 105ml/min to dispense 17.5ml of liquid. The command generated by the math *D,17.50,0.17)and it does not turn on pump at all. It should in theory, run the motor for 10.2s at max flow rate, and dispense the exact amount. 
Thought it was most likely a formatting issue, with the dots and the comas in the command generated and whatnot and checked in Atlas scientific sample code. It generated Error 2 when I enter the command. I have played around with numbers and have observed that pump works differently in this mode, eg. it works in short pulses vs one continuous run, and the effective max flow rate is limited to 50ml/min. (please see below for rate) issues with error 2 has to do with overstepping 50ml/min flow limit, thus creating the Error 2.

[image]

While I understand your argument about broader use case, maybe we should take a step back and add one mode of the pump at the time. Dispensing certain volume and calibration  feature in Mycodo would be awesome and highly appreciated. Im here to test anything you need. This pump is mostly useful to dose exact small amounts of liquid in 0.5ml granularity accurately after calibration.

eyespoint, 2019-09-15:
[image]

eyespoint, 2019-09-15:
This is the limit of the pump flow rate of 25ml in 30sec in this mode.

kizniche, 2019-09-16:
I read the flow rate is limited by the calibration. Perhaps you need to calibrate your pump before it will work properly in the flow rate mode. In any case, I will work on adding an option in the output settings to select either "Fastest Dispense" or "Select Flow Rate".

kizniche, 2019-09-16:
Perhaps it's a decimal issue. Try truncating the float value to "0.1" rather than "0.17"

kizniche, 2019-09-16:
Working on the calibration method.

kizniche, 2019-09-16:
Just pushed a commit that adds the ability to calibrate the pump. Let me know if it works as expected. I'll try to work on the output option to select what mode to run the pump in soon.

kizniche, 2019-09-16:
Well, I didn't think it would be so easy to add modes to the pump output settings, but it looks like it's now working, and you can now select "Fastest Flow Rate" or "Select low Rate".

eyespoint, 2019-09-30:
Hi Kyle,

I have just tested the latest build 7.7.9 with EZO-PMP and the max flow rate is now working as it should as well as the dispensing over time. Thanks for fixing that. I am trying to use the pump as output in PID but I am having issues configuring it. The PID setting allows only a number of seconds for output to be turned on, not the volume in ml to be dispensed, which would be the desired output mangement. At this time the pump never turns on to dispense. My PID setting below and daemon log as well. The amounts dispensed by pump, are me manually dispensing the amounts.
Regarding the decimal points when in "max flow rate" the pump will only dispense increments of 0.5ml and nothing else. (0.5, 1, 1.5, 2, 2.5............) and I have tested this. I do not see PID generating any commands for the pump. Calibration works good as well.
Please help me set up the PID to dispense ml from the ezo-pmp.

PID settings below

[image]

Daemon output below

2019-09-29 23:56:47,818 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.178 @ 2019-09-29 23:56:40
2019-09-29 23:56:47,911 - INFO - mycodo.utils.pid_autotune - 
2019-09-29 23:56:47,912 - INFO - mycodo.utils.pid_autotune - Cycle: 12
2019-09-29 23:56:47,912 - INFO - mycodo.utils.pid_autotune - found peak: 6.178
2019-09-29 23:56:47,912 - INFO - mycodo.utils.pid_autotune - peak count: 2
2019-09-29 23:57:17,829 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.173 @ 2019-09-29 23:57:10
2019-09-29 23:57:47,851 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.172 @ 2019-09-29 23:57:40
2019-09-29 23:57:47,949 - INFO - mycodo.utils.pid_autotune - 
2019-09-29 23:57:47,949 - INFO - mycodo.utils.pid_autotune - Cycle: 14
2019-09-29 23:57:47,949 - INFO - mycodo.utils.pid_autotune - found peak: 6.172
2019-09-29 23:57:47,950 - INFO - mycodo.utils.pid_autotune - peak count: 3
2019-09-29 23:58:17,894 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.17 @ 2019-09-29 23:58:10
2019-09-29 23:58:47,844 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.173 @ 2019-09-29 23:58:40
2019-09-29 23:59:17,861 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.182 @ 2019-09-29 23:59:10
2019-09-29 23:59:17,979 - INFO - mycodo.utils.pid_autotune - 
2019-09-29 23:59:17,979 - INFO - mycodo.utils.pid_autotune - Cycle: 17
2019-09-29 23:59:17,980 - INFO - mycodo.utils.pid_autotune - found peak: 6.182
2019-09-29 23:59:17,980 - INFO - mycodo.utils.pid_autotune - peak count: 4
2019-09-29 23:59:47,795 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.177 @ 2019-09-29 23:59:40
2019-09-30 00:00:17,899 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.194 @ 2019-09-30 00:00:10
2019-09-30 00:00:47,829 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.181 @ 2019-09-30 00:00:40
2019-09-30 00:01:17,874 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.202 @ 2019-09-30 00:01:10
2019-09-30 00:01:17,959 - INFO - mycodo.utils.pid_autotune - 
2019-09-30 00:01:17,960 - INFO - mycodo.utils.pid_autotune - Cycle: 21
2019-09-30 00:01:17,960 - INFO - mycodo.utils.pid_autotune - switched state: relay step down
2019-09-30 00:01:17,960 - INFO - mycodo.utils.pid_autotune - input: 6.202
2019-09-30 00:01:47,883 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.201 @ 2019-09-30 00:01:40
2019-09-30 00:02:17,834 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.185 @ 2019-09-30 00:02:10
2019-09-30 00:02:47,865 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.184 @ 2019-09-30 00:02:40
2019-09-30 00:02:47,955 - INFO - mycodo.utils.pid_autotune - 
2019-09-30 00:02:47,955 - INFO - mycodo.utils.pid_autotune - Cycle: 24
2019-09-30 00:02:47,955 - INFO - mycodo.utils.pid_autotune - found peak: 6.184
2019-09-30 00:02:47,956 - INFO - mycodo.utils.pid_autotune - peak count: 5
2019-09-30 00:02:47,956 - INFO - mycodo.utils.pid_autotune - 
2019-09-30 00:02:47,956 - INFO - mycodo.utils.pid_autotune - amplitude: 0.006000000000000079
2019-09-30 00:02:47,956 - INFO - mycodo.utils.pid_autotune - amplitude deviation: 0.9999999999999755
2019-09-30 00:03:17,869 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.203 @ 2019-09-30 00:03:10
2019-09-30 00:03:47,875 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.199 @ 2019-09-30 00:03:40
2019-09-30 00:04:17,900 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.205 @ 2019-09-30 00:04:10
2019-09-30 00:04:18,005 - INFO - mycodo.utils.pid_autotune - 
2019-09-30 00:04:18,005 - INFO - mycodo.utils.pid_autotune - Cycle: 27
2019-09-30 00:04:18,006 - INFO - mycodo.utils.pid_autotune - found peak: 6.205
2019-09-30 00:04:18,006 - INFO - mycodo.utils.pid_autotune - peak count: 6
2019-09-30 00:04:18,006 - INFO - mycodo.utils.pid_autotune - 
2019-09-30 00:04:18,006 - INFO - mycodo.utils.pid_autotune - amplitude: 0.0030000000000001137
2019-09-30 00:04:18,007 - INFO - mycodo.utils.pid_autotune - amplitude deviation: 1.0
2019-09-30 00:04:47,831 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.204 @ 2019-09-30 00:04:40
[... cut ...]
2019-09-30 12:26:59,215 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.2 @ 2019-09-30 12:26:55
2019-09-30 12:27:29,167 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.194 @ 2019-09-30 12:27:25
2019-09-30 12:27:59,196 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.192 @ 2019-09-30 12:27:55
2019-09-30 12:28:29,118 - DEBUG - mycodo.controllers.controller_pid_03a2de37 - Latest (CH0, Unit: pH): 6.204 @ 2019-09-30 12:28:2

kizniche, 2019-10-01:
For testing, use the most simple set of options. Disable setpoint tracking (method), do not use autotune, set the setpoint significantly higher (2+ pH) than the current pH, and, of course, just have water in the pump reservoir rather than an actual acid/base, and have it pump into a waste reservoir (or back into the holding reservoir), so as not to actually affect the system.

Let me investigate the PID pump output issue and I will get back to you.

kizniche, 2019-10-02:
I pushed the fix, so you can upgrade to master to test the new code.

kizniche, 2019-10-07:
If you have any further issues, please open a new issue. This one has been complete with the addition of the output.

closed 2019-10-07 by kizniche