kizniche/Mycodo #563: Suggestion: adding support for Energenie USB and LAN power switches EG-PM

pewi81, 2018-11-14:
Hi!
Perhaps Mycodo already support this relay switch or similar ones!
These power relays are connected either by WiFi or USB and requires no skill in electrical wiring which is the main reason I would like buy them!

Regards
Pete

kizniche, 2018-11-30:
If you would like to purchase one to test, I will work with you to add support in Mycodo.

kizniche, 2018-11-30:
If you would like me to review documentation for an item you are interested in purchasing, please make a post with the item, datasheet, code, etc. and I'll take a look before you purchase to give you an idea if it will be possible/feasible to integrate into Mycodo.

pewi81, 2018-11-30:
I will see what I can do!

pmunz75, 2019-01-06:
Hello,

INFO Links : 
https://github.com/perryflynn/energenie-connect0r
https://github.com/asig/energenie
https://github.com/e-cite/eg-pms-2-lan

PDF:
[A1PbPxqy6OS.pdf](https://github.com/kizniche/Mycodo/files/2730490/A1PbPxqy6OS.pdf)
[Uploading B1ku4G1V03S.pdf…]()

kizniche, 2019-01-08:
I don't think integrating support into Mycodo is warranted, since this device is network-enabled, and the tools for communicating with it are already available. It seems the easiest solution to getting this working is to merely add 4 "On/Off (Linux Command)" Outputs, then configure the on and off commands using on as ```~/energenie-connect0r/geniecli.py http://192.168.2.5 passw0rd --on 2``` and off as ```~/energenie-connect0r/geniecli.py http://192.168.2.5 passw0rd --off 2```. Let me know what you think.

closed 2019-10-07 by kizniche