kizniche/Mycodo #564: PH Circuit no Data over i2c

HugoPenner, 2018-11-14:
Hi,

first of all Thanks for all your hard work on this project.

## Mycodo Issue Report:

  - Mycodo Version: 6.4.5
  - Pi3 B+
  - Atlas EZO pH Circuit
  - Atlas Electrically Isolated EZO Carrier Board
  - i2c and GPIO pins 3 ( SDA.1),5 (SCL.1)
 
#### Problem Description
Please list:

After a fresh installation of MyCodo i added the Atlas EZO Ph Circuit in the Input Tab.

[image]

But i do net get any reading on the Live tab.
[image]

Then i tried the Sample Code from atlas-scientific and it works fine.

[image]

 
### Errors

i think i dont have any Errors in the Daemon Log.

Thanks for any intake you have on this!

Hugo

kizniche, 2018-11-14:
The Atlas code in Mycodo is pretty difficult to debug without the sensor (I don't have the EC sensor). Could you provide me SSH access (user pi) and an admin user for the Mycodo UI? If so, I could probably find the issue fairly quickly. You can contact me at http://kylegabriel.com/contact

HugoPenner, 2018-11-14:
I send you an PM to http://kylegabriel.com/contact

closed 2018-11-16 by kizniche