kizniche/Mycodo #565: Mycodo 5.7.3 cannot update

alonrab, 2018-11-15:
## Mycodo Issue Report:
  Version 5.7.3
 Hi All, I cannot update to new version 
Thanks much!

  trying to update
### Errors
 [2018-06-20 12:05:39] 
[2018-06-20 12:05:39] 
[2018-06-20 12:05:39] Installed version: 5.7.3
[2018-06-20 12:05:39] Latest version: usage: github_release_info.py [-h] [-c] [-i] [-l] [-m MAJORNUMBER] [-v]
[2018-06-20 12:05:39] github_release_info.py: error: argument -m/--majornumber: expected one argument
[2018-06-20 12:05:39] 
[2018-06-20 12:05:39] #### Upgrade to vusage: github_release_info.py [-h] [-c] [-i] [-l] [-m MAJORNUMBER] [-v]
[2018-06-20 12:05:39] github_release_info.py: error: argument -m/--majornumber: expected one argument initiated 2018-06-20_12-05-35 ####
[2018-06-20 12:05:39] 
[2018-06-20 12:05:39] #### Beginning Upgrade: Stage 1 of 2 ####
[2018-06-20 12:05:55] Stopping the Mycodo daemon...Done.
[2018-06-20 12:05:55] Downloading latest Mycodo version to /home/pi/mycodo-usage: github_release_info.py [-h] [-c] [-i] [-l] [-m MAJORNUMBER] [-v]
[2018-06-20 12:05:55] github_release_info.py: error: argument -m/--majornumber: expected one argument.tar.gz...Failed: Error while trying to wget new version.
[2018-06-20 12:05:55] File requested: usage: github_release_info.py [-h] [-c] [-i] [-l] [-m MAJORNUMBER] [-v]
[2018-06-20 12:05:55] github_release_info.py: error: argument -m/--majornumber: expected one argument -> /home/pi/mycodo-usage: github_release_info.py [-h] [-c] [-i] [-l] [-m MAJORNUMBER] [-v]
[2018-06-20 12:05:55] github_release_info.py: error: argument -m/--majornumber: expected one argument.tar.gz
[2018-06-20 12:05:55] 
[2018-06-20 12:05:55] 
[2018-06-20 12:05:55] #### ERROR ####
[2018-06-20 12:05:55] There was an error detected during the upgrade. Please review the log at /var/log/mycodo/mycodoupgrade.log

kizniche, 2018-11-15:
Not sure what the issue is, it should have given you a message stating upgrading to the next major revision number is not possible. Versions 5.x cannot be upgraded to versions 6.x. You will have to perform a fresh install.

closed 2018-11-15 by kizniche

alonrab, 2018-11-15:
That make sense, Thanks!