kizniche/Mycodo #566: Method not updating Setpoint

mrplow, 2018-11-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.5
 
#### Problem Description
- My room heater function does work. I created a method to drop the temperature to 14C during the day and 18C during the night, however, the setpoint never changes on it's own and I have to manually change it.

 
### Errors
  - no messages or errors in daemon log.

### Additional Notes
Since 7.0.0 is being worked on and, I assume, that a lot of this code is going to be rewritten this isn't a high priority issue. I'm hoping this is a misconfiguration issue.

[image]
[image]

kizniche, 2018-11-17:
It does look like an issue with Mycodo and not a misconfiguration. I'll try to replicate it on the 7.0 branch.

mrplow, 2018-11-19:
Thanks @kizniche 
I manually merged these changes into my 6.4.5 install and my method is now working properly

drgrumpy, 2019-02-13:
@mrplow please can you tell me precisely what you merged, I have a similar issue with 6.4.5, and I don't want to risk upgrading....

I have a lighting PID and a daily method to set the setpoint from 06:00 to 18:00, but it seems that it is being ignored.

mrplow, 2019-02-13:
@drgrumpy see the changelog here 6d361e0
Make backups of the files first 😎

drgrumpy, 2019-02-14:
Thanks @mrplow. I suppose what I was really interested to know was did you just update the single line in controller_pid.py ? Or overwrite the whole file ?

The other stuff in that changelog is all about web display I think.

mrplow, 2019-02-15:
I just manually merged ALL the changes from that commit (except the readme.)
Good luck

kizniche, 2019-02-16:
It looks like that one indentation change of controller_pid.py was all that was needed to fix this issue. The rest of the edits were for button styles.

drgrumpy, 2019-02-16:
Thanks. That's what I thought. I just changed the indent on that one line.