kizniche/Mycodo #567: v7.0 Dashboard - only first graph configure displays

Theoi-Meteoroi, 2018-11-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.0
 
#### Problem Description
Please list:  

1.  Configure inputs 
2.  Configure Graph output in dashboard for configured inputs. Configure at least two or three.
3. Only the first graph renders.  Subsequent graphs do not render in the dashboard view.

### Errors

No reported errors seen or noticed.

unable to print missing graph objects as there is no context menu.

### Steps to Reproduce the issue:
How can this issue be reproduced?

Follow the list of actions under "problem description"

### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-11-18:
I wasn't able to replicate this issue. What commit are you running from the 7.0 branch?

Theoi-Meteoroi, 2018-11-18:
I didn't record which commit -  I pulled it and installed it yesterday.

Anything I can do to help diagnose?

I'll pull the latest commit where you fixed #568 to test that fix.

kizniche, 2018-11-18:
I'd have to look at the source code of the Dashboard page and see what isn't appearing in the Javascript that is causing issues with the graphs. It would be difficult to explain my diagnostic steps, since I can merely look at the source and identify the issue by recognizing what doesn't appear in the source vs. what should. Otherwise, I think we may go back and forth a lot before solving the issue.
If you could provide SSH (user pi) and a Mycodo admin user I could probably rapidly identify the issue.

kizniche, 2018-11-19:
oh, I forgot to mention that the latest commit changed the database schema and creating a new database would be necessary.

kizniche, 2018-11-19:
You can merely delete ~/Mycodo/databases/mycodo.db, restart the frontend (```sudo service mycodoflask restart```), then reload the browser page.

kizniche, 2018-11-19:
I would also restart the backend after the database has been recreated (before adding any inputs) with ```sudo service mycodo restart```

Theoi-Meteoroi, 2018-11-20:
Chose complete re-install with latest commit.  All graphs are generated now, as long as there are no conversions in the data sources.  When testing with AM2315 - enabled conversion of temp to F - no error.  Then also enabled conversion for dew-point to F - I get this stack in the Dashboard -

Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 584, in page_dashboard
    y_axes = utils_dashboard.graph_y_axes(dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 472, in graph_y_axes
    elif y_axes[each_graph.unique_id] and each_measure.convert_unit_to not in y_axes[each_graph.unique_id]:
AttributeError: 'DeviceMeasurements' object has no attribute 'convert_unit_to'

kizniche, 2018-11-20:
The latest commit (referenced above) should fix this issue.

kizniche, 2018-11-20:
Just pushed a commit that fixes the same issue for asynchronous graphs.

closed 2018-11-30 by kizniche