kizniche/Mycodo #568: v7.0 Async graph page seems completely broken (missing code changes)

Theoi-Meteoroi, 2018-11-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:   v7.0
 
#### Problem Description

Asynchronous Graph page has missing code variables:

AttributeError: 'InputMeasurements' object has no attribute 'converted_unit'

Attempted to generate async data graph 
 
### Errors
 Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 754, in page_graph_async
    selected_ids_measures)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 586, in graph_y_axes_async
    if each_measure.converted_unit:
AttributeError: 'InputMeasurements' object has no attribute 'converted_unit'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Configure inputs - configuring with or without unit conversion
 2. Collect data for some period of time ( so there is data to display )
 3. Go to Async graph page and attempt to generate a graph of available data.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-11-18:
oops, I see another issue. Working no a fix.

kizniche, 2018-11-18:
Scratch that, I don't know what I was thinking.

Theoi-Meteoroi, 2018-11-20:
Chose complete re-install.

Tested Async graph function with latest commit - it worked fine with the inputs tested earlier that didn't work.  Thanks Kyle!

Theoi-Meteoroi, 2018-11-20:
Closing as fixed for v7.0

closed 2018-11-20 by Theoi-Meteoroi