kizniche/Mycodo #569: Sensors with external dependencies require 'import sys' for path to env/bin

Theoi-Meteoroi, 2018-11-20:
## Mycodo Issue Report:
  - Specific Mycodo Version: v7.0
 
#### Problem Description
Sensors with external module dependencies  (example MCP9808) need to have 

"import sys"

Added to the python module in "mycodo/inputs" otherwise the module in env/src will not be found at runtime.

 
### Errors

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 529, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 173, in __init__
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/mcp9808.py", line 67, in __init__
    from Adafruit_MCP9808 import MCP9808
ImportError: No module named 'Adafruit_MCP9808'

### Steps to Reproduce the issue:
Add a sensor with external sensor driver requirement
 
### Additional Notes

kizniche, 2018-11-20:
This is strange, because I don't experience this issue. I noticed you write about this in the past (or perhaps someone else).

kizniche, 2018-11-20:
I believe the issue is the daemon needs to be restarted after a dependency is installed and you may have done this in an attempt to fix the issue.

kizniche, 2018-11-20:
If this is indeed the issue (not restarting the daemon after installing a module), it may be trickier than I anticipated to solve if the desired solution involves refreshing the available modules without restarting the daemon. I believe I've researched a solution for this in the past without success.

kizniche, 2018-11-30:
I wasn't able to find a way to fix this issue without restarting the daemon, so I merely added a daemon restart to the end of the dependency install process.

closed 2018-11-30 by kizniche