kizniche/Mycodo #570: RTD Probe showing 715℃ at room temperature

electricspaceman, 2018-11-23:
## Mycodo Issue Report:
  - Specific Mycodo Version:

Version 6.3.9
 
#### Problem Description

  -  Cannot properly read temperature with [RTD100 probe 4 wire](http://www.farnell.com/datasheets/1918942.pdf?_ga=2.237103938.378747944.1542933915-211189162.1522673430) using [MAX31865](https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/python-circuitpython) and Raspberry Pi 3.
  - Wired it all up, and set up in Mycodo without issue.
[image]
[image]
[image]

**PINOUTS:**
5V -> Vin
Ground -> GND
BCM 10 MOSI -> SDI
BCM 9 MISO -> SDO
BCM 11 SCLK -> CLK
BCM 23 -> CS

Can't figure this out! Have I got faulty hardware??

Any help would be greatly appreciated.

kizniche, 2018-11-23:
This is the code I'm using: https://github.com/steve71/MAX31865

Can you test whether that works outside of Mycodo?

```cd```
```git clone github.com/steve71/MAX31865```
```cd MAX31865```

Edit max31865.py and change these pins to reflect your connections:

https://github.com/steve71/MAX31865/blob/9f30bb383c3fd9df7650a39aa892ed03f770d07a/max31865.py#L210-L213

Then run the code using Mycodo's python3:

```~/Mycodo/env/bin/python ./max31865.py```

kizniche, 2018-11-23:
Also, I see you're running v6.3.9, which is not the latest stable version (6.4.5). I would always recommend upgrading the latest stable version because your issue may have already been resolved. Upgrade, and if you're still experiencing the same issue, run the code in comment https://github.com/kizniche/Mycodo/issues/570#issuecomment-441255663 and report back. Thanks.

kizniche, 2018-11-23:
Mycodo only runs in python3, so if you could provide the error(s) from running in the python3 environment, that will help me port it from py2 -> py3.

kizniche, 2018-11-23:
Also, make sure you're using the proper reference resistor value. This can throw off the measurement if it's incorrect.

electricspaceman, 2018-11-26:
> Mycodo only runs in python3, so if you could provide the error(s) from running in the python3 environment, that will help me port it from py2 -> py3.

I see that there are 8 pre-existing forks, I wonder perhaps if one of them is Python 3 compliant. I will check it out. Quite new to this sort of thing. :)

> Also, make sure you're using the proper reference resistor value. This can throw off the measurement if it's incorrect.

So double and triple checked the hardware and I'm certain that I've not wired things incorrectly. Breakout board that I'm using advises to use 430ohm as the base/reference resistance, with that it reads 715C. Playing around I found:

_Base resistance |Temperature_
600 | 1209
430 | 715
215 | 198
180 | 120
140 | 35.75
130 | 14.78
125 | 4.34
120 | 577.09
107.5 | 578.81
53.75 | 578.78

Have you ever seen anything like this before?

It's so weird.

kizniche, 2018-11-27:
I don't have any of the MAX chips or probes to test with, so unfortunately I won't be much help debugging the issue. As for how to send the updated file, you can paste the code in a comment in this issue. If you surround the code with three backtics at the top and bottom of the code, it will preserve indentations.

Theoi-Meteoroi, 2018-11-27:
Howdy,
               I think I have one of these that I haven't put together yet.  I'll review the engineering information, build a test environment and see if I can help with the code review for this part.  

This is the vendor [Primer for MAX31865](https://www.maximintegrated.com/en/design/videos.html/vd_5775767051001#popupmodal)

Theoi-Meteoroi, 2018-11-28:
The Maxim data sheet specifies the reference resistor value as:

**_A reference resistor equal to four times the RTD’s 0 degrees C resistance is optimum for a platinum  RTD. Therefore, a PT100 uses a 400 ohm reference resistor, and a PT1000 uses a 4k ohm reference resistor._**

This is a high precision, low drift resistor value ( 430 ohms sounds like a %5 value ).  But I don't think that would introduce that much error.  Are you really using a 4-wire connection?   Which breakout are you using? Do you have a schematic of the breakout circuitry?

The error values ( lack of apparent linearity ) look to me to be more of a problem with the SPI interface.   The Max31865 tries to be clever in sensing the polarity of the clock during CS-low.  It could be noise on the power supply.  A 10-25 uf tantalum capacitor and 0.1 ceramic capacitor low-pass filter is good. Just put across the supply lines close to the device or breakout, being careful of the tantalum cap polarity.

electricspaceman, 2018-11-28:
Hello! :)

> This is a high precision, low drift resistor value ( 430 ohms sounds like a %5 value ). But I don't think that would introduce that much error. Are you really using a 4-wire connection? Which breakout are you using? 

I am using [Adafruit MAX31865](https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/python-circuitpython) and [4 wire connection](http://www.farnell.com/datasheets/1918942.pdf?_ga=2.237103938.378747944.1542933915-211189162.1522673430).

>Do you have a schematic of the breakout circuitry?

Here is a copy of the schematic
[image].
 
> **_A reference resistor equal to four times the RTD’s 0 degrees C resistance is optimum for a platinum RTD. Therefore, a PT100 uses a 400 ohm reference resistor, and a PT1000 uses a 4k ohm reference resistor._**

I changed the Reference Resistance from 430ohm to 400ohm but no dice, and I noticed the temperature now reads 636C.

[image]

I also noticed, and I'm not sure how this is the case, there are now two options to select the clock pins. Please see the above screenshot, as well as this one...

 [image]

Perhaps the Mycodo update has meant the addition of a new an addtioanl CS field? @kizniche  what values should I have for each? I actually have two MAX31865, sorry I didn't mention this earlier but I just wanted to get one working before trying the second unit, is the second field for a second MAX31865? 
I'd like to use two so I have a humidity sensor that relies on evapouration to work (ie Wet Bulb Dry Bulb).

> The error values ( lack of apparent linearity ) look to me to be more of a problem with the SPI interface. The Max31865 tries to be clever in sensing the polarity of the clock during CS-low. It [...]

I've been trying my best to imitate Kyle's build and so I do have a 5v power supply (CUI VOF-10-5) at hand which I could use - presumably that will produce a clean/quiet electrical source. So far I've been using the onboard power supply of the Raspberrry Pi, which in turn is powered by a little 5w iPhone usb thingy. How much difference will it make, or is it a case of trying rule out possibilities? Merely curious.

Thank you both for your time, I do really appreciate some help.

kizniche, 2018-11-28:
There are some options appearing that shouldn't be (again, fixed in 7.0). I'll review the code later and let you k ow which aren't used to set the options for the module.

kizniche, 2018-11-30:
Don't worry about the second Clock Pin option. As long as you have them the same, one will be used to set the clock pin. I noticed you changed the CS pin from the default. Have you tried using the default pins?

electricspaceman, 2018-12-05:
Hi there,
So:

> The error values ( lack of apparent linearity ) look to me to be more of a problem with the SPI interface. The Max31865 tries to be clever in sensing the polarity of the clock during CS-low. It [...]

Had to order a few bits but I now have a 5v power supply (CUI VOF-10-5) powering the PI and supply a common DC supply two terminal blocks.

[image]

[image]

Sadly the powersupply has had no effect on the signal and in mycodo, it's still reading ~715C.

I am correct in suggesting that using this power supply addressing any possible noise or interference? If not, should I pursue installing the snubber circuit?

> Don't worry about the second Clock Pin option. As long as you have them set to the same pin, one will be used to set the clock pin properly. I noticed you changed the CS pin from the default. Have you tried using the default pins?

Thanks for the idea, I've been really wracking my brain. I changed it over to the default CS as you suggested and that made no difference. 

I bet it's something like really obvious. Will spend a bit more time trying different things, a tough nut to crack!

Do you know if anyone else in the mycodo-verse has a built a wet bulb dry bulb sensor for mushroom cultivation?

kizniche, 2018-12-06:
> Do you know if anyone else in the mycodo-verse has a built a wet bulb dry bulb sensor for mushroom cultivation?

I have with two DS18B20 sensors, a conical vial of deionized water, and a hydrophilic hollow cotton wick surrounding the wet-bulb thermometer that was placed in front of a continuously moving airflow in the chamber. It worked very well. There were no issues with condensation throwing off the measurements as digital sensors often experience, however you must make sure to have a microbial control regimen in place because the wick will begin to be colonized after a few weeks exposed to the environment. I never had any issue with the reservoir water so long as the water was free of contaminants and nutrients/minerals.

zsole2, 2018-12-06:
I have the wet bulb system running for about 2 months now, albeit not for mushroom, my temperature range is 5-10 degrees Celsius for cheese aging. This is good for avoiding contamination, but I would definitely agree to have distilled water and probably something against mold at higher temperatures.
 
I tested several DS18B20 sensors to make sure they give sufficiently close readings, then used a setup similar to Kyle's. Although the wick is not a specialized one that was discussed in another thread here, just a simple cotton diaper piece. Still get readings within a few % points of a control hygrometer.

In conclusion, I'll probably never go back to any humidity sensor, since I already have a big pile of failed DHT, HTU and BMP sensors, all stopped working after 3-6 months at 92-98% RH.

kizniche, 2018-12-13:
You can check out #335 for the issue developing this feature and photos of my wet-bulb setup. I essentially dipped my sensor/wire end in the [material typically used for coating tool handles that have degraded](https://www.homedepot.com/p/Plasti-Dip-14-5-oz-Black-Plasti-Dip-11603-6/202196703). For this particular application, I used [the spray version](https://www.homedepot.com/p/Plasti-Dip-11-oz-Black-Plasti-Dip-11203-6/100131010?MERCH=REC-_-PIPHorizontal2_rr-_-202196703-_-100131010-_-N).

electricspaceman, 2018-12-20:
Thank you for all your help with this,

kizniche, 2020-10-14:
Hello. Is there a reason this issue was reopened?

electricspaceman, 2020-10-14:
Hi, 

Sorry, I reopened it and then immediately had to go do something else. sorry. 

I wanted to have another go with the max31865 RTD PT1000 sensors in order to make my own super-accurate psychrometer for measuring humidity for a mushroom growing environment that I am building.

Side note: I tried using DS18B20 and with your help, I was able to get them working however the temperature readings were not accurate enough for my liking. The mushroom farm where I used to work used PT100 sensors for their psychrometer and it would bring me great satisfaction to be able to do the same.

Prior to initialising Mycodo, I connected my two adafruit max31865 breakout boards and using the tutorial on their website I was able to get norminal readings from the probes! I was extremely happy about this and thought the next best thing to do would be to fire up mycodo and see if it works there too. This is where things took a U-turn and I started getting funny readings like 800C etc.
I then, without doing anything else, decided to run my python script from the adafruit tutorial and the sensors were putting funny readings again but like -245C this time. All this while the hardware had not been touched and nothing powered down or rebooted.

This makes me wonder if the module for the max31865 is set up correctly. I'm not 100% confident with SPI but I'm guessing once Mycodo initialized that it reconfigured the way the pins are read on the GPIO for SPI (complete conjecture). The pins must've been connected correclty as the code ran correctly in my script and I quadruple checked that I had the same pins (using BCM numbering) were configured in mycodo.

Normally I would include screenshots but I'm not at work right now.

If I recall correctly you said that you aren't able to debug the max31865 as you don't have any, but I wonder if you could help by checking the code in the module and the code in the tutorial?

https://learn.adafruit.com/adafruit-max31865-rtd-pt100-amplifier/python-circuitpython

I also used the code on this page for multiple sensors, adapting it for 2 channels as the adafruit tutorial only shows how to code for one sensor and I'm not a sufficiently adept coder to synthesize code.

`import time
import board
import busio
import digitalio
import adafruit_max31865
import wiringpi

print('MAX31865 4 Channel')

#initialize

spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs1 = digitalio.DigitalInOut(board.D5)
cs2 = digitalio.DigitalInOut(board.D6)
cs3 = digitalio.DigitalInOut(board.D12)
cs4 = digitalio.DigitalInOut(board.D13)

sensor1 = adafruit_max31865.MAX31865(spi, cs1, rtd_nominal=100, ref_resistor=431.0, wires=2)
sensor2 = adafruit_max31865.MAX31865(spi, cs2, rtd_nominal=100, ref_resistor=431.0, wires=2)
sensor3 = adafruit_max31865.MAX31865(spi, cs3, rtd_nominal=100, ref_resistor=431.0, wires=2)
sensor4 = adafruit_max31865.MAX31865(spi, cs4, rtd_nominal=100, ref_resistor=431.0, wires=2)

while True:
	temp1 = sensor1.temperature
	temp2 = sensor2.temperature
	temp3 = sensor3.temperature
	temp4 = sensor4.temperature
	print('Temperature: {0:0.3f}C'.format(temp1))
	print('Temperature: {0:0.3f}C'.format(temp2))
	print('Temperature: {0:0.3f}C'.format(temp3))
	print('Temperature: {0:0.3f}C'.format(temp4))
	time.sleep(0.1)`

Many thanks for your time Kyle,

electricspaceman, 2020-10-15:
Hi Kyle, 

Back at work! I've got screenshots for you to look at. I rebooted the pi when I got in and tried to run my test script and it showed norminal readings. 
I think the problem might be due to the fact that I am using four wires which is possibly an unusual amount. Maybe the mycodo max31865 software module is configured for a different amount of wires?

[image]
[image]
[image]
[image]

electricspaceman, 2020-10-15:
I just retried 'activating' the sensors again and then trying my maxtest.py script again and this time the correct readings are showing in the terminal window which contradicts my earlier post. 

Data in mycodo is still showing -256.

[image]

electricspaceman, 2020-11-18:
Hey Kyle, 

Just wondering you'd be interested in helping me get this sensor working? I had a review of the code, my coding skills are weak! but I think I think the MAX31865 module is broken. It doesn't contain libraries that I think will make it functional and also it ommits things like a way to properly set the ref-resistance and the number of wires being used. 

I did try have a look at making my own custom input but it's an almost insurmountable challenge right now. I could do it with enough time, but perhaps with your help, it would be much less.

not5, 2020-11-30:
@electricspaceman I've been using the MAX31865 module for awhile now (over a year) in Mycodo and have only had small issues (unrelated to Mycodo, has to do with EM interference, I think) with it. What exactly is the issue you're having?

kizniche, 2020-12-02:
I think creating a new input with the adafruit library is the best way to go, however since I don't have this sensor, developing it myself may be difficult. When developing an input module, I like to be able to let the user select which pins to use, and referencing your code I notice there are variables used rather than integers for the pins. I have several modules that use Adafruit's Circuitpython, but only for I2C using Adafruit_Extended_Bus, but none with UART. If you can you provide example code that allow specific pins as integers to be used with the adafruit library, I can probably take it from there to make it into a module.

not5, 2020-12-02:
Relooking at this in further detail, you're using the wrong  pins... you're putting in Board numbering when it should be BCM numbering.

kizniche, 2020-12-14:
FYI a new MAX31865 input using the adafruit CircuitPython library was just added and will be available in the next release. You can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to test it out. (see #900)

electricspaceman, 2020-12-16:
Hey Kyle, 

Sorry that I didn't see this sooner. Thank you for time on this. Super grateful.

> @electricspaceman I've been using the MAX31865 module for awhile now (over a year) in Mycodo and have only had small issues (unrelated to Mycodo, has to do with EM interference, I think) with it. What exactly is the issue you're having?

I'm trying to add two PT1000 4-wire sensors using the adafruit MAX31865 breakout boards for a psychrometer I'm making.  It would come up with incorrect temperature readings as per the title. Yet when running a script outside of Mycodo, in the screenshot above it would work fine.

> I think creating a new input with the adafruit library is the best way to go, however since I don't have this sensor, developing it myself may be difficult. When developing an input module, I like to be able to let the user select which pins to use, and referencing your code I notice there are variables used rather than integers for the pins. I have several modules that use Adafruit's Circuitpython libraries, but only for I2C using Adafruit_Extended_Bus, none for UART. If you can you provide example code that allows specific pins (as integers) to be used with the adafruit library, I can probably take it from there to make it into a module.

My friend and I had also been working on this but it seems like you have beaten us to it! 🥇  Thanks for implementing this. 

I upgraded to the master branch to test this out and added the new MAX31865_CIRCUITPYTHON data inputs using the GPIO numbering (5,6), this yielded 0C for both sensors. Out of curiosity I also tried BCM numbering (29,31) and this gave the same 0C reading. I'll stick to the GPIO numbering as the tooltip suggests, just felt like it needed trying! 

Something I have noticed, if I boot up the raspberry pi and run my script outside of mycodo it works fine, but then when I try the script again having added the sensors in mycodo it gives readings of -242C. Which is really weird.

[image]
[image]
[image]
[image]
[image]

not5, 2020-12-16:
Based on your images above, the BCM numbering for the cs pins is 5 and 6. Make sure SPI is enabled on your pi: https://localhost/settings/pi this may be a silly question, but make sure your board is soldered correctly.

electricspaceman, 2020-12-16:
> Based on your images above, the BCM numbering for the cs pins is 5 and 6. Make sure SPI is enabled on your pi: https://localhost/settings/pi this may be a silly question, but make sure your board is soldered correctly.

I’m pretty sure the hardware is configured correctly as I was able to acquire correct readings outside mycodo just fine using a python script I cobbled together. 

It’s when we activate the sensors in mycodo that things go awry.

electricspaceman, 2020-12-17:
Good news!

I think I have found the issue. I have commented out the SPI try lock code, as shown in the screenshot below, I get temperature readings in mycodo :D

I don't understand what the SPI try lock does, is there something on my system config, is this needed?

[image]

not5, 2020-12-17:
Not sure what the SPI lock was added for... SPI is a party bus, so it shouldn't be "locked." Kyle, was there a reason you added that portion to the input code?

kizniche, 2020-12-17:
I was testing. I'll remove it for release.

not5, 2020-12-17:
@electricspaceman sounds like we can close this issue?

kizniche, 2020-12-17:
This is the info I was reading when I started testing locking: https://www.digikey.com/en/maker/projects/circuitpython-basics-i2c-and-spi/9799e0554de14af3850975dfb0174ae3

electricspaceman, 2020-12-18:
Thanks for getting this working now. I did a fresh install at it's working great!

closed 2020-12-18 by electricspaceman