kizniche/Mycodo #571: setup>functions>conditional measurement stops working all the time

SAM26K, 2018-11-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.5
 
#### Problem Description
Setup conditional measurement control to turn on a relay.  Conditional control stops working after some time.  Works for hours sometimes.  Only way to fix is change some parameters in the conditional measurement control and save the changes.

  - what were you trying to do
  using analog sensors to turn a relay on and off depending on the voltage from the sensors.

kizniche, 2018-11-25:
Does the daemon log contain errors?

Please screenshot both the conditional and input settings.

SAM26K, 2018-11-27:
Hi Kyle,
Sorry for the delay.  When it happens the controls still switchd the relays on and off manually. Here's the screenshots:
[image]
[image]
[image]
[image]

SAM26K, 2018-11-27:
All those screenshots are from the master pi on 192.168.1.17 and has the A/D shield mounted on it.  The pi controlling the relays is on 192.168.51.

SAM26K, 2018-11-27:
Screenshots from remote relay pi. 8 Channel relay board controlled with GPIO  has daemon errors but works ok from master pi using manual controls:
[image]

kizniche, 2018-11-28:
Do all conditionals stop working at the same time?

kizniche, 2018-11-30:
Do you have the ability to test the 7.0 branch for this issue? I'm hesitant to put in a lot of time diagnosing an issue that may have already been fixed in the next release. [This comment](https://github.com/kizniche/Mycodo/issues/562#issuecomment-441150035) contains the commands to get the 7.0 branch running.

SAM26K, 2018-12-01:
Will try that.  FYI the only way to get the conditional working again is to
delete the comparison parameter and re-enter it again.
Does the 7.0 branch have the ADS1256 fixes?
Thanks

kizniche, 2018-12-01:
Yes, it should be working.

SAM26K, 2018-12-02:
Also noticed I have a lot more input choices with the ADS1256 A/D pcb.  Only 8 ch AD is connected to pi.  Have two inputs in the data page and 2 measurement displays on dash.  What are these extra inputs? TIA
[image]

SAM26K, 2018-12-02:
Data input control settings for 2nd  analog input:
[image]

kizniche, 2018-12-02:
Fixed the decimal issue in the last commit (above).

You'll have to explain in more detail what the ADC issue is, as it looks to me to be working properly. You have two ADC Inputs, each with 8 channels. That means each input (8 channels each) will appear as selection options (totaling 16 channels).

kizniche, 2018-12-02:
It looks from the screenshot you are only enabling 1 of the 8 channels for Input ACh1. This indicated you only have one thing connected to this ADC (on channel 2). I can't see the other Input to determine how you have configured it, but I suspect your second physical ADC only has one input connected as well and you have only one channel enabled.

I'm not sure why you don't have both connections connected to just one ADC chip, channel 1 and channel 2. It seems like a waste of the second ADC when you could have both inputs connected to just one ADC.

kizniche, 2018-12-02:
I also just noticed another Input in the screenshot, with a third ADC chip. I am now suspecting you are improperly configuring the ADC. You should only have one Input per ADC chip.

kizniche, 2018-12-02:
I need to add some sanity-checking to prevent more than one Input with the same I2C address and bus from being activated. This will prevent such issues with misunderstanding how to configure an ADC.

SAM26K, 2018-12-02:
Only using a single ADS1256 Waveshare PCB.  The new 7.0 controls seem to
generate multiple outputs

SAM26K, 2018-12-02:
What about the resolution?       Dash > measurement shows 0 decimal places.

kizniche, 2018-12-02:
> The new 7.0 controls seem to generate multiple outputs

Do you mean inputs? There will be as many channels as the chip will allow. for the ADS1256, there will be 8 channels. When you add the input, all 8 are enabled and you can disable any you don't need a measurement from.

> What about the resolution? Dash > measurement shows 0 decimal places.

I already pushed a fix, so you will need to grab the latest code with:

```cd ~/Mycodo && sudo git fetch && sudo git pull && sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize && sudo service mycodoflask restart```

SAM26K, 2018-12-03:
Decimals working ok but now if more than 2 channels from the ADS1256 is enabled, the all go to just above zero volts out.  Also notice that the input selection increases with every input enabled.
[image]
[image]
[image]
[image]

SAM26K, 2018-12-03:
[image]

kizniche, 2018-12-03:
First, you must start using the input properly before we can assess if there is an issue. Delete all inputs except for one.

SAM26K, 2018-12-03:
All inputs deleted and created a single ch0 data input,  now getting a 500 server error, should I re-install?
[image]

SAM26K, 2018-12-03:
How do I use the new inputs properly?

kizniche, 2018-12-03:
I just pushed a fix for the 500 error. You'll have to do the series of commands from above with git fetch/pull to update.

You should only have one ADC input and enable/disable the channels you want to use in the "Measurements Enabled" multiselect box.

SAM26K, 2018-12-03:
Ok back to working again but still confused about the new config of AD, Now one instance of the input has 8 channels each instead of 1 channel?
I think only the first channel is accurate using that method.  ie if I   pick channel 2 of instance one it reads  near a rail and know its not accurate, If I create a new instance of the ADS1256 with the next channel, that channel is accurate too,

SAM26K, 2018-12-03:
Where is the  ""Measurements Enabled" multiselect box"?

SAM26K, 2018-12-04:
Ok, I think I'm getting it now.  Will take a while to rebuild all the controls but so far seeing reasonable numbers for the 6 channels I have sensors connected to.

SAM26K, 2018-12-05:
Hit a 500 error when adding outputs to dashboard:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 685, in page_dashboard
    y_axes=y_axes)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 135, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 117, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["dashboard", _('Dashboard')] %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 259, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 63, in block "body"
    {% include 'pages/dashboard_options/mod_graph.html' %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard_options/mod_graph.html", line 190, in top-level template code
    {{' (' + dict_units[colors_graph[each_dashboard.unique_id][n]['unit']]['name']}})</label>
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: collections.OrderedDict object has no element Undefined

kizniche, 2018-12-06:
I just pushed a fix for the graph issue adding outputs.

SAM26K, 2018-12-07:
Thanks for the update. Seems to work ok so far but still rebuilding all the controls.  Not updating the slave pi with the relay GPIO PCB attached.  Seems to work ok with 6.45 so far.
Noticed that the dashboard 'measurement' display height parameter seems to no longer have any effect with 7.0.
Thanks again.

SAM26K, 2018-12-07:
Seems to be a problem with timed output.  If I set the conditional to turn on for 2 mins and then ignore for 5 mins the output turns on but never turns off. Manual controls ok.  e.g. 
[image]

kizniche, 2018-12-07:
It seems the conditional refractory period wasn't operating as expected. I just pushed a fix for that, however I've also pushed a commit recently that changed the database schema (added a start offset to PID and conditional controllers), so your current database will need to be regenerated if you want to use the latest code.

If you want to just make the very minor change yourself, all you need o do is change the following line:

https://github.com/kizniche/Mycodo/blob/42244940998a887603836e4d961642b6b9ce5622/mycodo/controller_conditional.py#L275

To this:

```python
        if self.refractory_period:
```

Save, then restart the backend.

SAM26K, 2018-12-08:
Thanks for the update.  Not updating dB yet as not using PID functions yet.  Focused on verifying if conditionals stop working in 7.0 or not.  Do you think the other changes could affect the conditionals reliability problem?
Thanks again for the code fix.

SAM26K, 2018-12-08:
> I'm not sure what your original issue was, as I wasn't able to replicate
> it. If you can verify there is an issue with the 7.0 code, I'll
> investigate. The refractory period issue should now be fixed.

The latest issue was that the conditional function stopped working after a
while and had to rewrite the conditional parameter and resave the control
to get it to respond again.
Would you prefer I start a new issue with new issues related to conditional
functions?

kizniche, 2018-12-08:
What I meant was I understood what the issue was, but I was not able to replicate it. Don't open a new issue unless you can replicate it in the 7.0 branch.

SAM26K, 2018-12-09:
Everything seems to be working so far except for function > conditional > refractory period.  If I set it to 300 secs it only runs for 2.5 minutes before it activates again.

kizniche, 2018-12-09:
I just released 7.0.0. Can you install this release fresh and test if this still occurs? I wasn't able to replicate it. When I set the refractory period to 300, it waits 5 minutes from execution to execute again (if conditional remains true).

SAM26K, 2018-12-09:
Is there a way to use the web UI "upgrade" function or how should I upgrade?  It tells me I'm already up to date even though my version is edited.  Is there a way to keep all the controls?
TIA

SAM26K, 2018-12-09:
> I changed the database schema in a commit a day or two ago, so if you're
> database was generated prior to that, you're going to have to start over
> anyway.
>
> If you have a very recent database, you can wait until I release 7.0.1
> that includes a few LCD bug fixes. I should have that released by the end
> of tomorrow.

Were you able to confirm the dashboard measurement display's height
parameter doesn't seem to have any effect?

kizniche, 2018-12-09:
Are you suggesting only Measurements on the dashboard exhibit this behavior? To my understanding, the new version of bootstrap (v4) doesn't allow different height divs on the same row. Therefore, you should see this represent as all elements on the same row will be the height of the tallest element.

SAM26K, 2018-12-09:
Ok, you're right.  No problem there.
Thanks again.

SAM26K, 2018-12-09:
Will need more time to confirm operation to close the original issue though.

kizniche, 2018-12-09:
> Ok, you're right. No problem there.

I spent a good while trying to solve this issue, but it seems fairly difficult. I may return to it soon. Having different heights in the same row is something I liked.

SAM26K, 2018-12-09:
Is there any to reduce font size below 1?  I'm using 1 for all font sizes
and could use 1/4 that size with a 1080 monitor resolution.

kizniche, 2018-12-09:
Font size of what? You can zoom out in your browser.

SAM26K, 2018-12-09:
Was referring to the font size parameter in various display widgets.  Using
1 for everything but maybe a completely different font could produce
smaller screen fonts? Know about the browser zoom function but that is not
the best solution.
No big deal just a suggestion when you get time.
Thanks again.

kizniche, 2018-12-09:
You can set the font size options to less than 1.

SAM26K, 2018-12-09:
Did a fresh install from the released  7.0 and rebuilt the web UI controls. 
The problem now is that when I create  > function > conditional,  it no longer lists the id parameter in curly brackets. Instead, it generates a generic text that is the same for all new function conditional controls.
ie its not clear what  {} is needed for the comparison test:
[image]

kizniche, 2018-12-09:
I make a quick update to make the conditional system a little more powerful, now that full, long Python code can be used. It's at the expense of usability to the inexperienced, but I'd like to get your feedback on how the documentation can be improved. The screenshot, below, has the ID circled that you need to use in the code.

[image]

kizniche, 2018-12-09:
I should change the [ID] in that disabled input/button to {ID} to make it a little more suggestive that is the correct ID.

SAM26K, 2018-12-09:
Ok thanks for the update.  Do I need to push that change manually or how should  I do that?

kizniche, 2018-12-09:
Since 7.0.0 is released, you'll get the updates as you normally would when I release 7.1.1, which should be sometime today.

SAM26K, 2018-12-09:
OK, thanks again.

kizniche, 2018-12-09:
I also just added a sanity check for the actual code. The IDs will be attempted to be replaced and the code executed before it's allowed to be saved (screenshot, below). If there's an error, a large detailed error will appear and the code will be prevented from being saved.

[image]

SAM26K, 2018-12-10:
Hi Kyle,
With fresh install using released 7.0.0 conditional controls no longer work at all.  With 7.0.1 I get an error similair to above.  The ID was copy and pasted from the condition tag.
[image]
[image]
[image]

kizniche, 2018-12-10:
Back at my computer. For syntactically-correct Python code, our conditional statement should be:

```python
if {66493fa3} > 1.8:
    print(1)
```

SAM26K, 2018-12-10:
Thanks, I got it working now.  Will run for a few days to confirm it keeps working.
Thanks.

SAM26K, 2018-12-10:
It still seems to have a timing problem.  I'm programming it to turn on a relay for 2 minutes then ignore for five minutes.  Instead of five minutes, it gets triggered again 2.5 minutes later instead of 5 minutes.  The weird thing is the graph shows five minute intervals between relay activations.  My stopwatch shows 2.5 minutes between activations.

SAM26K, 2018-12-10:
It seems I was measuring wrong.  If I time it from the start of the trigger to the start of the following trigger it is 5 minutes.

kizniche, 2018-12-10:
While setting up conditionals for one of my production systems today with the new 7.0, I found what can be an issue for particular setups and wanted to let you know about my findings.

The Max Age settings for a measurement determines how long ago to look in the database for a measurement. If it finds a measurement, it will replace the {ID} with the actual measurement before running the Python code. However, if it doesn't find a measurement, it will replace {ID} with "None". Therefore, if you don't want the code to throw an error (for trying to compare None to a numerical value, such as "None < 23", then you will need to add a test for None-ness before the actual conditional comparing it to a value. This can easily be accomplished a number of ways, but here are a few examples:

Example statement:

```python
if {asdf1234} < 24:
    print(1)
```

This statement will cause an error if {asdf1234} is replaced with None. However, if we add a check for None before the comparison to 24, we can avoid the error:

```python
if {asdf1234} is not None and {asdf1234} < 24:
    print(1)
```

or

```python
if {asdf1234} is not None:
    if {asdf1234} < 24:
        print(1)
```

I duplicated the check during saving of the conditional for properly-formatted code to include replacing with None (in addition to replacing with a numerical value). Now, if the code isn't formatted properly if a value has a None value, then the error will appear, like below:

[image]

kizniche, 2018-12-10:
However, I find it useful to include a conditional that makes sure measurements are being acquired, and if they aren't (returns None), then it will email me (or turn off the system and email me).

```python
if {asdf1234} is None:
    print(1)
```

[16 more replies]

closed 2018-12-22 by kizniche