kizniche/Mycodo #572: Suggestion: Adding VPD calculation to Inputs on Dashboard

BobDDole, 2018-11-27:
Along with the Temp, Humidity, & Dewpoint Inputs is it possible to add Vapor Pressure Deficit
as a Input as well?

BobDDole, 2018-11-28:
Yes indoors VPD is a very useful metric and would be a welcome addition to inputs. 
I would add it myself if I knew python. Calculation values seem to be done in mycodo/inputs/sensorutils.py And the valid ones are defined in the sensors related python file (IE mine is dht22.py).
Really wish I knew python enough to whip this up but alas I suck at programming languages.
BTW Theoi, I was going to include the EXACT same link you did for calculation, LOL
Thanks for giving the info

SAM26K, 2018-11-29:
Suggest creating a general math module input where real inputs from sensors can be parameters of a formula that generates an output that can be used for logging and triggering events.

kizniche, 2018-11-29:
I could add VPD as a measurement for inputs that return temperature and RH, as well as a math controller that can have temperature and RH selected from one or two different inputs. In 7.0, the user can select which measurements are stored from the sensor, so you can have many potential measurements returned and storage space isn't wasted if you don't want them by disabling them. One of the easy features I implemented with the new measurement system.

kizniche, 2018-11-29:
I implemented VPD returning for inputs that measure at least temperature and RH. I'll add the math controller later, but just wanted to give an update showing it calculating the proper VPD from the measurements (dummy data) that align with the expected result from the example in the provided link (above), using the equations for SVP and VPD.

[image]

kizniche, 2018-11-30:
Closing issue because both Input and Math VPD will be in the 7.0 release.

closed 2018-11-30 by kizniche

BobDDole, 2018-12-01:
Thanks, you the man!