kizniche/Mycodo #574: fresh install dht22 sensors not rworking

tomkm7p, 2018-11-30:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.5

#### Problem Description
Please list:

Raspi 3B  - raspbian stretch full version - fresh install 
Mycodo 6.4.5 - fresh install 

Tried connecting 2 separate dht22 sensors to GPIO #4 (bcm pin 4, physical pin 7)  and couldn't get them to read any data.  Tried using different data pins with no luck.  I am a complete noob to Pi, linux and programming, so i really dont know how to diagnose or troubleshoot this issue.  Tried looking for resolved issues with a similar problem but none could be found with sensors not working after a fresh install.  
Tried sudo apt-get install pigpio, got pigpio is already the newest version (1.64-1).
Tried sudo pigpiod , got 2018-11-30 00:32:23 initInitialise: Can't lock /var/run/pigpio.pid
Can't initialise pigpio library
Tried ~/Mycodo/env/bin/pip install Adafruit_GPIO got Installing collected packages: adafruit-pureio, Adafruit-GPIO
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/home/pi/Mycodo/env/lib/python3.5/site-packages/Adafruit_PureIO-0.2.3.dist-info'
Consider using the `--user` option or check the permissions.

Dont know how to proceed from here, any help would be greatly appreciated.  

I have reinstalled Stretch and reinstalled a fresh copy of Mycodo once again
### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1 - find a noob
 2. step 2... - install raspbian stretch and mycodo
 3. etc - plug the data wire into GPIO 4 (BCM 4, physical 7)
 
### Additional Notes
you are dealing with someone who created a github account just for this, so please take that into consideration

kizniche, 2018-11-30:
Can you provide a photo of your sensor connected to your Pi, with the wiring in clear view?

tomkm7p, 2018-12-01:
Sure,
here it is https://drive.google.com/open?id=0BxMGDEc9TLKEWFhxMFpERWdFMEE5QWstY0NYZ0dJZGtXUFF3 

I also read  somewhere online to try using sudo loldht X(PIN NUMBER) but i got an error saying command not found. 
Then I tried installing Mycodo on a fresh copy of Raspbian Jesse (older version of raspbian), and i couldnt load the web client it said error 502 bad gateway .... not my fault, create an issue on github 

I now have once again a fresh copy of mycodo on stretch.  In the web interface, under GPIO i put it under number 4 .... just to confirm, for this software you use BCM numbering and not GPIO numbering?

Thanks

tomkm7p, 2018-12-01:
the pic above shows a dht22, i also tried this with a wired dht22 (am2302) with a built in resistor with the same negative results

kizniche, 2018-12-01:
BCM numbering is correct. Try disabling the input, change your VCC to 5 volts, then activate it.

BobDDole, 2018-12-01:
I also had a problem with DHT22 sensors not working. For me it was because PIGPIOD wasn't running and wouldn't start on reboot so I ended up doing this:
sudo systemctl enable pigpiod
then
sudo pigpiod 
To start it immediately 
This worked for me YMMV.
For the record I'm using DHT22's on 12Ft of shielded wire on 3.3V rail with no problems.

mrplow, 2018-12-01:
DHT22 was working fine in 6.4.5 without issue and without needing to install any additional packages for me.
Still works fine after upgrading to the 7.0 branch.

tomkm7p, 2018-12-03:
Tried 5 V, same result
Tried pi@raspberrypi:~ $ ps aux|grep pigpoid
pi        1035  0.0  0.0   4376   528 pts/0    S+   23:28   0:00 grep --color=auto pigpoid

tomkm7p, 2018-12-03:
Tried pi@raspberrypi:~ $ sudo systemctl enable pigpiod
Created symlink /etc/systemd/system/multi-user.target.wants/pigpiod.service → /lib/systemd/system/pigpiod.service.
pi@raspberrypi:~ $ sudo pigpiod
pi@raspberrypi:~ $ 

the above works, thank you very much ..... now am I gonna have to type this in everytime my pi restarts?

kizniche, 2018-12-03:
So it appears pigpiod was the issue. Mycodo has it's own system for enabling/disabling pigpiod and it's suggested to allow it to control that process. This is accessed from Config [Gear Icon] -> Configure -> Raspberry Pi, where you will see "Configure pigpiod".

I would recommend disabling pigpiod using the method you used to enable it, then use the Mycodo config menu to enable the 1ms option of pigpiod.

tomkm7p, 2018-12-03:
How would I go about disabling it ? Whats the command?
Also, I noticed that now my raspi would stay on for no more then 2-3 hours before it would reboot by itself, it didnt do that before.  Could this this pigpoid issue some how caused this?

kizniche, 2018-12-03:
> How would I go about disabling it ?

Go to the configuration option I mentioned.

> Could this this pigpoid issue some how caused this?

Not sure. I've never had that happen with any of my Pis and I've had probably over 5000 days combined run time with DHT sensors.

tomkm7p, 2018-12-04:
i also attempted to change the configure options to 5 ms, still not recording 
Then i tried : pi@raspberrypi:~ $ sudo systemctl enable  pigpiodFailed to enable unit: Refusing to operate on linked unit file pigpiod.service
pi@raspberrypi:~ $ sudo pigpoid
sudo: pigpoid: command not found
pi@raspberrypi:~ $ 

now its not working again and i cant seem to turn it on

kizniche, 2018-12-04:
Did you read the information at the top of the configuration page?

tomkm7p, 2018-12-04:
Are you referring to the need to reboot for changes to take effect? Yes, rebooted after the config changes with still the same result, no data last 16 sec

kizniche, 2018-12-04:
I'm not sure the issue, but would recommend a complete SD format and install of raspbian and Mycodo.

tomkm7p, 2018-12-04:
but then again how come when i type sudo pigpoid it says command not found?

kizniche, 2018-12-04:
It's pigpiod, not pigpoid, and I not sure why it's not working. Perhaps try the 7.0 branch, however this is constantly changing, until realase.

tomkm7p, 2018-12-08:
Gotcha, thank you very much for your help, gonna try out this software over the weekend

closed 2018-12-08 by tomkm7p

tomkm7p, 2018-12-08:
As a side question without opening up a new issue, does the raspi change ips ?

After i got mycodo working i had a power interruption (router as well, whole house interruption) .  Afterwards i couldn't log in using my usual raspi ip address.  I thought it was daemon but then after reading your diagnosis page and  doing a sudo ip addr show, i realized the ip address was changed.  The raspi is connected to my network via wifi while im testing it.  In the future i plan to not connect it via a router or wifi, i plan on using an ethernet cord pluged into my laptop to access the web interface.  In that scenario, will the raspi change IPs, or is it dependent on the router as well to change ips?

jacopienaar, 2019-01-06:
I have the exact same problem and none of the above workarounds actually work. I tried everything. The DHT22 sensor works in previous versions of Mycodo. I'm clutching at straws here but it seems the feature of being able to control the pigpiod from within Mycodo interface might be what's causing the issue. 
"> So it appears pigpiod was the issue. Mycodo has it's own system for enabling/disabling pigpiod and it's suggested to allow it to control that process. "

It seems the only solution is to not use the crappy DHT's

kizniche, 2019-01-06:
Did you allow Mycodo to handle pigpio since the install of raspbian?

jacopienaar, 2019-01-06:
Yes. I did a clean install of the latest raspbian followed by installation of Mycodo. Sensor didn't work from the outset, however, after above workaround it is working

kizniche, 2019-01-06:
Was the "1 ms" option tried in the Mycodo config?

jacopienaar, 2019-01-06:
Yes, I tried 1ms & 5ms - both didn't work.

kizniche, 2019-01-10:
I just did a completely fresh install (Raspbian and Mycodo) and after adding the DHT22 (and installing the pigpio dependency it forces to occur), pigpiod was running in the background (ad the default 1 ms sample rate. Even after a reboot pigpiod was still running. So, I would recommend a fresh Raspbian and Mycodo install if you're still experiencing this issue.