kizniche/Mycodo #575: Pi-Plates

shortyishere, 2018-12-02:
Can we add the Pi-Plates and automation hat to your software. That would be awesome!

shortyishere, 2018-12-02:
Also here is the Pi-Plates python code if interested
https://pypi.org/project/pi-plates/

shortyishere, 2018-12-02:
Also here is the Automation Hat python code if interested...
https://pypi.org/project/automationhat/

kizniche, 2018-12-05:
Out of the box, Mycodo should be able to integrate with most of these pieces of hardware. Mycodo uses input modules that can be created and imported by the user that will allow measurements to be acquired and used within the Mycodo system. Additionally, a Command output can be used to execute commands that would turn on/off relays on the relay plate.

Do you have a particular piece of hardware that you're looking to get working with Mycodo and need assistance with the configuration?

shortyishere, 2019-01-09:
Ok time to get the Pi-Plates working! What would you recommend where I should start

kizniche, 2019-01-09:
Which ones do you have?

kizniche, 2019-01-09:
First, you'll need to install the pi-plates python module into the Mycodo virtualenv:

```
~/Mycodo/env/bin/pip install pi-plates
```

From the virtualenv, you now you have commands available via importing piplates.DAQC2plate

```python
~/Mycodo/env/bin/python
>>> import piplates.DAQC2plate as DAQC2
```

I'm just heading out so I can't show you much more, but that should get you started with testing commands from the user manual. There are a number of places in Mycodo that you can use Python or Linux commands that will allow you to use the Mycodo virtualenv to execute commands. I can come up with some examples later when I have more time.

kizniche, 2019-01-11:
To use the analog-to-digital converter, I would recommend copying one of the current ADC Input modules, such as https://github.com/kizniche/Mycodo/blob/master/mycodo/inputs/ads1256.py

You will at the very least change the ```'input_name_unique': 'ADS1256',``` to a unique name for it to be able to be imported under Config -> Input. This will allow you to create an ADC module that accesses the Pi-plate's 8 ADC channels.

kizniche, 2019-01-11:
Here is one I just threw together that may work:

```python
# coding=utf-8
import logging
from collections import OrderedDict

from flask_babel import lazy_gettext

from mycodo.databases.models import DeviceMeasurements
from mycodo.inputs.base_input import AbstractInput
from mycodo.utils.database import db_retrieve_table_daemon

# Measurements
measurements_dict = OrderedDict()
for each_channel in range(8):
    measurements_dict[each_channel] = {
        'measurement': 'electrical_potential',
        'unit': 'V'
    }

# Input information
INPUT_INFORMATION = {
    'input_name_unique': 'PI_PLATE_DAQC2',
    'input_manufacturer': 'Pi-Plates',
    'input_name': 'DAQC2',
    'measurements_name': 'Voltage (Analog-to-Digital Converter)',
    'measurements_dict': measurements_dict,
    'measurements_rescale': True,
    'scale_from_min': 0.0,
    'scale_from_max': 5.0,

    'options_enabled': [
        'measurements_select',
        'custom_options',
        'period',
        'pre_output'
    ],
    'options_disabled': ['interface'],

    'dependencies_module': [
        ('pip-pypi', 'piplates', 'pi-plates')
    ],
    'interfaces': ['GPIO'],

    'custom_options': [
        {
            'id': 'address',
            'type': 'select',
            'default_value': '0',
            'options_select': [
                ('0', '0'),
                ('1', '1'),
                ('2', '2)'),
                ('3', '3'),
                ('4', '4'),
                ('5', '5'),
                ('6', '6'),
                ('7', '7')
            ],
            'name': lazy_gettext('Pi-Plate Address'),
            'phrase': lazy_gettext('The address of the Pi-Plate')
        }
    ],
}

class InputModule(AbstractInput):
    """ ADC Read """
    def __init__(self, input_dev, testing=False, run_main=True):
        super(InputModule, self).__init__()
        self.logger = logging.getLogger('mycodo.ads1256')
        self.run_main = run_main
        self._measurements = None

        if not testing:
            import piplates.DAQC2plate as DAQC2

            self.logger = logging.getLogger(
                'mycodo.ads1256_{id}'.format(id=input_dev.unique_id.split('-')[0]))

            self.daqc2 = DAQC2
            self.adc_calibration = None

            self.device_measurements = db_retrieve_table_daemon(
                DeviceMeasurements).filter(
                    DeviceMeasurements.device_id == input_dev.unique_id)

            # Check if all measurements are enabled
            self.all_enabled = True
            for each_measure in self.device_measurements.all():
                if not each_measure.is_enabled:
                    self.all_enabled = False

            if input_dev.custom_options:
                for each_option in input_dev.custom_options.split(';'):
                    option = each_option.split(',')[0]
                    value = each_option.split(',')[1]
                    if option == 'address':
                        self.address = int(value)

    def get_measurement(self):
        self._measurements = {}

        return_dict = measurements_dict.copy()

        if self.all_enabled:
            # Acquire ADC measurements from all 8 channels
            list_measurements = self.daqc2.getADCall(self.address)
            for each_measure in self.device_measurements.all():
                return_dict[each_measure.channel]['value'] = list_measurements[each_measure.channel]
        else:
            # Acquire ADC measurements from only enabled channels
            for each_measure in self.device_measurements.all():
                if each_measure.is_enabled:
                    return_dict[each_measure.channel]['value'] = self.daqc2.getADC(
                        self.address, each_measure.channel)

        return return_dict

if __name__ == "__main__":
    from types import SimpleNamespace
    settings = SimpleNamespace()
    settings.id = 1
    settings.unique_id = '0000-0000'
    settings.adc_gain = '1'
    settings.adc_sample_speed = '10'
    settings.channels = 8

    measurements = InputModule(settings, run_main=True).next()
    print("Measurements: {}".format(InputModule(settings).next()))
```

kizniche, 2019-01-11:
This module will measure the 8 ADC channels. We can create separate modules for each part of the Pi-plate, or we could have one DAQC2 module allow acquisition of all potential measurements. Go ahead and import that module and let me know if it works. If not, check the daemon log for errors.

shortyishere, 2019-01-13:
Doing it now I'll report back. Thank you

shortyishere, 2019-07-18:
hi kizniche, I have been away. and I am back trying these. I now have an even better board and it's this board. https://www.sequentmicrosystems.com/megaio-bas.html
 now have both board on each rasp pi. I will try and make them both work now.

closed 2020-07-23 by kizniche