kizniche/Mycodo #576: issue timeout system information

alcemyco, 2018-12-04:
## Mycodo Issue Report:
  - Specific Mycodo Version:6.4.5
 
#### Problem Description
Please list: 

  - trying to open system information 

 
### Errors
  -gateway timeout 504

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
no other issues hve been presented in the gui.

kizniche, 2018-12-04:
If you don't provide steps to reproduce the error, there's little chance this can ever be resolved.

kizniche, 2018-12-05:
I must have missed the "system information" part. I've found bugs that create perpetual processes that cause the fronted to timeout, but I've never experienced it on the system information page. @Theoi-Meteoroi makes a good point, that you should see if the issue persists with nothing connected, then add hardware one at a time until the issue reappears.

closed 2019-01-04 by kizniche