kizniche/Mycodo #577: DHT22 inaccurate humidity?

mschatte, 2018-12-04:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.4.5
 
#### Problem Description
Please list:

- No specific issue with Mycodo, just seeing if anyone else has this issue. 
- Have 2 DHT22 sensors in a mini greenhouse along with a analog humidor humidity sensor. 
- Both DHT22 always seems to be 25-30% above humidor hygrometer. 
- Have recently calibrated hygrometer to 100% RH
- Hygrometer shows 70% RH but DHT22 shows 99.9% RH. 

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Bought 2 DHT22 sensors
https://www.amazon.com/Gowoops-Temperature-Humidity-Measurement-Raspberry/dp/B073F472JL/ref=sr_ph_1?ie=UTF8&qid=1543950826&sr=sr-1&keywords=dht22 
 2. Setup PID to regulate RH in the greenhouse (this works fine). 
 3. DHT22 sensor in greenhouse will quickly spike to 99.9% RH even though my hygrometer never goes over 70%. 
 
### Additional Notes
I just wanted to see if anyone else with the DHT22 sensor has the same issue.

mschatte, 2018-12-05:
Also, I'm using these in my mushroom growing so the humidity is 90% and up

Theoi-Meteoroi, 2018-12-05:
This may be of interest to you: [Humidity Sensors Compared](http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html)

My personal preference is to use at least two different sensor technology types if possible for humidity  value comparison.  I haven't had any saturation issues with the AM2315 - I think the probe design may prevent internal condensation - which seems to be what you report with the DHT22.  I have used the SHT31 with success in higher humidity environments, but not with Mycodo.

mschatte, 2018-12-05:
That's great info! I was considering the SHT3x sensors already to help with this issue. Thanks!

closed 2018-12-05 by mschatte

kizniche, 2018-12-06:
I actually just added support for the SHT31 the other day in the 7.0 branch.