kizniche/Mycodo #578: v7.0 LCD - Python error

mrplow, 2018-12-06:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.0, fresh install was 32acf19
 
#### Problem Description
Please list:

  - what were you trying to do: Setup the LCD in the latest beta
  - specific setup details that are involved: Fresh install
 
### Errors
  - Upon clicking on Setup - LCDs I receive the following error:
```
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 968, in page_lcd
    input_dev, math, pid, output, dict_units, dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 364, in choices_lcd
    choices = form_output_choices(choices, each_output)
TypeError: form_output_choices() missing 2 required positional arguments: 'dict_units' and 'dict_measurements'
```

kizniche, 2018-12-06:
I just pushed a fix for loading the LCD page

mrplow, 2018-12-06:
Thanks @kizniche, that allowed me to setup the LCD, unfortunately it doesn't initialize
```
2018-12-06 07:19:29,488 - mycodo.system_pi - ERROR - None, None
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/system_pi.py", line 54, in return_measurement_info
    measurement = device_measurement.rescaled_measurement
AttributeError: 'NoneType' object has no attribute 'unit'
2018-12-06 07:19:29,490 - mycodo.lcd_1208f88e - ERROR - Error: 'NoneType' object has no attribute 'measurement'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 181, in __init__
    each_lcd_display.line_1_measurement)
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 432, in setup_lcd_line
    self.lcd_line[display_id][line]['measure'] = measurement.measurement
AttributeError: 'NoneType' object has no attribute 'measurement'
2018-12-06 07:19:29,492 - mycodo.lcd_1208f88e - INFO - Activated in 141.4 ms
2018-12-06 07:19:30,495 - mycodo.lcd_1208f88e - ERROR - Could not initialize LCD. Check your configuration and wiring. Error: 'LCDController' object has no attribute 'bus'

```

kizniche, 2018-12-06:
Thanks. I'll look into it and push a fix to test shortly.

kizniche, 2018-12-06:
Just pushed a potential fix.

mrplow, 2018-12-07:
Getting there! LCD is working, except the IP address line output has 'no data'
```
2018-12-06 15:02:54,780 - mycodo.system_pi - ERROR - None, None
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/system_pi.py", line 56, in return_measurement_info
    unit = device_measurement.unit
AttributeError: 'NoneType' object has no attribute 'unit'
2018-12-06 15:02:55,628 - mycodo.lcd_5f51870a - INFO - Activated in 1045.7 ms
```

kizniche, 2018-12-07:
The most recent commit should fix the IP address issue.

mrplow, 2018-12-08:
Yup all working. Thanks!

closed 2018-12-08 by mrplow

kizniche, 2018-12-09:
I just released 7.0.0. If you could install fresh and test if the LCD still works it would be appreciated. Thanks.

kizniche, 2018-12-09:
I just found an issue. Working on a fix.

```
2018-12-08 21:13:16,342 - mycodo.lcd_3a48fe61 - INFO - Activated in 981.7 ms
2018-12-08 21:13:23,510 - mycodo.lcd_3a48fe61 - ERROR - KeyError: Unable to output to LCD.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 225, in run
    self.output_lcds()
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 395, in output_lcds
    if self.lcd_string_line[display_id][3]:
KeyError: 3
2018-12-08 21:13:30,588 - mycodo.lcd_3a48fe61 - ERROR - KeyError: Unable to output to LCD.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 225, in run
    self.output_lcds()
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 395, in output_lcds
    if self.lcd_string_line[display_id][3]:
KeyError: 3
```

kizniche, 2018-12-09:
The last commit fixed the display issue. I'll make the 7.0.1 release soon to get the fix live.