kizniche/Mycodo #579: Display Addition: Adafruit PiOLED

Magnum-Pl, 2018-12-07:
Hi Kyle,

Can you add support for the Adafruit PiOLED 128x32 I2C display? 

I'd like to use it to display temperature and PID data. 

I've attached a Python script from the Adafruit GitHub repo.

Also included is a link to the tutorial I used to get the example code running.

I tested the script and was able to display the data.

[Adafruit_Python_SSD1306-master.zip](https://github.com/kizniche/Mycodo/files/2658712/Adafruit_Python_SSD1306-master.zip)

[https://learn.adafruit.com/adafruit-pioled-128x32-mini-oled-for-raspberry-pi/usage](url)

[image]

kizniche, 2018-12-07:
Thanks for the recommendation. Just made a commit that redesigned the LCD system to permit the use of LCD classes (different LCD modules), moved the generic 16x2 and 16x4 LCD code to a new class, then added the PiOLED code to a new class. This should allow the PiOLED LCD to now be used in Mycodo, however, none of this code is tested to work (either the generic LCD backpack or PiOLED), so we're likely to see errors when the LCD is activated. So, if you can test the new code, we can begin debugging the issues that pop up. I'm not in the presence of my development Pi or my 16x2 LCD to test, so I'll have to attempt that sometime in the near future.

Magnum-Pl, 2018-12-07:
Thanks for the quick response. I can help with this. Where do I start?

kizniche, 2018-12-07:
If you already have the latest release installed, this is how I would go about backing it up and installing the 7.0 branch:

```bash
sudo service mycodo stop
sudo service mycodoflask stop
sudo mv ~/Mycodo ~/Mycodo-backup
cd
git clone github.com/kizniche/Mycodo
cd ~/Mycodo
git checkout major_release_v7.0
sudo /bin/bash ~/Mycodo/install/setup.sh
```

kizniche, 2018-12-07:
Oops, I haven't added a key piece for the PiOLED code to be used. Let me push a fix then give you some commands to update your repo clone.

kizniche, 2018-12-07:
If you haven't performed the above commands to start the 7.0 branch install, you can now try and the code should be good to test. If you've already initiated or finished the install, you'll have to perform these commands to get the latest code:

```bash
sudo service mycodo stop
sudo service mycodoflask stop
cd ~/Mycodo
sudo git fetch
sudo git pull
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
sudo service mycodo start
sudo service mycodoflask start
```

kizniche, 2018-12-07:
I actually see some issues I'll have to resolve before you can properly test. Let me see if I can get them fixed before I have to leave for the evening. Stay tuned.

Magnum-Pl, 2018-12-08:
I was able to get 7.0.0 running, but the daemon isn't running. The  daemon log says file empty. Any advice?

kizniche, 2018-12-08:
Make sure the daemon isn't running with:

```ps aux | grep mycodo_daemon.py```

Then try to start the daemon manually to determine any errors during startup before logging begins:

```sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py```

kizniche, 2018-12-08:
You can terminate the daemon with:

```sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_client.py -t```

And to start the daemon via its service:

```sudo service mycodo start```

kizniche, 2018-12-08:
If running by the service, the daemon can be manipulated with the following commands"

```
sudo service mycodo start
sudo service mycodo stop
sudo service mycodo restart
```

Magnum-Pl, 2018-12-08:
When I ran this, 

`sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py` 

I got this:

Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 54, in <module>
    from mycodo.controller_lcd import LCDController
  File "/home/pi/Mycodo/mycodo/controller_lcd.py", line 72, in <module>
    from mycodo.devices.lcd_pioled import LCD_Pioled
  File "/home/pi/Mycodo/mycodo/devices/lcd_pioled.py", line 5, in <module>
    import Adafruit_SSD1306
ImportError: No module named 'Adafruit_SSD1306'

Then I ran this 

`sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_client.py -t`

I got this:

Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 54, in __init__
    self.rpyc_client = rpyc.connect("localhost", 18813)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/rpyc/utils/factory.py", line 90, in connect
    s = SocketStream.connect(host, port, ipv6 = ipv6, keepalive = keepalive)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/rpyc/core/stream.py", line 150, in connect
    return cls(cls._connect(host, port, **kwargs))
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/rpyc/core/stream.py", line 108, in _connect
    s.connect(sockaddr)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 289, in <module>
    daemon = DaemonControl()
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 56, in __init__
    raise Exception("Connection refused. Is the daemon running?")
Exception: Connection refused. Is the daemon running?

kizniche, 2018-12-08:
I just pushed a fix. Use the commands in [this comment](https://github.com/kizniche/Mycodo/issues/579#issuecomment-445398310) to update your code and see if the daemon starts.

Magnum-Pl, 2018-12-08:
`sudo git pull`   

gives me this error

Updating 28c2f110..1345359f
error: Your local changes to the following files would be overwritten by merge:
	mycodo/controller_lcd.py
	mycodo/devices/lcd_generic.py
Please commit your changes or stash them before you merge.
Aborting

Magnum-Pl, 2018-12-08:
I got it to work and the daemon is running now. Ill try to get the screen running

Magnum-Pl, 2018-12-08:
After installing the dependencies, I was able to add the display and activate it, but I wasn't able to get it to display any data. Deactivatiing it gives me this error:

> Error: Could not deactivate LCD controller with ID 0c1db50f-bb5e-448f-8c26-1469e368b178, it's not active.

"Reset LCD" gives me the error:

> Traceback (most recent call last):
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
>     response = self.full_dispatch_request()
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
>     rv = self.handle_user_exception(e)
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
>     reraise(exc_type, exc_value, tb)
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
>     raise value
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
>     rv = self.dispatch_request()
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
>     return self.view_functions[rule.endpoint](**req.view_args)
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
>     return func(*args, **kwargs)
>   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 996, in page_lcd
>     utils_lcd.lcd_reset_flashing(form_lcd_mod.lcd_id.data)
>   File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_lcd.py", line 219, in lcd_reset_flashing
>     return_value, return_msg = control.lcd_flash(lcd_id, False)
> TypeError: 'NoneType' object is not iterable

kizniche, 2018-12-08:
Pushed another fix to test.

kizniche, 2018-12-08:
To overwrite any changes you made to tracked files, use these commands:

```bash
sudo service mycodo stop
sudo service mycodoflask stop
cd ~/Mycodo
sudo git fetch --all
sudo git reset --hard origin/major_release_v7.0
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
sudo service mycodo start
sudo service mycodoflask start
```

Magnum-Pl, 2018-12-08:
Done. Now I'm able to activate, deactivate and reset the screen without any error, but the display shows "ID NOT FOUND" on all 4 lines. 

The log shows this:

> 2018-12-07 22:48:28,842 - mycodo.lcd_0c1db50f - ERROR - Error: argument of type 'NoneType' is not iterable
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 191, in __init__
>     self.lcd_out.lcd_write_lines(line_1, line_2, None, None)
>   File "/var/mycodo-root/mycodo/devices/lcd_pioled.py", line 61, in lcd_write_lines
>     draw.text((x, top + 16), message_line_3, font=font, fill=255)
>   File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 222, in text
>     if self._multiline_check(text):
>   File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 213, in _multiline_check
>     return split_character in text
> TypeError: argument of type 'NoneType' is not iterable
> 2018-12-07 22:48:28,844 - mycodo.lcd_0c1db50f - INFO - Activated in 736.4 ms
> 2018-12-07 22:48:40,896 - mycodo.lcd_0c1db50f - ERROR - Error: argument of type 'NoneType' is not iterable
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 191, in __init__
>     self.lcd_out.lcd_write_lines(line_1, line_2, None, None)
>   File "/var/mycodo-root/mycodo/devices/lcd_pioled.py", line 61, in lcd_write_lines
>     draw.text((x, top + 16), message_line_3, font=font, fill=255)
>   File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 222, in text
>     if self._multiline_check(text):
>   File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 213, in _multiline_check
>     return split_character in text
> TypeError: argument of type 'NoneType' is not iterable
> 2018-12-07 22:48:40,898 - mycodo.lcd_0c1db50f - INFO - Activated in 721.9 ms

kizniche, 2018-12-08:
I applied a potential fix and also added debugging lines, so please look out for the "TEST00" lines in the daemon log and paste them if it's still not working.

kizniche, 2018-12-08:
Do you have a measurement set for all 4 lines? What is your configuration? I just pushed a commit with more debugging lines.

Magnum-Pl, 2018-12-08:
For now I was just using the IP address option for each line. I don't have anything else added yet. I'll add a sensor and see if it makes a difference.

It won't let me activate the display if I don't have something chosen for each line. It would be nice to be able to set a line as blank. Is that something you can change?

kizniche, 2018-12-08:
Okay, I think that last commit now fixed it. I can add an option to leave a line blank.

Magnum-Pl, 2018-12-08:
That was the issue. I added RPi temp and cpu measurements and it works fine. Its only the IP address that won't display. Still says "ID NOT FOUND". 

Everything else is showing up fine.

[image]

Magnum-Pl, 2018-12-08:
The sensors are still displaying properly after the last changes, but the IP address still doesn't show up. Instead of "ID NOT FOUND", its just a blank line. 

The log shows this:

> 2018-12-07 23:51:41,231 - mycodo.lcd_0c1db50f - ERROR - Error: 'NoneType' object has no attribute 'ljust'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 380, in create_lcd_line
>     name_cropped = self.lcd_line[display_id][i]['name'].ljust(name_length)[:name_length]
> AttributeError: 'NoneType' object has no attribute 'ljust'
> 2018-12-07 23:51:48,853 - mycodo.lcd_0c1db50f - INFO - Deactivated in 1203.3 ms
> 2018-12-07 23:52:12,930 - mycodo.lcd_0c1db50f - INFO - Activated in 996.5 ms
> 2018-12-07 23:52:16,949 - mycodo.lcd_0c1db50f - ERROR - Error: 'NoneType' object has no attribute 'ljust'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 380, in create_lcd_line
>     name_cropped = self.lcd_line[display_id][i]['name'].ljust(name_length)[:name_length]
> AttributeError: 'NoneType' object has no attribute 'ljust'
> 2018-12-07 23:52:47,343 - mycodo.lcd_0c1db50f - ERROR - Error: 'NoneType' object has no attribute 'ljust'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 380, in create_lcd_line
>     name_cropped = self.lcd_line[display_id][i]['name'].ljust(name_length)[:name_length]
> AttributeError: 'NoneType' object has no attribute 'ljust'

kizniche, 2018-12-08:
Pushed another commit that should finally fix the IP issue.

Magnum-Pl, 2018-12-08:
Everything was working fine until I tried to set the Period to less than 5 seconds. Now every time I try to save, no matter what I set the period to, it gives me this error:

> Error: Error in the Address (I<sup>2</sup>C) field - This field is required.

I tired deleting the display and rebooting with no luck. It still shows the same info in the I2C address and bus sections.

kizniche, 2018-12-08:
That should be fixed now

kizniche, 2018-12-08:
I need to get going, but if you find any more issues, document them and I'll resume fixing them later.

Magnum-Pl, 2018-12-09:
I updated and was able to get all 4 lines to display correctly, but changing any of the fields results in 
the I2C address field showing up blank. If I hit "Reset LCD", I get this Error 500: Internal Server Error:

> Traceback (most recent call last):
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
>     response = self.full_dispatch_request()
>   File [...]

Here's the log:

> 2018-12-08 18:22:32,567 - mycodo.lcd_f7e7081e - ERROR - Error: invalid literal for int() with base 16: ''
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 104, in __init__
>     self.lcd_i2c_address = int(lcd_dev.location, 16)
> ValueError: invalid literal for int() with base 16: ''
> 2018-12-08 18:22:32,575 - mycodo.lcd_f7e7081e - INFO - Activated in 33.0 ms
> 2018-12-08 18:22:37,458 - mycodo.daemon - ERROR - Could not flash LCD (False): 'LCDController' object has no attribute 'lcd_out'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 666, in lcd_flash
>     return self.controller['LCD'][lcd_id].lcd_flash(state)
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 485, in lcd_flash
>     self.lcd_backlight(True)
>   File "/var/mycodo-root/mycodo/controller_lcd.py", line 472, in lcd_backlight
>     self.lcd_out.lcd_backlight(state)
> AttributeError: 'LCDController' object has no attribute 'lcd_out'
> 2018-12-08 18:24:05,870 - mycodo.lcd_f7e7081e - ERROR - Error: invalid literal for int() with base 16: ''

Rebooting or restarting Mycodo has no effect. Deleting and adding the display again gets everything working again.

kizniche, 2018-12-09:
I just released 7.0.0. I'll incorporate a fix for this into 7.0.1

Magnum-Pl, 2018-12-09:
I loaded the master branch and everything works fine now. 

Can you still add the option to set a line as blank? Also, is it possible to change the period field to allow a setting of less than 5 seconds? The range now is 5-86400 seconds. 

Regarding the "Add Display Set" option, I was going to ask you to change it to be able to add one line at a time, which still might be handy, but I think being able to set each line as blank accomplishes the same thing. It would be interesting to be able to activate and deactivate a display set so that I could configure one display set for a certain application, then deactivate it and activate a second display set for a second application. What was the intended purpose for being able to add extra display sets?

kizniche, 2018-12-09:
> Can you still add the option to set a line as blank?

Just added in the latest commit.

> It would be interesting to be able to activate and deactivate a display set so that I could configure one display set for a certain application, then deactivate it and activate a second display set for a second application.

Activate/deactivate controller is a trigger/conditional action, so this is possible.

> What was the intended purpose for being able to add extra display sets?

So you're not limited to 2 or 4 lines to display on the LCD. If you have more than one display set added, the next in the series will be displayed after the set Period.

closed 2018-12-22 by kizniche